RIOT-OS/RIOT #15626: vera++: reports "warning: keyword 'new' not followed by a single space"

miri64, 2020-12-14:
#### Description
In the static tests `vera++` oftentimes reports "warning: keyword 'new' not followed by a single space". This seems to be a C++-related style issue and should be silenced for C-files.

#### Steps to reproduce the issue
See e.g. https://github.com/RIOT-OS/RIOT/runs/1548755867?check_suite_focus=true#step:5:198

#### Expected results
Do not warn about the C++ keyword 'new' in C-files.

#### Actual results
There is a style issue report for the keyword C++ 'new' in C-files.

#### Versions
Current riotbuild docker.

jia200x, 2020-12-14:
Hmmmm one option would be to simply disable that Vera++ rule. According to Github only 5% of RIOT code is C++, so I don't think it would be an issue.

Alternatively we could try to run different rules for cpp files, but I'm not sure if it's worth the efforts.

miri64, 2020-12-14:
That (and that I really don't know how to configure Vera++) is the reason why I opened an issue rather than a PR to discuss, if it makes sense to also check for C++. In the end, even though we have only 5% C++ code, there [_are_ coding conventions for it](https://github.com/RIOT-OS/RIOT/blob/master/CODING_CONVENTIONS_C%2B%2B.md) which in the long run should also be checked for.

jia200x, 2020-12-14:
wait! I think there's an easy solution. We can add some `--exclusions` to exclude this rule for C files. This should be out of the box accoding to https://bitbucket.org/verateam/vera/wiki/Running

jia200x, 2020-12-14:
In fact, I forgot I added the exclusion file ^^. It's just a matter of adding the regex there

jia200x, 2020-12-14:
could you try the following patch in that PR?
```patch
diff --git a/dist/tools/vera++/exclude b/dist/tools/vera++/exclude
index e69de29bb2..323e335095 100644
--- a/dist/tools/vera++/exclude
+++ b/dist/tools/vera++/exclude
@@ -0,0 +1 @@
+[hc]pp:\d+:.*keyword 'new' not followed by a single space
```

miri64, 2020-12-14:
Since that PR is nearly done building (and on its way to mergin) I would prefer not to delay it by further experiments unrelated to it... AFAIK, static tests can also be tested locally ;-).

miri64, 2020-12-14:
On the branch of https://github.com/RIOT-OS/RIOT/pull/12648:

```
$ BASE_BRANCH=upstream/master dist/tools/vera++/check.sh 
error: invalid command name "hc"
    while executing
"hc"
    invoked from within
"[hc]:\d+:.*keyword 'new' not followed by a single space"
error: invalid command name "hc"
    while executing
"hc"
    invoked from within
"[hc]:\d+:.*keyword 'new' not followed by a single space"
```

jia200x, 2020-12-15:
I found the problem. This syntax is not valid for vera++ 1.2.1 :(. The older version used some TCL arrays:
```
--exclusions` _exclusionsfilename_
:   instructs the program to exclude some source files from rule checks, as described in the given file; the content of this file is a Tcl script that must set a `ruleExclusions` array, where keys are rule names and values are lists of files to omit for the given rule. For example:

        set ruleExclusions(L002) {
            some_file.cpp
        }

        set ruleExclusions(T005) {
            some_file.cpp
            some_other_file.cpp
        }

    Note that the given file names are compared for exact match with the source file names that are provided as parameters to Vera++. This means that links in paths are not resolved for comparison purposes. This option may be used several times.
```

Vera++ 1.3.0 doesn't seem to be available in the latest Ubuntu neither.
This TCL structure sucks because there's no regex. I think we could either:

- Manually add affected C files with TCL syntax
- Avoid this rule for now and hope that we have vera++ 1.3.0 soon in riotdocker.

NB we tried to compile vera++ from source in the beginning, but the Docker image was  ~500MB bigger.

miri64, 2020-12-15:
> * Avoid this rule for now and hope that we have vera++ 1.3.0 soon in riotdocker.

This seems to be no option. Even [Ubuntu 21.04 seems to go for vera++ 1.2](https://packages.ubuntu.com/hirsute/vera++).

jia200x, 2020-12-16:
Then let's add the affected files to the exclusion file

jia200x, 2020-12-16:
alternatively we can try to run Vera++ a third time (it's already running twice because of the 80 chars line warning)

miri64, 2020-12-16:
> alternatively we can try to run Vera++ a third time (it's already running twice because of the 80 chars line warning)

I would prefer that over amending the exclusions file everytime one decides to name a variable `new`.

jia200x, 2020-12-16:
would it be too hard to create a PPA or simply pack vera++ in a .deb? That would solve our problems

miri64, 2020-12-16:
> would it be too hard to create a PPA or simply pack vera++ in a .deb? That would solve our problems

I'm not sure... But I could imagine it being harder and its maintainance more cumbersome than adding an extra line to `dist/tools/vera++/check.sh` ;-).

closed 2021-01-06 by miri64