RIOT-OS/RIOT #15638: Possible memory leak in RIOT/build/pkg/ndn-riot/app.c

tluio, 2020-12-15:
I run make in the directory examples/ndn-ping and get some files in the directory build/pkg/ndn-riot which includes app.c.
I guess there is a memory-leak bug in the function **ndn_app_register_prefix2** if it returns on line 570.

benpicco, 2020-12-18:
That would be a bug in https://github.com/named-data-iot/ndn-riot then.
Can you please also report it upstream?

tluio, 2020-12-18:
Ok, I have reported it upstream just now (https://github.com/named-data-iot/ndn-riot/issues/15).

cgundogan, 2020-12-18:
My understanding is that ndn-riot is not maintained anymore, as ndn-lite is the replacement project. @yan-foto and @tianyuan129 might know more on this?

yan-foto, 2020-12-18:
@cgundogan yes, that's correct! ndn-lite as the generic library of NDN for IoT devices also has a RIOT port which is fairly outdated. Me and @Pairarma have been updating and simplyfing the port (see [my fork](https://github.com/yan-foto/RIOT/tree/feature/ndn-lite)) and @Pairarma has also initiated an effort to remove GNRC from our port (see [here](https://github.com/yan-foto/RIOT/tree/feature/no-gnrc)).

As soon as we have the non-GNRC version up and running, we will send a PR!

closed 2023-05-23 by bors[bot]