RIOT-OS/RIOT #15654: Clear BOARD parameters in the board documentation

LordTy, 2020-12-16:
#### Description
When looking up a board in the documentation for example [nRF52840-Dongle](https://riot-os.org/api/group__boards__nrf52840dongle.html), there is no information on which parameter to use in your makefile.
The disconnect in this case is that the board is called: nRF52840-Dongle while the makefile requires BOARD=nrf52840dongle

You could probably find this same problem in other boards.

It would be nice if there was a clear place in the board documentation that specified the make variable that has to be used. 

Some boards do show their make parameter in the Flash the Board instruction, but not all of them.

### Useful links

chrysn, 2020-12-16:
I'm looking through how to both solve this particular case and the generic one. A brief `grep -c BOARD boards/*/doc.txt` reveals that this affects about half the boards.

I'm playing around with different styles of generic boxes (or generic info lines) we could put in on all the board pages. The board name should be in there, not sure yet if anything else.

chrysn, 2020-12-18:
Just for cross-referencing: Apart from the quick (but failing-at-doxygen-obscurities) fix in #15658, this also spun up the discussion in <https://forum.riot-os.org/t/boards-metadata/161>.

LordTy, 2020-12-18:
Thanks for the quick response. I will look at the forum for further discussion related to the documentation.