RIOT-OS/RIOT #15685: esp32: can't use newer C++ standard than c++11

jenswet, 2020-12-19:
#### Description
RIOT projects cannot use a newer C++ standard like c++17 with the ESP32 .

In the nvs_flash vendor module the C++ version is fixed and cannot be overwritten.
https://github.com/RIOT-OS/RIOT/blob/54062971767a8218325991c34a0d0487a189ded5/cpu/esp32/vendor/esp-idf/nvs_flash/Makefile#L6

#### Steps to reproduce the issue
Adding `CXXEXFLAGS += -std=c++17` does not work. The c++11 standard is added later in the CXXFLAGS and therefore overwrites the newer standard.

#### Expected results
The newer user desired standard should be applied.

#### Actual results
std=c++11 is always used.

#### Potential fix
I'd suggest something like this:
```
ifeq (,$(findstring -std=,$(CXXFLAGS)))
	CXXFLAGS += -std=c++11
endif
```
A way to add the flag if no standard newer than c++11 is present would be better. But I don't know Makefiles very well. I usually work with CMake.

jenswet, 2020-12-20:
I just noticed the official RIOT ESP32 toolchain is still based on the GCC 5.2 version from Espressif. So this only makes sense when the RIOT toolchain is also updated to the newer GCC 8.4 version shipped with ESP-IDF 4 and newer. This is probably a bigger task.

gschorcht, 2021-10-04:
Unfortunately the RIOT port for ESP32 is much more than just using the ESP-IDF from Espressif. Since the ESP-IDF is based on FreeRTOS, we had to reimplement many functions provided by the ESP-IDF and we only use some extracts from the original SDK. In fact, the RIOT port is a mixture of functions needed for compatibility with the binary libraries of the ESP-IDF, modified functions extracted from ESP-IDF and the actual RIOT port. At the moment I don't see any way to update the RIOT port to a newer version of ESP-IDF, because it has changed a lot since version 3.1, if not too much.

jenswet, 2021-10-05:
Hi @gschorcht,

I see.

I was hoping that we don't need to update RIOT to the new ESP-IDF to fix this issue.

It should've been enough to use the new toolchain that Espressif published together with the new IDF version.

But unfortunately the new toolchain isn't officially supported for older ESP-IDF versions: https://www.esp32.com/viewtopic.php?t=7400

> This toolchain is compatible with ESP-IDF starting from commit 70eda3d2 (latest master branch as of 2019/04/11).
> This toolchain is not compatible with older versions of ESP-IDF.

That means a minimum of v4.0.0...

jenswet, 2022-01-18:
Seems like this is fixed with #17479.

Of course, there is still the problem of compiler support for newer standards, but the user's choice of standard is no longer forcibly overridden by C++11.

closed 2022-01-18 by jenswet