RIOT-OS/RIOT #15687: Current offset in CoAP blockwise transfer

Citrullin, 2020-12-20:
#### Description
In order to not generate the WoT TD multiple times for #14095, I need a way to receive the current offset of blockwise transfer.

### Useful links
- [Forum post](https://forum.riot-os.org/t/how-to-use-coap-blockwise-transfer/80)
- @bergzand mentioned it on Matrix 
  > Quickly skimming over it, it looks like the blockwise could use a function to retrieve the current offset, a generator could use that to skip certain parts

Citrullin, 2022-03-16:
It probably is still an issue. @JKRhb Do you remember if that was still an issue?

JKRhb, 2022-03-18:
Hmm, I think the blockwise serialization seems to work fine at the moment, right? If I am not mistaken, the offset should be calculated automatically from the block2 option when using `coap_block2_init`.