RIOT-OS/RIOT #15691: examples/posix_socket: doesn't work between 2 iotlab-m3

aabadie, 2020-12-22:
#### Description

The `examples/posix_socket` doesn't work when exchanging udp messages between 2 iotlab-m3 using link-local addresses. It was working in 2020.01 and no longer works since **2020.04**.

It works when using the broadcast address `ff02::1`.

#### Steps to reproduce the issue

To test this, I used the IoT-LAB testbed as follows:

- Submit an experiment with 2 iotlab-m3:
```
$ iotlab-experiment submit -n "riot-udp" -d 120 -l 2,archi=m3:at86rf231+site=saclay
$ iotlab-experiment wait
```
- Build and flash the `examples/posix_sockets` on both M3 of the experiment:
```
$ make BOARD=iotlab-m3 -C examples/posix_sockets
$ iotlab-node --flash examples/posix_socket/bin/iotlab-m3/posix_sockets_example.bin
```
- Open a terminal on both M3 (let's say you got m3-10.saclay and m3-11.saclay, m3-11.saclay will be used as server, m3-10 as client):
```
$ IOTLAB_NODE=m3-11.saclay.iot-lab.info make BOARD=iotlab-m3 -C examples/posix_sockets term
```
and in another terminal:
```
$ IOTLAB_NODE=m3-10.saclay.iot-lab.info make BOARD=iotlab-m3 -C examples/posix_sockets term
```
- On the server node, print the interface configuration using ifconfig and then start an udp server listening on port 8888:
```
> ifconfig
ifconfig
Iface  5  HWaddr: 0B:BE  Channel: 26  Page: 0  NID: 0x23  PHY: O-QPSK 
          
          Long HWaddr: E6:D3:6F:EC:DF:36:0B:BE 
           TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
          AUTOACK  ACK_REQ  CSMA  L2-PDU:102  MTU:1280  HL:64  6LO  
          IPHC  
          Source address length: 8
          Link type: wireless
          inet6 addr: fe80::e4d3:6fec:df36:bbe  scope: link  VAL
          inet6 group: ff02::1
> udp server start 8888
udp server start 8888
Success: started UDP server on port 8888
```
- On the client side, send an udp paquet to the server:
```
> udp send fe80::e4d3:6fec:df36:bbe 8888 hello
```

#### Expected results

The server prints the following message:

```
> Received data: hello
```

#### Actual results

The server remains silent => no paquet is received

#### Versions

Current master, but this is broken since 2020.04 and it was working in 2020.01

miri64, 2020-12-22:
Duplicate of https://github.com/RIOT-OS/RIOT/issues/15649?

aabadie, 2020-12-22:
Ok, thanks! The interface identifier must be part of the link-local address. This was not the case before.

closed 2020-12-22 by aabadie