RIOT-OS/RIOT #15692: Terminal program <...> is required but not found in PATH. Aborting.

Godlux, 2020-12-22:
Hello! I'm trying to run following code on my arduino Mega:

```
#include "crypto/ciphers.h"
cipher_t cipher;
uint8_t key[AES_KEY_SIZE] = {0},
        plain_text[AES_BLOCK_SIZE] = {0},
        cipher_text[AES_BLOCK_SIZE] = {0};
if (cipher_init(&cipher, CIPHER_AES_128, key, AES_KEY_SIZE) < 0)
    printf("Cipher init failed!\n");
if (cipher_encrypt(&cipher, plain_text, cipher_text) < 0)
    printf("Cipher encryption failed!\n");
else
    od_hex_dump(cipher_text, AES_BLOCK_SIZE, 0);
```
from: https://api.riot-os.org/group__sys__crypto.html

But after pasting this example i theese error:
**Terminal program /home/user/RIOT/RIOT/examples/hello-cipher-text1/bin/native/hello-cipher-text1.elf is required, but not foun in PATH. Aborting.**

Before adding theese code my "hello-world" had compiled without such errors. What i'm doing wrong?

UP:
When i do "make" i'm getting some alerts like:
    undefined reference to 'CIPHER_AES_128' in main.c:58

Citrullin, 2020-12-25:
I guess the issue is your 8-Bit Microcontroller. Not sure, if the libraries support such a Microcontroller.
What library to use with your Arduino Mega?

leandrolanzieri, 2021-01-06:
> Terminal program /home/user/RIOT/RIOT/examples/hello-cipher-text1/bin/native/hello-cipher-text1.elf is required, but not foun in PATH. Aborting.

If you look at the path it seems you are running `make term` for native and not for the platform you intend.  You should set the environmental variable like so: `BOARD=arduino-mega2560 make all flash term`.

> When i do "make" i'm getting some alerts like:
> undefined reference to 'CIPHER_AES_128' in main.c:58

That is a variable that is declared as `extern` by `crypto/ciphers.h`, so if the module has been added to the `USEMODULE` variable in the application Makefile it should be defined:

```Makefile
USEMODULE += crypto_aes
```

fjmolinas, 2021-01-20:
@Godlux where you able to try what @leandrolanzieri suggested?

fjmolinas, 2022-03-02:
Stale, closing.

closed 2022-03-02 by fjmolinas