RIOT-OS/RIOT #15710: periph_can is missing into nucleo-f767zi/Makefile.features

donsez, 2021-01-06:
#### Description
periph_can is missing into nucleo-f767zi/Makefile.features

nucleo-f767zi board and stm32f767zi CPU have 3 × CANs (2.0B Active) interfaces

### Useful links
* https://github.com/RIOT-OS/RIOT/blob/master/boards/nucleo-f767zi/Makefile.features
* Section 2.33 in https://www.st.com/resource/en/datasheet/stm32f767zi.pdf

donsez, 2021-01-07:
Few changes are required for add the can_periph feature.

       > git diff boards/nucleo-f767zi/Makefile.features
      diff --git a/boards/nucleo-f767zi/Makefile.features b/boards/nucleo-f767zi/Makefile.features
      index 672ce4400..4780e635f 100644
      --- a/boards/nucleo-f767zi/Makefile.features
      +++ b/boards/nucleo-f767zi/Makefile.features
      @@ -2,7 +2,9 @@ CPU = stm32
       CPU_MODEL = stm32f767zi
       
       # Put defined MCU peripherals here (in alphabetical order)
      +FEATURES_PROVIDED += periph_can
       FEATURES_PROVIDED += periph_dma
      +FEATURES_PROVIDED += periph_eth
       FEATURES_PROVIDED += periph_i2c
       FEATURES_PROVIDED += periph_rtc
       FEATURES_PROVIDED += periph_rtt
      @@ -10,7 +12,6 @@ FEATURES_PROVIDED += periph_spi
       FEATURES_PROVIDED += periph_timer
       FEATURES_PROVIDED += periph_uart
       FEATURES_PROVIDED += periph_usbdev
      -FEATURES_PROVIDED += periph_eth
       
       # Put other features for this board (in alphabetical order)
       FEATURES_PROVIDED += riotboot
      
      > git diff cpu/stm32/include/candev_stm32.h
      diff --git a/cpu/stm32/include/candev_stm32.h b/cpu/stm32/include/candev_stm32.h
      index 0d5afe5a7..3193848d5 100644
      --- a/cpu/stm32/include/candev_stm32.h
      +++ b/cpu/stm32/include/candev_stm32.h
      @@ -37,12 +37,13 @@ extern "C" {
       
       #include "can/candev.h"
       
      -#if defined(CPU_LINE_STM32F413xx) || defined(CPU_LINE_STM32F423xx)
      +#if defined(CPU_LINE_STM32F413xx) || defined(CPU_LINE_STM32F423xx) || \
      +       defined(CPU_LINE_STM32F767xx)
       #define CANDEV_STM32_CHAN_NUMOF 3
       #elif defined(CPU_FAM_STM32F1) || defined(CPU_FAM_STM32F2) || defined(CPU_FAM_STM32F4)
       #define CANDEV_STM32_CHAN_NUMOF 2
       #elif defined(CPU_FAM_STM32F0) || defined(CPU_FAM_STM32F3) || \
      -      defined(CPU_FAM_STM32L4) || DOXYGEN
      +      defined(CPU_FAM_STM32L4) || defined(CPU_FAM_STM32F7) ||DOXYGEN
       /** Number of channels in the device (up to 3) */
       #define CANDEV_STM32_CHAN_NUMOF 1
       #else
      
      > git diff sys/auto_init/can/auto_init_periph_can.c
      diff --git a/sys/auto_init/can/auto_init_periph_can.c b/sys/auto_init/can/auto_init_periph_can.c
      index 0c2ffb669..8fa49fbdd 100644
      --- a/sys/auto_init/can/auto_init_periph_can.c
      +++ b/sys/auto_init/can/auto_init_periph_can.c
      @@ -16,6 +16,7 @@
        * @}
        */
       
      +#include "periph/can.h"
       #include "can/device.h"
       #include "can_params.h"

fjmolinas, 2021-01-20:
Hi @donsez do you have the BOARD to test this? If so since you are already providing the patch would you mind opening a PR for this :)

donsez, 2021-01-20:
Hi @fjmolinas ,
I will test the patch ASAP with a CAN bus setup : F767ZI + F446RE + 2 L9616 CAN drivers.

wosym, 2021-11-20:
Is there any reason for not closing this issue yet? I see it has been added and removed to the Release milestone recently, even though a patch was merged almost a year ago?

fjmolinas, 2021-11-20:
> Is there any reason for not closing this issue yet? I see it has been added and removed to the Release milestone recently, even though a patch was merged almost a year ago?

thanks for bringing it back to light, closing!

closed 2021-11-20 by fjmolinas