RIOT-OS/RIOT #15776: skald: advertising channel configuration broken by kconfig remodel

haukepetersen, 2021-01-15:
#### Description
The adaption to kconfig broke the advertising channel configuration for `skald`: the purpose of the `SKALD_ADV_CHAN` define is to allow for any arbitrary number and order of channels to be defined. Although only 3 of the possible 39 channels should be used for advertising by the spec, there might be situations to also use some of the remaining data channels, e.g. for testing or experimentation.

Also the current form of parameters does not allow for changing the order of the given channels through kconfig...

It seems to me, that one should be able to simply specify a list of comma separated values (e.g. `37,38,39`) as configuration value through kconfig. However, I have no idea how to do that, so I leave this to the pros :)

#### Steps to reproduce the issue
n/a

#### Expected results
should be possible to define any set of channels to use for advertising, e.g. `17,39,7,12`

#### Actual results
fixed order and limited to the 3 advertising channels

#### Versions
master

akshaim, 2021-01-15:
Oops! This was done in PR #14824.  I could introduce an option to input arbitrary values. What could be the maximum number of channels used for advertising?  As per docs [1], we have 40 channels (39 indices), so should that be the limit?

[1] : https://ww2.mathworks.cn/help/comm/ug/ble-channel-selection-algorithms.html;jsessionid=5bf0ea011b230e4c320b1cd7165b

haukepetersen, 2021-01-15:
correct. Any number of channels, from 1 to 40, should be possible.

akshaim, 2021-01-18:
@haukepetersen Created a draft PR #15790 . Can you please test it on a device, if possible?

PR #15790 (2021-01-26): net/skald: Update Kconfig implementation
### Contribution description

This PR fixes issue #15776 

### Testing procedure

1. New documentation was built using Doxygen 

   The build works fine.

  2. Build successful. Tested this [example](https://github.com/akshaim/RIOT/commit/1d13658430c0ad77de08429e174c292a2e16dc85). Results below

      Compiled binaries for nrf52dk

#### Default State:

##### Firmware Output

```
main(): This is RIOT! (Version: 2021.04-devel-177-ga9da09-Kconfig_skald_update)
CONFIG_SKALD_INTERVAL=(1 * (1000000LU))
3
37
38
39
```

#### Usage with menuconfig :

`make menuc [... cut]
closed 2021-01-26 by leandrolanzieri