RIOT-OS/RIOT #15814: usb-serial/list-ttys.sh: Broken when a debugger offers multiple serial ports

bergzand, 2021-01-20:
#### Description

Some USB devices offer multiple serial interfaces. The `list-ttys.sh` bash script used to be able to show these all as belonging to the same USB device. This is probably broken since #15351 and now multiple directories found by `find` are returned as a single string.

#### Steps to reproduce the issue

run `make -C examples/hello-world list-ttys` with a hifive1b connected, or another board that has a debugger with multiple serial ports.

#### Expected results

```
/sys/bus/usb/devices/1-1.1.2.3: SEGGER HiFive, serial: '000979014634', tty(s): ttyACM5, ttyACM4
```

#### Actual results

```
find: ‘/sys/bus/usb/devices/1-1.1.2.3/1-1.1.2.3:1.2/tty\n/sys/bus/usb/devices/1-1.1.2.3/1-1.1.2.3:1.0/tty’: No such file or directory
```

#### Versions

```
GNU bash, version 5.0.18(1)-release (x86_64-pc-linux-gnu)
```

#### Possible fix

As the script is already an bash script, a fix would be to store the result from find in an array like this:
https://stackoverflow.com/questions/23356779/how-can-i-store-the-find-command-results-as-an-array-in-bash/54561526#54561526

aabadie, 2023-01-04:
Is this fixed by #17737 ?

maribu, 2023-01-05:
The python script indeed works in the given situation and now is used in `make list-ttys`.

I didn't dare to remove the old bash scripts, though. But every use case the bash scripts can handle, the python script can handle better. So maybe it is time to just drop them?

aabadie, 2023-01-05:
> So maybe it is time to just drop them?

I always happy when code can be removed :)

closed 2023-01-07 by bors[bot]