RIOT-OS/RIOT #15815: NIB Cache Entries

ad6sh, 2021-01-20:
#### Description

I am implementing an application using RIOT where I need to get the information about neighbors.My plan is to ping neighbors,which should automatically populate the neighbor cache .I could manually write and read NIB using the functions provided the following link. But after pinging, the neighbor information is not automatically added to the cache.Is there any specific way to do that(for eg. adding any modules to the makefile)?. I have already tried adding gnrc_ipv6_nib, auto_init_gnrc_ipv6_nib modules.

### Useful links
https://www.riot-os.org/api/group__net__gnrc__ipv6__nib__nc.html

miri64, 2021-01-20:
Have you tried deactivating [6Lo-ND](https://tools.ietf.org/html/rfc6775) in case you are on a 6LoWPAN device? This way link-local addresses (in case you use those for pinging) are also added to the neighbor cache. You can do so by setting [CONFIG_GNRC_IPV6_NIB_6LN](https://www.riot-os.org/api/group__net__gnrc__ipv6__nib__conf.html#gaff67513e6d48b405a4f8636df3a01857) to 0. You also might need to increase the maximum number of neighbor cache entries using [CONFIG_GNRC_IPV6_NIB_NUMOF](https://www.riot-os.org/api/group__net__gnrc__ipv6__nib__conf.html#ga9dc4943e8cb6c9b2fec3fb54ed1395a8). Don't let yourself get confused by the default values given in the documentation. Use `make menuconfig` to see the default values depending on your scenario.

ad6sh, 2021-01-20:
Thanks for your quick reply :) I am using [MSB-IoT](https://doc.riot-os.org/group__boards__msbiot.html). I have tried disabling 6Lo-ND and Increased neighbor cache size. Then the ipv6 address which i give for ping command gets added to the Neighbor cache,irrespective of whether it reply to the ping or not.But ping  gives 100% loss all the time and ipv6 address of device doesn't shows up using ifconfig  command.Is this an expected behavior ?

miri64, 2021-01-20:
Nope, I just forgot to mention that the [address-resolution state-machine](https://doc.riot-os.org/group__net__gnrc__ipv6__nib__conf.html#ga76a973340653df11c95840ca3a70d90e) also needs to be activated.

ad6sh, 2021-01-20:
Thank you much :) Everything works  works pretty well !!

closed 2021-01-20 by miri64