RIOT-OS/RIOT #15822: boards/rpi-pico: add support

tprrt, 2021-01-21:
#### Description
Add the support for the mcu RP2040 as weel as the  Raspberry-Pi Pico board.

Raspberry Pi Pico is a low-cost, high-performance microcontroller board with flexible digital interfaces. Key features include:

    RP2040 microcontroller chip designed by Raspberry Pi in the United Kingdom
    Dual-core Arm Cortex M0+ processor, flexible clock running up to 133 MHz
    264KB of SRAM, and 2MB of on-board Flash memory
    Castellated module allows soldering direct to carrier boards
    USB 1.1 with device and host support
    Low-power sleep and dormant modes

    Drag-and-drop programming using mass storage over USB
    26 × multi-function GPIO pins
    2 × SPI, 2 × I2C, 2 × UART, 3 × 12-bit ADC, 16 × controllable PWM channels
    Accurate clock and timer on-chip
    Temperature sensor
    Accelerated floating-point libraries on-chip
    8 × Programmable I/O (PIO) state machines for custom peripheral support

### Useful links
https://www.raspberrypi.org/blog/raspberry-pi-silicon-pico-now-on-sale/
https://raspberrypi.github.io/pico-sdk-doxygen/index.html

aabadie, 2021-07-05:
See #16609

maribu, 2021-07-14:
Maybe we can keep the issue open a bit to sync who's working on which feature

maribu, 2021-07-14:
- [x] `periph_timer`: @fabian18 is working on this, https://github.com/RIOT-OS/RIOT/pull/16627 is already in pretty good shape
- [x] DMA: @maribu is working on a HAL for memory to memory transfers (that should also be possible to implement for other MCUs) and some low level helper functions that should aid other RPx0xx peripheral drivers in implementing transfers from and to memory mapped I/O
- [x] `periph_uart`: Basic support is upstream, @maribu is working on DMA support as a test use case
- [ ] `periph_spi`: ?
- [ ] `periph_i2c: ?
- [ ] `periph_adc`: ?
- [ ] `periph_pwm`: ?
- [ ] `periph_pm`: ?
- [ ] `periph_usb`: ?
- [ ] `periph_pio`: I think @fabian18 did take a look into it. It certainly is one of the more interesting features of the MCU, IMO.

smurfix, 2022-05-03:
Is anybody working on enhancing this? there seem to be some missing peripherals …

fabian18, 2022-05-03:
> Is anybody working on enhancing this? there seem to be some missing peripherals …

Any unassigned peripheral is probably waiting to be implemented. Right now, I have more important work to do, than to continue to implement PIO support. Do you require any particular peripheral, or are you asking in general?

smurfix, 2022-05-03:
Besides PIO, ADC support would be nice to have, at least for my current project.

fengelhardt, 2023-03-21:
I am working on SPI support right now. I can maybe add some more, it seems not that difficult.

dylad, 2023-04-22:
I've started to work on an ADC driver for MCU.

dylad, 2023-04-28:
`periph_pwm` is my next target.

Mac6750, 2023-09-20:
I'm new here. Can I add i2c?

benpicco, 2023-09-20:
Sure, go ahead!
Currently I2C is implemented via PIO, but afaik there is also a native I2C peripheral in the RP2040

maribu, 2023-09-20:
yes there is :-) And @fabian18 already has already created the stubs where it would call to either PIO or native I2C implementation.

Thanks for picking this up :-)

MacT900, 2023-09-22:
Can I ask questions here or should I ask in forum? How and from what I create branch or pull request?

dylad, 2023-09-22:
If you have a question about something related to RP2040 (hardware or software). You can ask it here.
Otherwise, if you have questions about RIOT itself, or something else, it would be better to go to the forum. (This way this issue can focus on RP2040 only).

I strongly suggest you to have a look at a few entries in our wiki regarding RIOT development and contribution:
[How to make a pull request](https://docs.github.com/fr/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/about-pull-requests)
[Contribution guidelines](https://github.com/RIOT-OS/RIOT/blob/master/CONTRIBUTING.md)
[Coding Conventions](https://github.com/RIOT-OS/RIOT/blob/master/CODING_CONVENTIONS.md)

Feels free to join us on Matrix, we might also help you there.

studiosoftdev, 2023-10-30:
I'm currently doing a bachelors degree level project and intend to check some items off this list - looking at starting with pwm or i2c, then pm, usb and pio if there's no existing implementation for that. Posting here to ask what the current state of these features are so I don't try and do something that someone else is already doing :)

dylad, 2023-10-31:
I have the `periph_pwm` almost ready. Just need to do some testing on my side. Don't waste time on it ;)

MacT900, 2024-02-17:
Hi, I'm back from little break. Should I add an option to use I2C as a slave and  where I can find info about I2C drive strength?

dylad, 2024-02-22:
Hi @MacT900,
In the current state, RIOT doesn't support I2C slave. There is discussion about this [here](https://github.com/RIOT-OS/RIOT/issues/19560)
Regarding I2C drive strength, you could either set it to a compile-time default value in `periph_cpu.h` or create a special field in the I2C configuration structure defined in `periph_conf.h`. The current I2C API doesn't handle pull strength selection in the current state.

dylad, 2024-03-19:
`periph_pwm` support was just merged.

fengelhardt, 2024-03-23:
I will give `periph_usbus` a try.

dylad, 2024-03-23:
> I will give periph_usbus a try.

That's nice ! I have some experience with this, so don't hesitate to ask if you need help or have any questions.

fengelhardt, 2024-08-18:
I have my `periph_usbus` code almost ready. However, I have doubts if I understood correctly what the two event handlers should do, and what should happen in the ISR and in the xmit functions.

Maybe I could push my code as a WIP Pull Request to clarify?

The code is compiling but not working at all. I just ordered a Black Magic Probe for debugging just in case :smile:

dylad, 2024-08-18:
Please go ahead.
That's something I can help to polish.
No promises on how fast I'll be but that's something I'd like to see merged

tanneberger, 2025-01-31:
@dylad @MacT900 @benpicco 

I am right now building an application for which I require I2C running on the pico. What is the status of your implementation? I am happy to test your WIP implementations ;D

dylad, 2025-01-31:
@tanneberger 
Currently the easiest way to use I2C on rpi-pico is through PIO.

There is an open PR for hardware I2C opened by @studiosoftdev at #20677 but they didn't answer to review yet.

tanneberger, 2025-02-02:
@dylad I tried using I2C and I think it uses the PIO under the hood, but that crashes the kernel when calling `i2c_init`.

I will also try out that PR thank you.

dylad, 2025-02-03:
@tanneberger If you encountered an issue with I2C/PIO on RP2040, please open an issue to describe it. It should works at out of the box (at least that was the case last time I played with it).
If some regressions were introduced, I can have a look at it and try to fix it soon.

tanneberger, 2025-02-04:
@dylad I created Issue #21187

crasbe, 2026-10-07:
Considering we don't really have issues for every board that keeps track of the implementation status and the only missing pieces are USB, PM and (perhaps) hardware I2C, I'll close this issue now.

closed 2026-10-07 by crasbe