RIOT-OS/RIOT #15842: pkg/tinydtls: DTLS process gets completly borked

janosbrodbeck, 2021-01-23:
#### Description
I'm not quite sure whether an issue is right for such problem. I've run into an issue with DTLS/tinydtls with my gcoap development where I'm relatively sure it's not related to gcoap or my DTLS implementation. At first I thought this got fixed by another problem I found in the `sock_dtls_send_aux()` function, but sadly no. 

#### Steps to reproduce the issue
Only testable on my [gcoap dtls-dev branch](https://github.com/janosbrodbeck/RIOT/tree/gcoap/pr/dtls-dev). Contains some more printing/info at the moment.

* Flash my branch on two native instances 
#### Expected results

#### Actual results

#### Versions

janosbrodbeck, 2021-01-23:
@pokgak Maybe you have an idea what could be the problem here?

pokgak, 2021-01-23:
I cannot reproduce the zero address but I can see the server not responding to the ClientHello. I enabled the debug log of tinydtls and got the following after the client sends the second request:

```
> Jan 23 06:53:01 DEBG dtls_handle_message: FOUND PEER
Jan 23 06:53:01 DEBG got packet 22 (67 bytes)
Jan 23 06:53:01 ALRT No security context for epoch: 0
Jan 23 06:53:01 INFO decrypt_verify() failed
```

The server actually received the ClientHello but it cannot continue the handshake because it thinks that the session is still valid. I found in the [TLS RFC][1] that closing a session in (D)TLS requires both sides to send a close_notify. So when you terminate the client before the server closed the session, the server is still waiting for the close_notify from the client back.

In tinydtls, it set the session to `DTLS_STATE_CLOSING` after one side sends a close_notify and after receiving a close_notify from the other side, changes it to `DTLS_STATE_CLOSED` and **only then** it frees the allocated storage for the peer.

---
So I think you need a way to cleanly end the session when the client just goes offline for whatever reason. tinydtls has a function `dtls_reset_peer`, which should free the peer. Maybe you should try using this in the DTLS session management module e.g. after sending a close_notify, if the client does not responds with another close_notify, assume its offline and resets the peer.

Hope that helps :)

[1]: https://tools.ietf.org/html/rfc5246#section-7.2.1

pokgak, 2021-01-23:
btw in the section linked in the RFC, it says

> If the application protocol will not transfer any additional data, but will only close the underlying transport connection, then the implementation MAY choose to close the transport without waiting for the responding close_notify.

so for dsm module it should be safe to just resets session directly without waiting for close_notify from client

janosbrodbeck, 2021-01-24:
That really helped me, thank you :)
The line `ALRT No security context for epoch: 0` gave me a hint and I found it in the RFC again. If I understand it correctly, my scenario is perfectly described [here](https://tools.ietf.org/html/rfc6347#section-4.2.8). The conditions are the same: Client silently abandons the session and starts a new handshake which epoch 0. I do not think this scenario is unusual. IMHO this should not be handled on application layer. 

The fix I've hinted in my first post actually fixes this problem on client side (when the peer is already known but not connected). I have to do a bit more reading though...

pokgak, 2021-01-25:
> But that will lead to problems again later, when the node wants to make a request again.

I don't see how that could happen as the client will just have to redo handshake again if the server decides to reset its session with said client. But I can see that this might not be ideal because the client will need to do a full handshake almost every time.

Other than that, I cannot find any reference to session resumption in tinydtls. There is renegotiation but that seems like for another use case. So even though the RFC has suggested a solution for this, you might need to make changes to tinydtls to support it.

janosbrodbeck, 2021-01-25:
I had also been afraid of this, because I didn't find anything in tinyDTLS either. I guess for now I'll add this to "known issues", finish my gcoap approach and come back to this issue at a later point. IMHO this should not be handled in either gcoap nor session management.

Thanks again for your help ^^

closed 2021-01-25 by janosbrodbeck