RIOT-OS/RIOT #15850: Run emCute example on ESP32 board

thanosmanolis, 2021-01-24:
Hello,

I am trying to run the **emcute_mqttsn** example on an ESP32 board (I've already managed to run it on native, with no problem whatsoever). Ideally I would like to run the Mosquitto RSMB on a Raspberry Pi Zero, but for starters I am trying the example running it on my computer.

So in order to do that, after seeing #6956 I run **gnrc_border_router** on another ESP32 board. The problem is, when I try to add the IPv6 address generated from **gnrc_border_router** to my **emcute_mqttsn** node, I get the following message:

> error: unable to add IPv6 address

However, I am able to ping the **emcute_mqttsn** IPv6 from the **gnrc_border_router** and vice versa. Could you think of what I am doing wrong?

Thank you.

akshaim, 2021-01-26:
ping @haukepetersen

thanosmanolis, 2021-02-06:
Any help on that? I could provide more information of course if needed. :smiley:

jeandudey, 2021-03-20:
>error: unable to add IPv6 address

Have you tried to increase the maximum number of IPv6 addresses on your node?

You can do it either by,

- `make menuconfig` -> System -> Networking -> GNRC Network stack -> Configure GNRC network interface, and change
  - Select `(3) Maximum number of unicast and anycast addresses per interface` and increase it by 1 or any number of addresses you need.
  - Save changes with `S`, exit.
- By creating an `app.config` file on the application directory and setting the option `CONFIG_GNRC_NETIF_IPV6_ADDRS_NUMOF`
  - e.g: `GNRC_NETIF_IPV6_ADDRS_NUMOF=4`
- Adding it to the `CFLAGS` variable, e.g.: `CFLAGS="-DCONFIG_GNRC_NETIF_IPV6_ADDRS_NUMOF=4" make flash`

Do not forget to recompile and flash 👍🏻

thanosmanolis, 2021-03-22:
@jeandudey Thank you for your answer. The problem was solved!

However, another problem occurred. After I add the IPv6 address to my node (which is an ESP32 board), I try to connect to the Mosquitto RSMB broker (which is executed on my laptop) with the following command, just like in the emCute example:

`con 2001:db8::32ae:a4ff:fe30:83e5 1885` 

This IPv6 address is the one I added to my node, after it was produced by the border_router which is executed on another ESP32 board.

So after executing this command, the following error occurs:

> error: unable to connect to [2001:db8::32ae:a4ff:fe30:83e5]:1885

**P.S.** Should I open a new issue for this error? I am sorry if I should.

jeandudey, 2021-03-23:
>However, another problem occurred. After I add the IPv6 address to my node (which is an ESP32 board), I try to connect to the Mosquitto RSMB broker (which is executed on my laptop) with the following command, just like in the emCute example:
 
Have you a route to your border router node from the emCute one? you can check with `ipv6 nib route`, if not RPL maybe probably not initialized correctly (in theory you shouldn't add the IPv6 address manually on your nodes), you may need to intialize it on the root and other nodes, [see this guide for example](https://www.iot-lab.info/legacy/tutorials/riot-rpl-m3/index.html)

thanosmanolis, 2021-03-24:
@jeandudey By checking with `nib route` at my **emCute** node, I get the output 

> default* via fe80::260a:c4ff:fe00:93d3 dev #&#8203;8

which is the ip configured in **border_router** node. After many trials, I realized that after starting the **border_router** node, I can connect to the RSMB broker (from the **emCute** node) with the command `con 2a02:587:5421:258b:9358:4853:f7ec:3c9 1885` (This is the IPv6 I get as output by the `ifconfig` command in my laptop). Does that mean that everything works well? I thought that in the `con` command I should put the IP produced by the **border_router** node.

Here is also a screenshot of my terminals (upper left: **emCute** node, lower left: **border_router** node, upper right: RSMB broker, lower right: `ifconfig` output showing my laptop's IPv6).

[image]

jeandudey, 2021-03-25:
>Does that mean that everything works well? I thought that in the con command I should put the IP produced by the border_router node.

Yes, the border router has its own global IPv6 but only relays traffic from other nodes, in this case from your emCute one to your computer public IPv6 and vice versa

jeandudey, 2021-03-26:
I'm closing this since it's fixed, do not hesitate to open new issue if you find a bug. See also the [forum](https://forum.riot-os.org) about all related to RIOT 😉

closed 2021-03-26 by jeandudey