RIOT-OS/RIOT #15870: examples/micropython: floating point exception while testing on native

leandrolanzieri, 2021-01-27:
#### Description
While running the test for the `examples/micropython` application on native, a floating point exception is raised. I tried building in Docker and I get the same issue. Current master pointing to 4509dc681247c80064ebe2be850f3c6984439d40.

#### Steps to reproduce the issue
1. Build `examples/micropython` for `native`
2. Run `make test`

#### Expected results
- Passing test

#### Actual results
- Floating point exception is raised
```
/home/leandro/Work/RIOT/examples/micropython/bin/native/micropython.elf /dev/ttyACM0 
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2021.04-devel-338-g4509d)
-- Executing boot.py
boot.py: MicroPython says hello!
-- boot.py exited. Starting REPL..
MicroPython v1.4.2-6568-gbb8e51f6d on 2021-01-27; riot-native with native
Type "help()" for more information.
>>> print("echo this! " * 4)
print("echo this! " * 4)
echo this! echo this! echo this! echo this! 
>>> import riot
print(riot.thread_getpid())
import riot
>>> print(riot.thread_getpid())
2
>>> import utime
import utime
>>> utime.time()
utime.time()
3257
>>> utime.sleep_ms(0)
utime.sleep_ms(0)
>>> utime.time()
utime.time()
3358
>>> test_sleep(0, 1000): slept 101ms
utime.time()
utime.time()
3408
>>> utime.sleep_ms(50)
utime.sleep_ms(50)
>>> utime.time()
utime.time()
3560
>>> test_sleep(50, 111): slept 152ms
utime.time()
utime.time()
3610
>>> utime.sleep_ms(250)
utime.sleep_ms(250)
>>> utime.time()
utime.time()
3962
>>> test_sleep(250, 111): slept 352ms
utime.time()
utime.time()
4012
>>> utime.sleep_ms(500)
utime.sleep_ms(500)
>>> utime.time()
utime.time()
4614
>>> test_sleep(500, 111): slept 602ms
import xtimer
import xtimer
>>> a = 0
a = 0
>>> def inc_a(): global a; a+=1
def inc_a(): global a; a+=1
... 

>>> t = xtimer.xtimer(inc_a)
t = xtimer.xtimer(inc_a)
>>> utime.time()
utime.time()
4917
>>> t.set(500000)
t.set(500000)
>>> while a==0: pass
while a==0: pass
... 

>>> utime.time()
utime.time()
make[1]: *** [/home/leandro/Work/RIOT/Makefile.include:732: cleanterm] Floating point exception (core dumped)

Traceback (most recent call last):
  File "/home/leandro/Work/RIOT/examples/micropython/tests/01-run.py", line 84, in <module>
    sys.exit(run(testfunc))
  File "/home/leandro/Work/RIOT/dist/pythonlibs/testrunner/__init__.py", line 30, in run
    testfunc(child)
  File "/home/leandro/Work/RIOT/examples/micropython/tests/01-run.py", line 77, in testfunc
    duration = get_time() - before
  File "/home/leandro/Work/RIOT/examples/micropython/tests/01-run.py", line 11, in get_time
    res = int(child.readline().rstrip())
ValueError: invalid literal for int() with base 10: 'make[1]: *** [/home/leandro/Work/RIOT/Makefile.include:732: cleanterm] Floating point exception (core dumped)'
make: *** [/home/leandro/Work/RIOT/makefiles/tests/tests.inc.mk:22: test] Error 1
```

#### Versions
```

Operating System Environment
----------------------------
         Operating System: "Manjaro Linux" 
                   Kernel: Linux 5.10.7-3-MANJARO x86_64 unknown
             System shell: GNU bash, version 5.1.0(1)-release (x86_64-pc-linux-gnu)
             make's shell: GNU bash, version 5.1.0(1)-release (x86_64-pc-linux-gnu)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (GCC) 10.2.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (Arch Repository) 10.2.0
                  avr-gcc: missing
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 11.0.1

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.3.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: missing (missing)

Installed development tools
---------------------------
                   ccache: ccache version 4.1
                    cmake: cmake version 3.19.3
                 cppcheck: missing
                  doxygen: 1.9.1
                      git: git version 2.30.0
                     make: GNU Make 4.3
                  openocd: Open On-Chip Debugger 0.10.0
                   python: Python 3.9.1
                  python2: missing
                  python3: Python 3.9.1
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing
```

bergzand, 2021-01-27:
I'm able to reproduce this locally with the test.

My versions

```
Operating System Environment
----------------------------
         Operating System: Gentoo 
                   Kernel: Linux 5.7.14-gentoo x86_64 Intel(R) Core(TM) i7-10510U CPU @ 1.80GHz
             System shell: GNU bash, version 5.0.18(1)-release (x86_64-pc-linux-gnu)
             make's shell: GNU bash, version 5.0.18(1)-release (x86_64-pc-linux-gnu)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Gentoo 9.3.0-r2 p4) 9.3.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
                  avr-gcc: avr-gcc (Gentoo 9.3.0-r2 p4) 9.3.0
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 11.0.0

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.1.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: ccache version 4.1
                    cmake: cmake version 3.17.4
                 cppcheck: Cppcheck 1.88
                  doxygen: 1.8.17
                      git: git version 2.26.2
                     make: GNU Make 4.3
                  openocd: Open On-Chip Debugger 0.11.0-rc1+dev-g9c4787378-dirty (2021-01-07-16:40)
                   python: Python 3.8.6
                  python2: Python 2.7.18
                  python3: Python 3.8.6
                   flake8: 3.8.4 (mccabe: 0.6.1, pycodestyle: 2.6.0, pyflakes: 2.2.0) CPython 3.8.6 on
               coccinelle: missing
```

jia200x, 2021-01-29:
when I ran this test without the python script it seems to pass :/

There might be a race condition.

What I've noticed is that the timer seems to be fired way before than expected:
```
>>> t.set(500000)
t.set(500000)
>>> while a==0: pass
while a==0: pass
... 

/* It gets here almost immediately... */

>>> utime.time()
utime.time()
```

leandrolanzieri, 2021-02-01:
Hmm got another one. While running `tests/thread_float`:
```
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2021.04-devel-383-ga70ae-cpu/kinetis-float)
THREADS CREATED

THREAD 3 start
THREAD 4 start
THREAD 5 start
T(3): 141.466812
make: *** [/home/leandro/Work/RIOT/tests/thread_float/../../Makefile.include:725: term] Floating point exception (core dumped)
```

miri64, 2022-09-27:
Could this be related to #495?

leandrolanzieri, 2022-09-27:
> Could this be related to #495?

https://github.com/RIOT-OS/RIOT/issues/495#issuecomment-799201777 :)

miri64, 2022-09-27:
Sorry :sweat_smile:. But now at least, there is a cross reference here as well :grin:

maribu, 2023-01-05:
duplicate of https://github.com/RIOT-OS/RIOT/issues/495

closed 2026-09-01 by basilfx

basilfx, 2026-09-01:
The new MicroPython port now handles exceptions and interrupts on the interpreter thread, which "solves" this issue. In all the testing I did, I have not been able to reproduce this issue, but let's reopen if that is still the case.