RIOT-OS/RIOT #15878: examples/ccn-lite: floating point exception while testing on native

bergzand, 2021-01-28:
#### Description

While running the steps from the readme for the [examples/ccn-lite-relay](https://github.com/RIOT-OS/RIOT/tree/master/examples/ccn-lite-relay) application on native, a floating point exception is raised. I tried building in Docker and I get the same issue. Current master pointing to 8e47621e7ce1c5eb6a6832ea2ab885c8f1df5d7a.

#### Steps to reproduce the issue

Run the ccn-lite-relay example and follow the steps from the example.

#### Expected results

The steps from the [readme](https://github.com/RIOT-OS/RIOT/tree/master/examples/ccn-lite-relay#example-setup) work and it is possible to follow the example steps from the readme.

#### Actual results

```
koen@morgen ~/dev/RIOT-review $ make -C examples/ccn-lite-relay/ term
make: Entering directory '/home/koen/dev/RIOT-review/examples/ccn-lite-relay'
/home/koen/dev/RIOT-review/examples/ccn-lite-relay/bin/native/ccn-lite-relay.elf tap0 
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2021.04-devel-270-g8e4762-HEAD)
Basic CCN-Lite example
> ccnl_cs /riot/peter/schmerzl Hello World! Hello RIOT!
ccnl_cs /riot/peter/schmerzl Hello World! Hello RIOT!
make: *** [/home/koen/dev/RIOT-review/examples/ccn-lite-relay/../../Makefile.include:731: term] Floating point exception (core dumped)
make: Leaving directory '/home/koen/dev/RIOT-review/examples/ccn-lite-relay'
```

#### Versions

```
Operating System Environment
----------------------------
         Operating System: Gentoo 
                   Kernel: Linux 5.7.14-gentoo x86_64 Intel(R) Core(TM) i7-10510U CPU @ 1.80GHz
             System shell: GNU bash, version 5.0.18(1)-release (x86_64-pc-linux-gnu)
             make's shell: GNU bash, version 5.0.18(1)-release (x86_64-pc-linux-gnu)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Gentoo 9.3.0-r2 p4) 9.3.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
                  avr-gcc: avr-gcc (Gentoo 9.3.0-r2 p4) 9.3.0
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 11.0.0

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.1.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: ccache version 4.1
                    cmake: cmake version 3.17.4
                 cppcheck: Cppcheck 1.88
                  doxygen: 1.8.17
                      git: git version 2.26.2
                     make: GNU Make 4.3
                  openocd: Open On-Chip Debugger 0.11.0-rc1+dev-g9c4787378-dirty (2021-01-07-16:40)
                   python: Python 3.8.6
                  python2: Python 2.7.18
                  python3: Python 3.8.6
                   flake8: 3.8.4 (mccabe: 0.6.1, pycodestyle: 2.6.0, pyflakes: 2.2.0) CPython 3.8.6 on
               coccinelle: missing
```

PeterKietzmann, 2021-02-02:
Probably related to #15870. 

Does not fail for me:
```
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2021.01-devel-2346-g837b5)
Basic CCN-Lite example
> ccnl_cs /riot/peter/schmerzl Hello World! Hello RIOT!
ccnl_cs /riot/peter/schmerzl Hello World! Hello RIOT!
> ccnl_cs
ccnl_cs
CS[0]: /riot/peter/schmerzl [-1]: Hello World! Hello RIOT!
```

**Versions**
```
Operating System Environment
----------------------------
         Operating System: "Ubuntu" "20.04.2 LTS (Focal Fossa)"
                   Kernel: Linux 5.4.0-65-generic x86_64 x86_64
             System shell: /bin/dash (probably dash)
             make's shell: /bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                  avr-gcc: missing
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 10.0.0-4ubuntu1 

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.0.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: missing (missing)

Installed development tools
---------------------------
                   ccache: ccache version 3.7.7
                    cmake: cmake version 3.16.3
                 cppcheck: Cppcheck 1.90
                  doxygen: 1.8.17
                      git: git version 2.25.1
                     make: GNU Make 4.2.1
                  openocd: Open On-Chip Debugger 0.10.0
                   python: Python 2.7.18
                  python2: Python 2.7.18
                  python3: Python 3.8.5
                   flake8: 3.8.4 (mccabe: 0.6.1, pycodestyle: 2.6.0, pyflakes: 2.2.0) CPython 3.8.5 on
               coccinelle: missing
```

maribu, 2023-01-05:
duplicate of https://github.com/RIOT-OS/RIOT/issues/495

closed 2023-01-05 by maribu