RIOT-OS/RIOT #15888: tests/test_tools: test fails while testing on samr21-xpro/iotlab-m3

jia200x, 2021-01-29:
#### Description

`tests/test_tools` fails when ran on samr21-xpro or iotlab-m3. This is the output:

```
shellpong
true this should not be echoed
shellping
shellpong
toupper lowercase
LOWERCASE

Traceback (most recent call last):
  File "/home/jialamos/Development/RIOT/tests/test_tools/tests/01-run.py", line 84, in <module>
    sys.exit(run(testfunc))
  File "/home/jialamos/Development/RIOT/dist/pythonlibs/testrunner/__init__.py", line 30, in run
    testfunc(child)
  File "/home/jialamos/Development/RIOT/tests/test_tools/tests/01-run.py", line 77, in testfunc
    _test_clean_output(child)
  File "/home/jialamos/Development/RIOT/tests/test_tools/tests/01-run.py", line 56, in _test_clean_output
    assert retline.strip() == 'LOWERCASE'
AssertionError
make: *** [/home/jialamos/Development/RIOT/makefiles/tests/tests.inc.mk:22: test] Error 1
```

The task expects "LOWERCASE" and this is exactly what's read from the node, but for some reason the text contains some ANSI characters (actually [bracket paste mode](https://cirw.in/blog/bracketed-paste)). Therefore the line _contains_ "LOWERCASE" but doesn't match the exact result.

This issue was reproduced only on Arch Linux machines, Ubuntu machines don't seem to have any issues. I also tried different versions of python and still get the same issue.

Master points to 13b97be08cc90c8a11bbee8d67ddc15dbf91073e

#### Steps to reproduce the issue

Run e.g `BOARD=samr21-xpro make -C tests/test_tools flash test`.

#### Expected results

The test should pass
#### Actual results

As described above, the test fails.
#### Versions

```
Operating System Environment
----------------------------
         Operating System: "Arch Linux" 
                   Kernel: Linux 5.10.9-arch1-1 x86_64 unknown
             System shell: GNU bash, version 5.1.4(1)-release (x86_64-pc-linux-gnu)
             make's shell: GNU bash, version 5.1.4(1)-release (x86_64-pc-linux-gnu)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (GCC) 10.2.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
                  avr-gcc: avr-gcc (GCC) 10.2.0
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 11.0.1

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "2.5.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: missing
                    cmake: cmake version 3.19.3
                 cppcheck: missing
                  doxygen: 1.9.1
                      git: git version 2.30.0
                     make: GNU Make 4.3
                  openocd: Open On-Chip Debugger 0.10.0+dev-00226-g1c2e3d41d (2018-01-02-18:28)
                   python: Python 3.9.1
                  python2: Python 2.7.18
                  python3: Python 3.9.1
                   flake8: 3.8.4 (mccabe: 0.6.1, pycodestyle: 2.6.0, pyflakes: 2.2.0) CPython 3.9.1 on
               coccinelle: missing
```

jia200x, 2021-01-29:
note that this test passes if `RIOT_TERMINAL=socat`

OlegHahm, 2021-02-14:
Hm, I wasn't able to reproduce the bug here:
```
Operating System Environment
----------------------------
         Operating System: "Arch Linux" 
                   Kernel: Linux 5.9.1-arch1-1 x86_64 unknown
             System shell: GNU bash, version 5.0.18(1)-release (x86_64-pc-linux-gnu)
             make's shell: GNU bash, version 5.0.18(1)-release (x86_64-pc-linux-gnu)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (GCC) 10.2.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (Arch Repository) 10.2.0
                  avr-gcc: avr-gcc (GCC) 10.2.0
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: msp430-elf-gcc (GCC) 6.2.0
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 10.0.1 

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.3.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: /usr/lib/gcc/msp430-elf/6.2.0/cc1: error while loading shared libraries: libmpfr.so.4: cannot open shared object file: No such file or directory
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: "2.2.0"
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: ccache version 3.7.11
                    cmake: cmake version 3.18.4
                 cppcheck: Cppcheck 2.1
                  doxygen: 1.8.20
                      git: git version 2.29.0
                     make: GNU Make 4.3
                  openocd: Open On-Chip Debugger 0.10.0
                   python: Python 3.8.6
                  python2: Python 2.7.18
                  python3: Python 3.8.6
                   flake8: error: /bin/python3: No module named flake8
               coccinelle: missing
```

OlegHahm, 2021-02-14:
Ah, after a system update, I get the same error. I assume the Python version is responsible.

maribu, 2023-01-05:
Ping?