RIOT-OS/RIOT #15894: improve:mtd spi nor

kex2017, 2021-02-01:
#### Description
mtd:https://github.com/RIOT-OS/RIOT/blob/master/drivers/mtd_spi_nor/mtd_spi_nor.c#L396

 while (res < 0 || retries < MTD_POWER_UP_WAIT_FOR_ID);
should be
 while (res < 0 && retries < MTD_POWER_UP_WAIT_FOR_ID);

jia200x, 2021-02-01:
hi @kex2017 

could you please elaborate more about this issue? E.g why is this an issue, what's the expected result, what happens now, etc.

kex2017, 2021-02-02:
> hi @kex2017
> 
> could you please elaborate more about this issue? E.g why is this an issue, what's the expected result, what happens now, etc.

The maximum number of times the ID is always retried regardless of whether it is read successfully or not

kex2017, 2021-02-02:
> hi @kex2017
> 
> could you please elaborate more about this issue? E.g why is this an issue, what's the expected result, what happens now, etc.

I expect to end the retry loop after reading the correct return value

benpicco, 2021-02-02:
Indeed, good catch!
Do you want to submit a PR?

madokapeng, 2021-02-16:
Looks like @benpicco has a PR to fix the issue? if not, I can work on this.

PR #15987 (2021-02-22): drivers/mtd_spi_nor: fix poll on init

The while condition for polling the chip ID is broken, it will always poll for the max amount of retries.

Change the condition from `or` to `and` to fix the logic.

### Contribution description

### Testing procedure

A spi nor device is still detected by `tests/mtd_raw`

```
2021-02-11 22:52:00,752 # test 0
2021-02-11 22:52:00,752 # [START]
2021-02-11 22:52:00,980 # [SUCCESS]

2021-02-11 22:52:11,040 # info 0
2021-02-11 22:52:11,040 # sectors: 128
2021-02-11 22:52:11,041 # pages per sector: 16
2021-02-11 22:52:11,041 # page size: 256
2021-02-11 22:52:11 [... cut]
commit c1348c80 (2021-02-22): drivers/mtd_spi_nor: fix poll on init

The while condition for polling the chip ID is broken, it will always
poll for the max ammount of retries.

Change the condition from `or` to `and` to fix the logic.

fixes #15894
closed 2021-02-22 by benpicco