RIOT-OS/RIOT #15917: sys/riotboot/flashwrite: unaligned write when skipping  `RIOTBOOT_MAGIC` on stm32wb

fjmolinas, 2021-02-03:
#### Description

While testing  #15914 I realized that `riotboot/flashwrite` has been broken on stm32wb for a while. It seems to be caused by the following sections:

https://github.com/RIOT-OS/RIOT/blob/3b2a55a923fa3bbf215bc90ac6d882f48bf5b8a0/sys/riotboot/flashwrite.c#L141-L155

The offset is initially set to 4 and when the required alignment is 8. Weirdly enough I only get this error with stm32wb and not with stm32l4.

#### Steps to reproduce the issue

Any application using riotboot/flashwrite so `examples/suit_udpate` or `tests/riotboot_flashwrite`. https://github.com/RIOT-OS/RIOT/pull/15559 can be used to reproduce in any case:

- `sudo ./dist/tools/ethos/setup_network.sh riot0 2001:db8::/64`
- `make -C <examples/suit_update or tests/riotboot_flashwrite> flash test-with-config`

#### Expected results

Test passes

#### Actual results

```
          inet6 addr: fe80::2  scope: link  VAL
          inet6 group: ff02::2
          inet6 group: ff02::1
          inet6 group: ff02::1:fff0:17c1
          inet6 group: ff02::1:ff00:2
          
> _flashwrite_handler(): received data: offset=0 len=64 blockwise=1 more=1
_flashwrite_handler(): init len=64
riotboot_flashwrite: initializing update to target slot 1
riotboot_flashwrite: processing bytes 4-63
cpu/stm32/periph/flashpage.c:208 => 0x80017f5
*** RIOT kernel panic:
FAILED ASSERTION.

*** halted.

```

maribu, 2023-05-17:
@benpicco Is this still an issue?

benpicco, 2023-05-22:
I don't have the hardware to check this

aabadie, 2023-05-22:
Could it be related #19618 ?

maribu, 2023-05-22:
> I don't have the hardware to check this

I think I have an stm32wb somewhere. I'll give it a try when I find it, both with master and https://github.com/RIOT-OS/RIOT/pull/19618

crasbe, 2024-07-03:
tl;dr: This issue seems to be fixed. The suit_update example failed once for me but that could've had a lot of reasons. I retried it multiple times and it never failed.

IMO this issue can be closed @maribu.
The original author doesn't seem to be active anymore :(

<hr/>

The "riot_flashboot" test gives the following output:

```
chris@ThinkPias:~/flashdev-riot/RIOT/tests/riotboot_flashwrite$ BOARD=p-nucleo-wb55 make flash test-with-config
Building application "tests_riotboot_flashwrite" for "p-nucleo-wb55" with CPU "stm32".

"make" -C /home/chris/flashdev-riot/RIOT/pkg/cmsis/ 
"make" -C /home/chris/flashdev-riot/RIOT/boards/common/init
"make" -C /home/chris/flashdev-riot/RIOT/boards/p-nucleo-wb55
...
"make" -C /home/chris/flashdev-riot/RIOT/sys/stdio_null
compiling /home/chris/flashdev-riot/RIOT/dist/tools/riotboot_gen_hdr/bin/genhdr...
make: Nothing to be done for 'all'.
creating /home/chris/flashdev-riot/RIOT/tests/riotboot_flashwrite/bin/p-nucleo-wb55/riotboot_files/slot0.1720049003.bin...
   text	   data	    bss	    dec	    hex	filename
  68056	    176	  13032	  81264	  13d70	/home/chris/flashdev-riot/RIOT/tests/riotboot_flashwrite/bin/p-nucleo-wb55/tests_riotboot_flashwrite.elf
/home/chris/flashdev-riot/RIOT/dist/tools/openocd/openocd.sh flash /home/chris/flashdev-riot/RIOT/tests/riotboot_flashwrite/bin/p-nucleo-wb55/riotboot_files/slot0-extended.bin
### Flashing Target ###
Binfile detected, adding ROM base address: 0x08000000
Flashing with IMAGE_OFFSET: 0x08000000
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
hla_swd
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_assert_srst

Info : clock speed 500 kHz
Info : STLINK V2J40M27 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.236292
Info : stm32wbx.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32wbx.cpu on 0
Info : Listening on port 32821 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32wbx.cpu       hla_target little stm32wbx.cpu       reset

Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800038c msp: 0x20000200
Info : device idcode = 0x20016495 (STM32WB5x - Rev 2.1 : 0x2001)
Info : flash size = 1024kbytes
Info : flash mode : single-bank
Warn : Adding extra erase range, 0x08066200 .. 0x08066fff
auto erase enabled
wrote 418304 bytes from file /home/chris/flashdev-riot/RIOT/tests/riotboot_flashwrite/bin/p-nucleo-wb55/riotboot_files/slot0-extended.bin in 15.787079s (25.876 KiB/s)

verified 418304 bytes in 12.088158s (33.793 KiB/s)

Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
shutdown command invoked
Done flashing
make -C benchmark_udp
make[2]: Für das Ziel „all“ ist nichts zu tun.
make -C dose
make[2]: Für das Ziel „all“ ist nichts zu tun.
make -C ethos
make[2]: Für das Ziel „all“ ist nichts zu tun.
make -C uhcpd
make[2]: Für das Ziel „all“ ist nichts zu tun.
[... 872 lines cut ...]
ifconfig
Iface  5  HWaddr: E6:A6:36:71:18:E9 
          L2-PDU:1500  MTU:1500  HL:64  Source address length: 6
          Link type: wired
          inet6 addr: fe80::e4a6:36ff:fe71:18e9  scope: link  VAL
pinging node...
PING fe80::e4a6:36ff:fe71:18e9%riot0(fe80::e4a6:36ff:fe71:18e9%riot0) 56 data bytes

--- fe80::e4a6:36ff:fe71:18e9%riot0 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 40.572/40.572/40.572/0.000 ms
pinging node succeeded.

          inet6 addr: fe80::2  scope: link  VAL
          inet6 group: ff02::1
          inet6 group: ff02::1:ff71:18e9
          inet6 group: ff02::1:ff00:2
          
> suit seq_no

> 
> suit seq_no
seq_no: 0x6685dbcd
> TEST PASSED
```

maribu, 2024-07-04:
@crasbe: Thx for looking into this ❤️

closed 2024-07-04 by maribu