RIOT-OS/RIOT #15927: uri_parser: out-of-bounds read

nmeum, 2021-02-04:
#### Description

I did some testing of the `uri_parser` module as provided in `sys/uri_parser`. I believe I discovered an edge-case where the parser performs an out-of-bounds read of the provided buffer. The code causing this is:

https://github.com/RIOT-OS/RIOT/blob/9eb6a38a5adf55335f14e582834dad63610889d7/sys/uri_parser/uri_parser.c#L76-L80

which advances `result->host` without a bounds check and even if `userinfo_len` is zero `result->host` is still advanced by one byte.

#### Steps to reproduce the issue

Application code:

```C
#include <stdio.h>
#include <stdlib.h>
#include <stdint.h>
#include <uri_parser.h>

#define BUFFER_SIZE 5

static char buf[] = { 65, 58, 47, 47, 64 };

int main(void)
{
    size_t len = sizeof(buf);
    uri_parser_result_t result;

    uri_parser_process(&result, buf, len);
    return 0;
}
```

Minimal `Makefile`:

```Makefile
APPLICATION = uri
DEVELHELP = 1

BOARD = native

USEMODULE += uri_parser

RIOTBASE ?= $(CURDIR)/../..
include $(RIOTBASE)/Makefile.include
```

Afterwards, compile as:

	$ make -C examples/uri all-asan

And run the application using:

	$ make -C examples/uri term

#### Expected results

The application shouldn't crash.

#### Actual results

```
main(): This is RIOT! (Version: 2021.04-devel-440-g4cc04)
=================================================================
==24259==ERROR: AddressSanitizer: global-buffer-overflow on address 0x565e6105 at pc 0x565dce2b bp 0x565eb6f8 sp 0x565eb6ec
READ of size 1 at 0x565e6105 thread T0
    #0 0x565dce2a in _consume_authority /root/RIOT/sys/uri_parser/uri_parser.c:130
    #1 0x565dd5ed in _parse_absolute /root/RIOT/sys/uri_parser/uri_parser.c:222
    #2 0x565ddabc in uri_parser_process /root/RIOT/sys/uri_parser/uri_parser.c:280
    #3 0x565d3bcb in main /root/RIOT/examples/uri/main.c:15
    #4 0x565d40ed in main_trampoline /root/RIOT/core/init.c:58
    #5 0xf766353a in makecontext (/lib/i386-linux-gnu/libc.so.6+0x4153a)

0x565e6105 is located 0 bytes to the right of global variable 'buf' defined in '/root/RIOT/examples/uri/main.c:8:13' (0x565e6100) of size 5
SUMMARY: AddressSanitizer: global-buffer-overflow /root/RIOT/sys/uri_parser/uri_parser.c:130 in _consume_authority
Shadow bytes around the buggy address:
  0x2acbcbd0: f9 f9 f9 f9 00 00 00 00 00 00 00 00 00 00 00 00
  0x2acbcbe0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x2acbcbf0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x2acbcc00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x2acbcc10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x2acbcc20:[05]f9 f9 f9 f9 f9 f9 f9 00 00 00 00 f9 f9 f9 f9
  0x2acbcc30: f9 f9 f9 f9 f9 f9 f9 f9 00 00 00 00 f9 f9 f9 f9
  0x2acbcc40: 00 00 00 00 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9
  0x2acbcc50: f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9
  0x2acbcc60: f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 00 00 00 00
  0x2acbcc70: f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==24259==ABORTING
make: *** [/root/RIOT/examples/uri/../../Makefile.include:725: term] Error 1
make: Leaving directory '/root/RIOT/examples/uri'
```

#### Versions

I don't think this is needed, if you need more information let me know.

cgundogan, 2021-02-04:
thanks for this detailed report (: I was able to reproduce on native and while going through the code I also adjusted the unit tests for the case you presented "A://@".  That's correct, right?

nmeum, 2021-02-04:
> That's correct, right?

Yes, that's correct. I also found a separate issue while testing my fix for this issue. I documented this separately in #15930. Hope that's ok. Thanks for your quick fix!

PR #15929 (2021-02-05): uri_parser: fix out-of-bounds and additional enhancements
### Contribution description

* Fixes the out of bounds for the case where the host part is empty, but a userinfo exists.
* Additionally allow empty host parts, even if userinfo and port exist (look at the ABNF in https://tools.ietf.org/html/rfc3986#appendix-A)
* Further, do not attempt to parse path if end-of-uri is reached.

### Testing procedure
run the unittests and the example in #15927

### Issues/PRs references
fixes #15927
closed 2021-02-05 by miri64