RIOT-OS/RIOT #15954: bmx280: Unable to compile due to spi3w_en parameter

memiks, 2021-02-09:
When I included bmx module in my project:
```
#include "bmx280.h"
#include <bmx280_params.h>
```

 I received :
`./domotique/RIOT/drivers/bmx280/include/bmx280_params.h:94:1: error: missing initializer for member 'bmx280_params_t::spi3w_en' [-Werror=missing-field-initializers]`

after a search on all RIOT I found the declaration but this 'spi3w_en' is never used... So I proposed to remove it ?
https://github.com/RIOT-OS/RIOT/blob/82ab75c2b4cf4deaa2bd630a3af63f1299c3fd48/drivers/include/bmx280.h#L198

after removing this line I can compile without issue.

Best regards,
Memiks.

benpicco, 2021-02-11:
Indeed, might be a leftover - do you want to submit a patch?

memiks, 2021-02-15:
Hello @benpicco ,

Of course, I've just pushed a PR.

Best regards,
Memiks.

PR #16016 (2021-02-16): remove unused parameter spi3w_en to solve issue #15954
Hi RIOT-OS Team,

### Contribution description

I removed spi3w_en parameter in drivers/include/bmx280.h since it is never used.

### Testing procedure

compile bmx280 tests by including bmx280.h header.

### Issues/PRs references

Fixes #15954 .

Have a nice day,
Memiks.
closed 2021-02-16 by benpicco