RIOT-OS/RIOT #16002: Provide compile_commands.json

chrysn, 2021-02-13:
There is a convention of how CFLAGS etc are transported to build-related tools in [compile_commands.json](https://clang.llvm.org/docs/JSONCompilationDatabase.html); AFAICT they always describe a single file, so the process would need to produce a compile_commands.json file that describes how any given input file would be processed (although it may be that it's actually immaterial to the rest of the commands which file that is).

Producing these from the build system would be beneficial for two reasons:

* IDEs can use them to provide suitable development support (@jnohlgard and @bergzand probably know more about that, judging from recent IRC discussions)
* In Rust support (#9799) I currently build my own compile_commands.json using data snooped from the environment inside the Makefiles. Having a possibility to get them out without being part of the build system would have the advantage that I could get the CFLAGS that RIOT would use for LLVM out of the build system, immaterial of which toolchain it's currently being run for (which may be GCC because that may produce better object files for the C code in a particular situation) -- thus solving [riot-sys/#3](https://gitlab.com/etonomy/riot-sys/-/issues/3).

miri64, 2021-02-14:
Not sure if I understand this correctly, but could this also maybe solve https://github.com/RIOT-OS/RIOT/issues/14669?

chrysn, 2021-02-14:
I don't see any direct relation (but asked for clarification there).

The compile_commands.json are very specific about the selected
configuration; getting them for the alternative TOOLCHAIN=llvm
configuration variant is a specific thing that'd help Rust integration,
and not a feature of the JSON format. (It's just that once there's a way
to extract the specific config, it follows that there's also a way to
extract the specific-config-with-toolchain-overridden).

pokgak, 2021-02-17:
I have tried using [Bear](https://github.com/rizsotto/Bear) to generate the compile command when running make. Though it produces output that is specific to the application that is being compiled only and not the whole RIOT codebase.

kaspar030, 2021-02-17:
> I have tried using [Bear](https://github.com/rizsotto/Bear) to generate the compile command when running make. Though it produces output that is specific to the application that is being compiled only and not the whole RIOT codebase.

I'm using bear, too, to get linting in vim with ale.

aabadie, 2021-05-20:
#16129 was merged, so I think this can be closed.

closed 2021-05-20 by aabadie