RIOT-OS/RIOT #16022: `make term` no longer works with JLinkExe v6.94

bartcerneels, 2021-02-16:
While starting to use RTT as the stdio for an STM32L496VG based board `make term` was not working. It still worked with JLink v6.92. 

#### Steps to reproduce the issue
I've configured a board to use JLink RTT in Makefile.include:
```
JLINK_DEVICE = STM32L496VG

RIOT_TERMINAL = jlink
include $(RIOTMAKE)/tools/serial.inc.mk

include $(RIOTMAKE)/tools/jlink.inc.mk
```
Then run `make term`

#### Expected results
```
$ JLINK=/opt/SEGGER/JLink_Linux_V692_x86_64/JLinkExe make term
/home/bart/Bond/Code/iot_firmware/RIOT/dist/tools/jlink/jlink.sh term-rtt 
### Starting RTT terminal ###
Twisted not available, please install it if you want to use pyterm's JSON capabilities
2021-02-16 11:36:56,254 # Host name for TCP connection is missing, defaulting to "localhost"
2021-02-16 11:36:56,254 # Connect to localhost:19021
Welcome to pyterm!
Type '/exit' to exit.
2021-02-16 11:36:57,258 # SEGGER J-Link V6.92 - Real time terminal output
2021-02-16 11:36:57,258 # SEGGER J-Link Ultra V4.0, SN=504504798
2021-02-16 11:36:57,258 # Process: JLinkExe
2021-02-16 11:36:57,259 # main(): This is RIOT! (Version: 2021.01-devel-2-g46a9e-HEAD)
```

#### Actual results
```
$ JLINK=/opt/SEGGER/JLink_V694d/JLinkExe make term
/home/bart/Bond/Code/iot_firmware/RIOT/dist/tools/jlink/jlink.sh term-rtt 
### Starting RTT terminal ###
Twisted not available, please install it if you want to use pyterm's JSON capabilities
2021-02-16 11:09:05,532 # Host name for TCP connection is missing, defaulting to "localhost"
2021-02-16 11:09:05,532 # Connect to localhost:19021
2021-02-16 11:09:05,534 # Something went wrong connecting to localhost:19021
Killing 68643
/home/bart/Bond/Code/iot_firmware/RIOT/dist/tools/jlink/jlink.sh: 281: kill: No such process
```

#### Versions
Our clone of RIOT is not based on the latest and has 2 fixes related to UART and building on windows. These should not affect the jlink script. I've checked out the current master (ce0d58938447af94da6def4163f49426318e32f1) and it has the same problem. 

```
Operating System Environment
----------------------------
         Operating System: "KDE neon" "5.20"
                   Kernel: Linux 5.4.0-59-generic x86_64 x86_64
             System shell: /usr/bin/dash (probably dash)
             make's shell: /usr/bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)
                  avr-gcc: missing
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: missing

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.3.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: missing (missing)

Installed development tools
---------------------------
                   ccache: missing
                    cmake: cmake version 3.16.3
                 cppcheck: missing
                  doxygen: 1.8.17
                      git: git version 2.25.1
                     make: GNU Make 4.2.1
                  openocd: Open On-Chip Debugger 0.10.0+dev-g4e85dcf4 (2020-11-24-10:58)
                   python: Python 3.8.5
                  python2: Python 2.7.18
                  python3: Python 3.8.5
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing
```

haukepetersen, 2021-05-10:
I can confirm this issue: just updated from JLink V6.80d to V7.20a, and for me `make term` using jlink and `stdio_rtt` also stopped to work, same behavior as stated above.

I am unsure what is causing this, but I am hopeful to find the cause of this soon, as this is annoying me a bit :-)

@bartcerneels I do have a temporary workaround: 
- run `JLinkExe` stand-alone in a terminal window
- connect manually to your board, in my case (using nrf52) I do this by typing `connect` -> `nrf52` -> `s` -> `2000`
- in a second terminal, invoke `make term` as usual, Now it should succeed.

haukepetersen, 2021-05-10:
I have been starring at this all evening now and can't figure this out. When running JLinkExe manually as background job on my Linux host, as in
```
JLinkExe -nogui 1 -exitonerror 1 -device 'nrf52' -speed '1000' -if 'SWD' -jtagconf -1,-1 -commandfile '/home/hauke/dev/riot/RIOT/dist/tools/jlink/term.seg' >/dev/null 2>&1 &
```
it will run jlink as expected.
But when passing that exact same command as a string argument to my dash shell
```
sh -c "JLinkExe -nogui 1 -exitonerror 1 -device 'nrf52' -speed '1000' -if 'SWD' -jtagconf -1,-1 -commandfile '/home/hauke/dev/riot/RIOT/dist/tools/jlink/term.seg' >/dev/null 2>&1 &"
```
jlink will not continue to run in the background, but will be closed instantly. I was able to see this behavior for any jlink version of V6.94 and higher, while it works perfectly fine for V6.92 and below. As far as I can tell, there is no hint in the jlink release nodes that would point to any changes in jlink that could be causing this... So for now I really can't tell, somebody else with an enlightenment?