RIOT-OS/RIOT #16043: Riotboot DFU does not compile

ant9000, 2021-02-18:
#### Description

I'm trying to flash the DFU bootloader on a SAMR34-XPRO board  - compilation fails.

#### Steps to reproduce the issue

Simply trying

```BOARD=samr34-xpro make -C bootloaders/riotboot_dfu/ all flash```

produces the following error:

```
riot/core/include/assert.h:116:54: error: expected ',' before ')' token
 #define static_assert(...) _Static_assert(__VA_ARGS__)

riot/sys/riotboot/flashwrite.c:50:5: note: in expansion of macro 'static_assert'
     static_assert(!(FLASHPAGE_SIZE % RIOTBOOT_FLASHPAGE_BUFFER_SIZE));
     ^~~~~~~~~~~~~
```

Looks like static_assert macro is not used correctly in riotboot/flashwrite.c

fjmolinas, 2021-02-18:
I'm unable to reproduce this issue, can you print your local toolchain `make print-versions`, when I build in docker everything is fine for me:

```
BUILD_IN_DOCKER=1 BOARD=samr34-xpro make -C bootloaders/riotboot_dfu/ all -j3
Launching build container using image "riot/riotbuild:latest".
docker run --rm --tty --user $(id -u) -v '/usr/share/zoneinfo/Europe/Paris:/etc/localtime:ro' -v '/home/francisco/workspace/RIOT:/data/riotbuild/riotbase:delegated' -e 'RIOTBASE=/data/riotbuild/riotbase' -e 'CCACHE_BASEDIR=/data/riotbuild/riotbase' -e 'BUILD_DIR=/data/riotbuild/riotbase/build' -e 'RIOTPROJECT=/data/riotbuild/riotbase' -e 'RIOTCPU=/data/riotbuild/riotbase/cpu' -e 'RIOTBOARD=/data/riotbuild/riotbase/boards' -e 'RIOTMAKE=/data/riotbuild/riotbase/makefiles'      -e 'BOARD=samr34-xpro'  -w '/data/riotbuild/riotbase/bootloaders/riotboot_dfu/' 'riot/riotbuild:latest' make     -j3 all
Building application "riotboot_dfu" for "samr34-xpro" with MCU "saml21".

"make" -C /data/riotbuild/riotbase/boards/samr34-xpro
"make" -C /data/riotbuild/riotbase/core
"make" -C /data/riotbuild/riotbase/cpu/saml21
"make" -C /data/riotbuild/riotbase/cpu/cortexm_common
"make" -C /data/riotbuild/riotbase/drivers
"make" -C /data/riotbuild/riotbase/drivers/periph_common
"make" -C /data/riotbuild/riotbase/cpu/cortexm_common/periph
"make" -C /data/riotbuild/riotbase/cpu/sam0_common
"make" -C /data/riotbuild/riotbase/cpu/sam0_common/periph
"make" -C /data/riotbuild/riotbase/cpu/saml21/periph
"make" -C /data/riotbuild/riotbase/sys
"make" -C /data/riotbuild/riotbase/sys/checksum
"make" -C /data/riotbuild/riotbase/sys/div
"make" -C /data/riotbuild/riotbase/sys/event
"make" -C /data/riotbuild/riotbase/sys/fmt
"make" -C /data/riotbuild/riotbase/sys/luid
"make" -C /data/riotbuild/riotbase/sys/malloc_thread_safe
"make" -C /data/riotbuild/riotbase/sys/newlib_syscalls_default
"make" -C /data/riotbuild/riotbase/sys/pm_layered
"make" -C /data/riotbuild/riotbase/sys/riotboot
"make" -C /data/riotbuild/riotbase/sys/stdio_uart
"make" -C /data/riotbuild/riotbase/sys/usb/usbus
"make" -C /data/riotbuild/riotbase/sys/xtimer
"make" -C /data/riotbuild/riotbase/sys/usb/usbus/dfu
   text	   data	    bss	    dec	    hex	filename
  11492	     36	   3896	  15424	   3c40	/data/riotbuild/riotbase/bootloaders/riotboot_dfu/bin/samr34-xpro/riotboot_dfu.elf
```

ant9000, 2021-02-18:
Yes, sure - I'm on a fully updated Debian Stable:

```
Operating System Environment
----------------------------
         Operating System: "Debian GNU/Linux" "10 (buster)"
                   Kernel: Linux 4.19.0-14-amd64 x86_64 unknown
             System shell: /bin/dash (probably dash)
             make's shell: /bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Debian 8.3.0-6) 8.3.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                  avr-gcc: avr-gcc (GCC) 5.4.0
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: missing

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.1.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: ccache version 3.6
                    cmake: cmake version 3.13.4
                 cppcheck: Cppcheck 1.86
                  doxygen: 1.8.13
                      git: git version 2.20.1
                     make: GNU Make 4.2.1
                  openocd: Open On-Chip Debugger 0.10.0
                   python: Python 2.7.16
                  python2: Python 2.7.16
                  python3: Python 3.7.3
                   flake8: 3.6.0 (mccabe: 0.6.1, pycodestyle: 2.4.0, pyflakes: 2.0.0) CPython 3.7.3 on Linux
               coccinelle: spatch version 1.0.4 with Python support and with PCRE support
```

dylad, 2021-02-18:
That's weird, I mainly used SAML21-XPRO which is the same MCU while I was developing USB DFU and I never get such error.

ant9000, 2021-02-18:
Can confirm it is a toolchain issue.
I created a new Debian Buster VM with build-essential, unzip, git, gcc-arm-none-eabi - compilation ends with the very same error.

fjmolinas, 2021-02-18:
I was able to reproduce by switching the toolchain.

fjmolinas, 2021-02-19:
Fixed by #16044

closed 2021-02-19 by fjmolinas