RIOT-OS/RIOT #16054: pkg/tinydtls: auxiliary data API does not work for async sockets

janosbrodbeck, 2021-02-19:
This most likely also affects the (WIP) port of wolfSSL.

#### Description

While #14706 works fine for synchronous DTLS sockets and is easily extensible, the aux API does not work for asynchronous DTLS sockets at all. The major problem lies in the way how async sockets work for DTLS:

__aux_rx:__ When encrypted data is received on the underlying (also asynchronous) UDP socket, the internal `_udp_cb()` handles the incoming event. It hands the encrypted data over to tinydtls to decrypt it. After successful decrypt, __tinydtls calls__ the `_read()`-function of the tinydtls riot implementation. This function receives the buffer holding the decrypted data, buffer len and the concerned session from tinydtls and sets it in provided fields in the associated DTLS socket. Now it finally fires the DTLS event for the user application. If the user now receives the data via `sock_dtls_recv_aux()` it just returns the previously set data in the socket. The aux object remains untouched. 

__aux_tx:__ No direct call of `sock_udp_send_aux()`, data is handed over to tinydtls and after encryption it calls `_write()` in the tinydtls riot implementation to send the data. 

#### Expected results
Getting the requested aux fields

#### Actual results
Empty aux objects

---

The only fix that comes in my mind is similar how we "solved" #15755. Attach the aux objects to the socket and retrieve them after tinydtls has done it's job. For the aux_tx object this is easily doable without any potential problem afaik. But for the receiving aux object there would be no way to set custom `aux_flags` to determine which fields should be retrieved, since when the user event is fired the `sock_udp_recv()` has already been called. In this case we could only do all or nothing. 

Any ideas?

maribu, 2021-02-22:
> While #14706 works fine for synchronous DTLS sockets and is easily extensible, the aux API does not work for asynchronous DTLS sockets at all.

By asynchronous you mean using this API? https://doc.riot-os.org/group__net__sock__async__event.html

There is neither API nor implementation for retrieving auxiliary data via the async API (yet). So this is very much expected, isn't it?

janosbrodbeck, 2021-02-22:
> > While #14706 works fine for synchronous DTLS sockets and is easily extensible, the aux API does not work for asynchronous DTLS sockets at all.
> 
> By asynchronous you mean using this API? https://doc.riot-os.org/group__net__sock__async__event.html

Yes

> There is neither API nor implementation for retrieving auxiliary data via the async API (yet). So this is very much expected, isn't it?

Oh, good point. Missed that the aux API does not yet support the async API. For UDP sockets it looked like it should work.

So I think we can close this for now.

maribu, 2021-02-22:
Wait, on UDP it should and does work. I am even using this successfully in my PTP client. I will take a closer look when my kids are sleeping.

janosbrodbeck, 2021-02-22:
I mean UDP sockets using the async API.