RIOT-OS/RIOT #16072: What's the meaning of riot‘s tickless scheduler?

liyue75, 2021-02-23:
I find the following description in **sched.h**:
```
 * RIOT features a tickless, preemptive, priority based scheduler.
 * Context switches can occur either preemptively (i.e. on interrupts),
 * voluntarily, or when a blocking operation (like `msg_receive()`) is
 * executed.
 * Being tickless means it does not have a timer that fires
 * periodically in order to emulate concurrent execution by switching
 * threads continuously.
```
and the following:
```
 * A higher priority means that the scheduler will run this thread
 * whenever it becomes runnable instead of a thread with a lower
 * priority.
```
Aren't these two descriptions contradictory?
If there is no  scheduler runs periodically, how a thread with higher priority can be sched when there is no preemptively, voluntarily or blocking operation?

maribu, 2021-02-23:
> If there is no scheduler runs periodically, how a thread with higher priority can be sched when there is no preemptively, voluntarily or blocking operation?

Maybe it is important to understand what voluntarily giving up the CPU means: This is done by a call to `thread_yield()` and allows another thread of the same priority to be run. A thread of lower priority will not run until all threads with higher priority are blocked.

If a low priority thread is running, all high priority threads are blocked. Let's say we have the threads A (highest priority), B (medium priority) and C (low priority). Let's also assume that A is waiting for an event from hardware, B is blocking because it wants to lock a mutex and C is currently holding that very mutex. The very moment C releases the mutex, the call to `mutex_unlock()` runs the scheduler. As B is no longer blocked once the mutex is release, B would get CPU time.

If an IRQ comes to handle the hardware event A is waiting for, the interrupt service routine (ISR) will unblock A. This sets a flag. At the end of each ISR, this flag is checked. If it is set (so that a thread was unblocked) the scheduler is run at the end of the ISR.

As you see: The only thing that can unblock a high priority thread are IRQs or the running thread. But whenever this is done, the scheduler is also run. (Or in case of the IRQ slightly delayed at the end of the ISR.) Hence, a high priority thread will run right when it is unblocked (or in case of the ISR just a few CPU cycles later).

maribu, 2021-02-23:
Btw: We will have to update the description once optional round-robin scheduling is added.

@kfessel is going to add this to allow multiple VM threads to run. The issue with those is that they are not trusted to give up the CPU voluntarily. To solve that, VM threads get the lowest priority (so that native RIOT threads can always run) and within VM threads a round robin scheduler can fix the issue that one VM thread takes all the CPU time available to that priority level while the other VM threads starve.

liyue75, 2021-02-23:
@maribu 
I know, thank you.

closed 2021-02-23 by liyue75