RIOT-OS/RIOT #16115: How can I disconnect with the node and rasp3b+ in the IPv6-over-BLE example?

sada45, 2021-03-01:
Hi
  I have been working on the IPv6-over-BLE example [https://github.com/RIOT-OS/RIOT/blob/master/pkg/nimble/README.ipv6-over-ble.md](url). With the turtorial, I can connect the nrf52840dk with RaspberryPi's native bluetooth.
But here is the problem, How can I disconnect with the node? I have tried the command:
`echo "disconnect F6:A3:FA:05:96:BC 2" > /sys/kernel/debug/bluetooth/6lowpan_control`
but it returns:
`bash: echo: write error: No such file or directory`
and the connection keeps alive.
Then I find, if I change the address type in the disconnect command from Random Device Address (which is 2) to Public Device Address (which is 1) then I can disconnect with the node, and the 'bt0' interface is gone. However, the node still shows that the GAP is still connected. Is this normal? If not, how can I correct it?
[image]
Thanks!

janosbrodbeck, 2021-03-01:
Hey @sada45,

if you want to disconnect the connection via your Pi you can use the command-line tool `bluetoothctl`. Type in once in your terminal and you should enter a interactive terminal application. 

`info` prints all currently connected devices. When it prints `Missing device address argument` then no device is connected.
You can disconnect with `disconnect <ble addr>`.

sada45, 2021-03-02:
> Hey @sada45,
> 
> if you want to close the connection via your Pi you can use the command-line tool `bluetoothctl`. It should be installed on most distributions when a bluetooth stack is installed by default. Start it in your terminal and you should enter a interactive terminal application.
> 
> `info` prints all currently connected devices. When it prints `Missing device address argument` then no device is connected.
> You can disconnect with `disconnect <ble addr>`.
> 
> Maybe @haukepetersen knows more ways to disconnect.

The `bluetoothctl` works fine. Thanks a lot!
But I still dont know why the 6lowpan_control cannot work. There is code to handle disconnection in the Linux kernel source code, which located in [https://github.com/torvalds/linux/blob/e767b3530acbf651593e3d357fe1168a024d8061/net/bluetooth/6lowpan.c](url) Line 1119.

janosbrodbeck, 2021-03-02:
Wild guess without knowing the link between the linux kernel and bluetooth: Linux kernel may disconnect the node on 6LoWPAN network level but maybe it does not communicate to bluez (default bluetooth stack on most/nearly all linux distros) to disconnect on BLE level. Even when using IPv6-over-BLE you are still having a valid bluetooth GATT connection between your devices specified as in the core bluetooth specification.

sada45, 2021-03-02:
> Wild guess without knowing the link between the linux kernel and bluetooth:
> Linux kernel may remove the node on 6LoWPAN network level but maybe it does not communicate to your bluetooth stack (default is bluez on most/nearly all linux distros) to disconnect on BLE level. Even when using IPv6-over-BLE you are still having a valid bluetooth GATT connection between your devices specified as in the core bluetooth specification.

That sounds reasonable. I'll use the `bluetoothctl` for further use. Thanks @janosbrodbeck .

janosbrodbeck, 2021-03-02:
I close this issue then. If something comes up on this again, feel free to re-open/write again ;)

closed 2021-03-02 by janosbrodbeck