RIOT-OS/RIOT #16124: lwip_sock_tcp.c: incorrect socket buffer last_offset computation

ChristophRh, 2021-03-02:
#### Description

Issue when calling 
`ssize_t sock_tcp_read(sock_tcp_t *sock, void *data, size_t max_len, uint32_t timeout)`
multiple times to read the socket buffer in multiple(at least 3) steps.
If there is data left in the buffer, the offset relative to the total buffer length `sock->last_offset` is not computed correctly.
That leads to the fact that the next incoming data is not read from the beginning but from the previously wrongly computed last_offset, since the function is in the state that it thinks there is still data from the previous buffer left to read.

#### Example
There is a buffer of 10 Bytes available and in the first call we read 8 Bytes.
We’ll have sock->last_buf set and sock->last_offset is 8.

    346 /* post-process buf */
    347 if (buf_len > copylen) {
    348     /* there is still data in the buffer */
    349     sock->last_buf = buf;
    350     sock->last_offset = copylen;
    351 }

In the next call we read 1 Byte (copylen = 1). The remaining buffer length is computed by

    337 buf_len = buf->tot_len - sock->last_offset;

and obviously is 2.
There is still data in the buffer left to read, so again we’ll have sock->last_buf set and sock->last_offset now is 1, since copylen = 1.

If we want to read the last remaining Byte, we’ll compute the remaining buffer length to

    buf_len = buf->tot_len - sock->last_offset
                    10     -         1          = 9

which should not be the case.
The remaining buffer length should now be 0.

The problem occurs when I want to read new data from the socket buffer. Then I get the pointer of the old buffer with a wrong sock->last_offset value and with the previous example I would start reading at offset 1 instead of reading from the beginning of the new data.

#### Issue resolution
    350 sock->last_offset += copylen; 
should solve the issue, since it properly increments the last offset and the remaining buffer length will be computed correctly.

iosabi, 2021-04-03:
Hey, I ran into this exact issue. I agree with @ChristophRh with the resolution. Do you want me to send a PR for this?

miri64, 2021-04-04:
I currently have no time to look in-depth into this issue, so I'll be happy about any help I can get :-).

iosabi, 2021-04-04:
> I currently have no time to look in-depth into this issue, so I'll be happy about any help I can get :-).

No problem, I just created #16278 to fix this. It makes sense that without this fix it worked anyway in certain cases (if the application buffer size is larger than the TCP payload size).

PR #16278 (2021-04-07): lwip_sock: Fix computation of last_offset on partial reads.
### Contribution description

lwip receives network buffers that are made available to lwip_sock_tcp
calling `netconn_recv_tcp_pbuf`. The size of these buffers depends on
the size of the network packets. An application calling `sock_tcp_read`
can pass any arbitrary buffer size to read (copy) from these internal
network buffers, which may be smaller. lwip_sock_tcp keeps around the
last network buffer (`struct pbuf last_buf`) and the offset into this
buffer already consumed by the application (`last_offset`).

However, when multiple app [... cut]
closed 2021-04-07 by miri64