RIOT-OS/RIOT #16155: Question: DS3231 drivers, no alarms, can I add my own implem ?

nicoHarel, 2021-03-05:
Hello everyone,

### Context

I've been using the NRF52840 and after upgrading to a newer version of RIOT I've encountered a problem with the NRF52''s i2C. For some reason, after a few calls my application gets stuck while reading some i2C register on a DS3231 RTC. I've narrowed the problem to the "finish" function (nrf52/periph/i2c.c line 68). At one point I never seem to be able to get that mutex. I'm stuck forever waiting for it. 

Not noticing that a driver was being developed on RIOT, I've started making my own a year ago, it looks pretty similar, except I have developed some alarm function in mine but no aging offset fcts. 

### Question

Our conundrum : before, I had a functional driver with alarms, now it seems either I have to choose to use the official driver (but I loose the alarm setting abilities which is not a viable solution for us) or I keep my driver which doesn't seem to work anymore for some unknown reasons. 

So, since we can't really choose any of the "solution" above, my question is: is there any way I could add some alarm functions on top of the official driver which would not be considered a messy/disgusting solution ? 

I'd also gladly share my alarm functions but they're not very clean and they also miss some of the DS3231 alarm functionalities. 

Thank you

nicoHarel, 2021-03-05:
Okay so since it seems I have no fast solution to remedy this I have switched to the official DS3231 driver and started adapting my alarm implementation to something that would be more appropriate inside the driver. 

Should I open a pull request ?

jeandudey, 2021-03-06:
Hi @nicoHarel! Thanks for asking

>Is there any way I could add some alarm functions on top of the official driver which would not be considered a messy/disgusting solution?

Drivers are mostly designed to be general to support a wide use case, in this case if the alarm function that the DS3231 RTC provides isn't implemented in the driver it would be really good if you provided a Pull-Request with the desired changes and one of the maintainers will review it and help you in the process of the contribution.

This functionality could be added as an extension of the existing functionality, as in this case you need the driver to support the alarms and do the setup of the interrupts you may add additional functions for this, for example, most RIOT drivers use an additional function to set up the interrupts in the given device and initialize the GPIO interrupts, see [itg320x_init_int](https://github.com/RIOT-OS/RIOT/blob/master/drivers/itg320x/itg320x.c#L90) fas an example, it's gated behind a `MODULE_ITG320X_INT` so that function is only compiled when the pseudo-module `itg320x_int` is included, this because some CPUs may not provide support for IRQs on GPIOs, so using the `itg320x_int` dependency requires the feature `periph_gpio_irq` of the CPU.

---

As a side note, see also the [CONTRIBUTING.md](https://github.com/RIOT-OS/RIOT/blob/master/CONTRIBUTING.md) file and it's highly encouraged to follow the coding style of RIOT, see [CODING_CONVENTIONS.md](https://github.com/RIOT-OS/RIOT/blob/master/CODING_CONVENTIONS.md)

nicoHarel, 2021-03-08:
Thanks you for your answer.

I get the idea of including only basic features in riot drivers, although one could argue RTC alarms are features that do not necessarily require a GPIO interrupt to be usable. A simple GPIO polling can be enough in some instances (even though it's not optimal). In my case it was my application managing the interrupt, I considered that the driver's job is only to configure/enable/disable alarms and not take responsibility of the gpio's interrupt. After all it's up to the user to decide what to do with the alarm. In some cases the SQW pin of the RTC is not even connected to the MCU. In one of my platform this pin is connected to the enable pin of a voltage regulator, therefore I don't need/want any gpio interrupt. 

Don't you agree ? 

Anyway, I've adapted my alarm functions to the RIOT DS3231 driver, deleted my driver, and it seems that my problem disappeared. I don't know what changed exactly but I'm not blocked by the mutex anymore. The implementation is dirty so there is still of course a bit of work to do. I will open a Pull-Request as soon as I have got a bit of time.

jeandudey, 2021-03-08:
>A simple GPIO polling can be enough in some instances (even though it's not optimal).

A simple function can be provided for the user by the driver and with the proper documentation to poll the GPIO if IRQ functions aren't not enabled/needed.

> I considered that the driver's job is only to configure/enable/disable alarms and not take responsibility of the gpio's interrupt. After all it's up to the user to decide what to do with the alarm. [...]

Yeah that's how it should go, the driver initialization function of the alarm, or the function that sets up the alarm takes a function pointer parameter from the user and the user handles the IRQ when it's triggered in their own function.

nicoHarel, 2021-03-10:
PR opened :wink:

PR #16180

jeandudey, 2021-04-13:
Closing since #16180 got merged, thanks for your contribution!

closed 2021-04-13 by jeandudey