RIOT-OS/RIOT #16184: unittests/tests-rtc: fail on native

miri64, 2021-03-11:
#### Description
Currently, in the weekly release test runs, the unittests of [spec 1](https://github.com/RIOT-OS/Release-Specs/blob/676d417c97b7d4202d46b1d87f9aa724325862d1/01-ci/README.md) fail (so tasks [2](https://github.com/RIOT-OS/Release-Specs/blob/676d417c97b7d4202d46b1d87f9aa724325862d1/01-ci/README.md#task-02---unittests-on-native) and [3](https://github.com/RIOT-OS/Release-Specs/blob/676d417c97b7d4202d46b1d87f9aa724325862d1/01-ci/README.md#task-03---unittests-on-native-separated)). Specifically, `tests-rtc` is failing.

#### Steps to reproduce the issue
I can't reproduce this locally, even when running

```sh
```

#### Expected results

#### Actual results

#### Versions

kaspar030, 2021-03-11:
This one is indeed weird. Any ideas on how to debug this?

Seems to me that the github action builds in the same riotdocker container, and the test is executed in "ubuntu-latest", right?

miri64, 2021-03-11:
> Seems to me that the github action builds in the same riotdocker container, and the test is executed in "ubuntu-latest", right?

Yepp. What I did not try yet, was running the tests for tasks 1.2 and 1.3 [locally using pytest](https://github.com/RIOT-OS/Release-Specs#pytest-runner). Not sure this could reveal something. When https://github.com/RIOT-OS/Release-Specs/pull/210 was merged, the unittests were working (see [linked action there](https://github.com/RIOT-OS/Release-Specs/pull/210#issuecomment-780549092)), so maybe something changed about those tests in the meantime?

miri64, 2021-03-11:
BTW, the weekly run after RIOT-OS/Release-Specs#210 was merged, [also succeeded in those tests](https://github.com/RIOT-OS/RIOT/runs/1993539557) (the only fail there was due to https://github.com/RIOT-OS/Release-Specs/pull/212 not being merged yet). So something must have happened in those tests or how the tests are handled by Linux between February 27, 2021 and March 6, 2021.

miri64, 2021-03-22:
Ok, I ran the test now isolated for a30184b5a931c0de08f816ad19d6deb [here](https://github.com/miri64/RIOT/runs/2166017924) but it also fails there.

miri64, 2021-03-22:
@benpicco was expressed hope to me offline that https://github.com/RIOT-OS/RIOT/pull/15763 might fix this issue, but sadly [this is not the case](https://github.com/RIOT-OS/RIOT/runs/2166439828) (ignore the `iotlab_creds` runs, those should be unrelated...)

miri64, 2021-04-12:
Could it be that `mktime` is working differently on the Github runners than on a vanilla Ubuntu and does not normalize?

miri64, 2021-04-12:
Just for fun, I ran the tests with some output as shown in <details><summary>this patch</summary>

```diff
diff --git a/.github/workflows/release-test.yml b/.github/workflows/release-test.yml
index 8f339914e9..a5c9e7fb96 100644
--- a/.github/workflows/release-test.yml
+++ b/.github/workflows/release-test.yml
@@ -46,8 +46,8 @@ jobs:
     strategy:
       fail-fast: false
       matrix:
-        pytest_mark:
-          - "iotlab_creds"
+        # pytest_mark:
+        #   - "iotlab_creds"
         include:
           - pytest_mark: "not iotlab_creds"
             sudo: "sudo"
@@ -80,7 +80,7 @@ jobs:
     - name: Checkout RIOT
       uses: actions/checkout@v2
       with:
-        repository: RIOT-OS/RIOT
+        repository: miri64/RIOT
         path: RIOT
         fetch-depth: 1
         ref: ${{ github.event.inputs.riot_version }}
@@ -134,7 +134,7 @@ jobs:
           TTN_DEV_ID="riot_lorawan_1" \
           TTN_DEV_ID_ABP="riot_lorawan_1_abp" \
           RIOTBASE=${RIOTBASE} \
-          $(which tox) -- ${TOX_ARGS} -m "${{ matrix.pytest_mark }}"
+          $(which tox) -- ${TOX_ARGS} -m "${{ matrix.pytest_mark }}" -k 'spec01 and task02'
     - name: junit2html and XML deploy
       if: always()
       run: |
diff --git a/tests/unittests/tests-rtc/tests-rtc.c b/tests/unittests/tests-rtc/tests-rtc.c
index eeac8ea472..68735b812f 100644
--- a/tests/unittests/tests-rtc/tests-rtc.c
+++ b/tests/unittests/tests-rtc/tests-rtc.c
@@ -12,6 +12,7 @@
  * @file
  */
 #include <string.h>
+#include <stdio.h>
 #include <errno.h>
 
 #include "embUnit.h"
@@ -47,7 +48,15 @@ static void test_rtc_compat(void)
 
     struct tm t2 = t1;
 
+    printf("pre t1: sec=%i, min=%i, hour=%i, mday=%i, mon=%i, year=%i, "
+           "wday=%i, yday=%i, isdst=%i\n", t1.tm_sec, t1.tm_min, t1.tm_hour,
+           t1.tm_mday, t1.tm_mon, t1.tm_year, t1.tm_wday, t1.tm_yday,
+           t1.tm_isdst);
     mktime(&t1);
+    printf("post t1: sec=%i, min=%i, hour=%i, mday=%i, mon=%i, year=%i, "
+           "wday=%i, yday=%i, isdst=%i\n", t1.tm_sec, t1.tm_min, t1.tm_hour,
+           t1.tm_mday, t1.tm_mon, t1.tm_year, t1.tm_wday, t1.tm_yday,
+           t1.tm_isdst);
     rtc_tm_normalize(&t2);
     _test_equal_tm(&t1, &t2);
 }
@@ -68,7 +77,15 @@ static void test_rtc_sec_wrap(void)
 
     struct tm t2 = t1;
 
+    printf("pre t1: sec=%i, min=%i, hour=%i, mday=%i, mon=%i, year=%i, "
+           "wday=%i, yday=%i, isdst=%i\n", t1.tm_sec, t1.tm_min, t1.tm_hour,
+           t1.tm_mday, t1.tm_mon, t1.tm_year, t1.tm_wday, t1.tm_yday,
+           t1.tm_isdst);
     mktime(&t1);
+    printf("post t1: sec=%i, min=%i, hour=%i, mday=%i, mon=%i, year=%i, "
+           "wday=%i, yday=%i, isdst=%i\n", t1.tm_sec, t1.tm_min, t1.tm_hour,
+           t1.tm_mday, t1.tm_mon, t1.tm_year, t1.tm_wday, t1.tm_yday,
+           t1.tm_isdst);
     rtc_tm_normalize(&t2);
     _test_equal_tm(&t1, &t2);
 }
```

</details>

`mktime` does [indeed *not* do the normalization](https://github.com/miri64/RIOT/runs/2324467213?check_suite_focus=true#step:12:323) described [in `man 3 mktime`](https://linux.die.net/man/3/mktime):

```
.pre t1: sec=42, min=37, hour=13, mday=23, mon=5, year=84, wday=0, yday=0, isdst=1
post t1: sec=42, min=37, hour=13, mday=23, mon=5, year=84, wday=0, yday=0, isdst=1

rtc_tests.test_rtc_compat (tests/unittests/tests-rtc/tests-rtc.c 30) exp 0 was 6
.pre t1: sec=360, min=58, hour=23, mday=30, mon=5, year=100, wday=0, yday=0, isdst=1
post t1: sec=360, min=58, hour=23, mday=30, mon=5, year=100, wday=0, yday=0, isdst=1
```

<details><summary>(compared to output on my Ubuntu 20.04 machine)</summary>

```
.pre t1: sec=42, min=37, hour=13, mday=23, mon=5, year=84, wday=0, yday=0, isdst=1
post t1: sec=42, min=37, hour=13, mday=23, mon=5, year=84, wday=6, yday=174, isdst=1
.pre t1: sec=360, min=58, hour=23, mday=30, mon=5, year=100, wday=0, yday=0, isdst=1
post t1: sec=0, min=4, hour=0, mday=1, mon=6, year=100, wday=6, yday=182, isdst=1
```

</details>

@benpicco would it make sense to check against a statically initialized `t1` instead?

benpicco, 2021-04-12:
> @benpicco would it make sense to check against a statically initialized `t1` instead?

The way the test works now it also makes sure that the RTC driver accepts non-normalized time (as it may happen often when just increasing a single element of the time struct), so I'd rather no lose this implicit test of the API contract.

Maybe updating Riotdocker to Ubuntu 20.04 would fix this?

miri64, 2021-04-12:
> Maybe updating Riotdocker to Ubuntu 20.04 would fix this?

Not sure the riotdocker version is the problem. We don't face the problem on Murdock which also uses the docker image.

miri64, 2021-04-12:
Just to confirm: this is with the current `riot/riotbuild:latest` on my Ubuntu 20.04 machine:

```
BUILD_IN_DOCKER=1 make -C tests/unittests/ tests-rtc test
make: Entering directory '/home/mlenders/Repositories/RIOT-OS/RIOT/tests/unittests'
Launching build container using image "riot/riotbuild:latest".
docker run --rm --tty --user $(id -u) -v '/usr/share/zoneinfo/Europe/Berlin:/etc/localtime:ro' -v '/home/mlenders/Repositories/RIOT-OS/RIOT:/data/riotbuild/riotbase:delegated' -e 'RIOTBASE=/data/riotbuild/riotbase' -e 'CCACHE_BASEDIR=/data/riotbuild/riotbase' -e 'BUILD_DIR=/data/riotbuild/riotbase/build' -e 'RIOTPROJECT=/data/riotbuild/riotbase' -e 'RIOTCPU=/data/riotbuild/riotbase/cpu' -e 'RIOTBOARD=/data/riotbuild/riotbase/boards' -e 'RIOTMAKE=/data/riotbuild/riotbase/makefiles'        -w '/data/riotbuild/riotbase/tests/unittests/' 'riot/riotbuild:latest' make      tests-rtc
Building application "tests_unittests" for "native" with MCU "native".

"make" -C /data/riotbuild/riotbase/boards/native
"make" -C /data/riotbuild/riotbase/boards/native/drivers
"make" -C /data/riotbuild/riotbase/core
"make" -C /data/riotbuild/riotbase/cpu/native
"make" -C /data/riotbuild/riotbase/cpu/native/periph
"make" -C /data/riotbuild/riotbase/cpu/native/stdio_native
"make" -C /data/riotbuild/riotbase/drivers
"make" -C /data/riotbuild/riotbase/drivers/periph_common
"make" -C /data/riotbuild/riotbase/sys
"make" -C /data/riotbuild/riotbase/sys/embunit
"make" -C /data/riotbuild/riotbase/sys/test_utils/interactive_sync
"make" -C /data/riotbuild/riotbase/tests/unittests/tests-rtc
   text	   data	    bss	    dec	    hex	filename
  34510	    796	  51864	  87170	  15482	/data/riotbuild/riotbase/tests/unittests/bin/native/tests_unittests.elf
r
/home/mlenders/Repositories/RIOT-OS/RIOT/tests/unittests/bin/native/tests_unittests.elf /dev/ttyACM0 
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2021.04-devel-1315-g9926e3)
Help: Press s to start test, r to print it is ready
READY
s
START
.........
OK (9 tests)

make: Leaving directory '/home/mlenders/Repositories/RIOT-OS/RIOT/tests/unittests'
```

miri64, 2021-04-12:
Also, aren't the test run on the host and the libc linked dynamically with `native`? So I think at this point the docker container has no influence on the test anymore.

miri64, 2021-04-12:
Just noticed: https://github.com/RIOT-OS/RIOT/runs/1993539573 which was the last where the unittests succeeded in the release tests where the last ones run on Ubuntu 18.04. The once [after](https://github.com/RIOT-OS/RIOT/actions/runs/626359578) where run on Ubuntu 20.04.

miri64, 2021-04-13:
A system update before the release test run does not yield success either: https://github.com/miri64/RIOT/runs/2324970605

miri64, 2021-04-13:
> _edit_: Ah sorry I got confused. This is about `tests-rtc/tests-rtc.c`, not `periph_rtc/main.c`. So yea, statically initializing `t1` should be fine.

Okay, then this probably should be the road to go :-).

miri64, 2021-04-13:
Okay, but then again, I wondered while doing that: why is this only happening for the first to tests, while the other usages of `mktime` are working perfectly fine. The only difference between those tests and the failing ones is that for the failing ones, `tm_isdst` is set. If `tm_isdst` is set to `-1` (DST-info unavailable), the tests succeed in the [Github Actions](https://github.com/miri64/RIOT/runs/2333434802?check_suite_focus=true). While it does not really explain, why `mktime` is basically a NOP with `isdst` set, it would be more in line with the function under test, `rtc_tm_normalize`, which does not care about DST.

Simultaneously, @benpicco and I found offline, that when `TZ` is set in the environment (in any way, even an empty string) makes the test failure locally reproducible:

```
TZ='' make -C tests/unittests/ -j tests-rtc test
make: Entering directory '/home/mlenders/Repositories/RIOT-OS/RIOT/tests/unittests'
Building application "tests_unittests" for "native" with MCU "native".

"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/boards/native
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/core
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/native
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/drivers
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/tests/unittests/tests-rtc
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/boards/native/drivers
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/drivers/periph_common
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/embunit
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/test_utils/interactive_sync
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/native/periph
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/native/stdio_native
/usr/bin/ld: /home/mlenders/Repositories/RIOT-OS/RIOT/tests/unittests/bin/native/cpu/tramp.o: warning: relocation against `_native_saved_eip' in read-only section `.text'
/usr/bin/ld: warning: creating DT_TEXTREL in a PIE
   text	   data	    bss	    dec	    hex	filename
  34635	    772	  51912	  87319	  15517	/home/mlenders/Repositories/RIOT-OS/RIOT/tests/unittests/bin/native/tests_unittests.elf
r
/home/mlenders/Repositories/RIOT-OS/RIOT/tests/unittests/bin/native/tests_unittests.elf /dev/ttyACM0 
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2021.04-devel-1336-ge38e0-tests/fix/tests-rtc)
Help: Press s to start test, r to print it is ready
READY
s
START
.
rtc_tests.test_rtc_compat (tests/unittests/tests-rtc/tests-rtc.c 29) exp 0 was 6
.
rtc_tests.test_rtc_sec_wrap (tests/unittests/tests-rtc/tests-rtc.c 23) exp 360 was 0
.......
run 9 failures 2
```

I am currently running a test run, testing if `unset TZ` in the workflow does change anything.

miri64, 2021-04-13:
> running a test run, testing if `unset TZ` in the workflow does change anything.

[Nope](https://github.com/miri64/RIOT/runs/2333592594?check_suite_focus=true#step:12:61). So I guess we go the 'disable DST' route.

miri64, 2021-04-13:
See https://github.com/RIOT-OS/RIOT/pull/16325

closed 2021-04-13 by miri64