RIOT-OS/RIOT #16207: CAN: raw_can_set/get_bitrate doesn't work on native board.

Ciusss89, 2021-03-18:
#### Steps to reproduce the issue
Compile test **conn_can** for native board.

#### Expected results
This issue regards only NATIVE board,  it works fine on my STM32 board 

#### Actual results
both callback set/get bitrate fails:
```
> test_can list
test_can list
CAN #0: can0
CAN #1: can1
> test_can get_bitrate 0
test_can get_bitrate 0
no link data found
Error when getting bitrate: res=-95

```
#### Versions
current master.

vincent-d, 2021-03-19:
There are 2 things with bitrate on native IIRC:
- virtual CAN interface do not have a bitrate
- changing bitrate on a real CAN interface with libsocketCAN requires root access

Ciusss89, 2021-03-19:
> There are 2 things with bitrate on native IIRC:
> 
>     * virtual CAN interface do not have a bitrate
> 
>     * changing bitrate on a real CAN interface with libsocketCAN requires root access

Ok, this means that is not a bug.

Just a question, to test a specific bitrate with the native board, can I use the userspace **ip** command, eg : `sudo ip link set vcan0 type can bitrate 1000000` 
should be equal or not?

vincent-d, 2021-03-19:
> can I use the userspace ip command, eg : sudo ip link set vcan0 type can bitrate 1000000

yes, this is equivalent as what libsocketCAN does (note that you are using the `sudo` command ;) )

Ciusss89, 2021-03-19:
> > can I use the userspace ip command, eg : sudo ip link set vcan0 type can bitrate 1000000
> 
> yes, this is equivalent as what libsocketCAN does (note that you are using the `sudo` command ;) )

Thanks for the support. I close the issue :beers:

closed 2021-03-19 by Ciusss89