RIOT-OS/RIOT #16216: gnrc_border_router hands out prefix, but does not do routing

benpicco, 2021-03-21:
#### Description

After a few clients have connected to the border router, it will stop routing packets for new clients, even though they receive a valid prefix.

#### Steps to reproduce the issue

1. start the `gnrc_border_router` on `native`:
     `make -C examples/gnrc_border_router BOARD=native all term`

2. start a `gnrc_networking` node
     `make -C examples/gnrc_networking BOARD=native USE_ZEP=1 all term`
3. make sure you can reach the host

```
ping fdea:dbee:f::1
12 bytes from fdea:dbee:f::1: icmp_seq=0 ttl=63 rssi=255 dBm time=0.554 ms
12 bytes from fdea:dbee:f::1: icmp_seq=1 ttl=63 rssi=255 dBm time=0.613 ms
12 bytes from fdea:dbee:f::1: icmp_seq=2 ttl=63 rssi=255 dBm time=0.746 ms

--- fdea:dbee:f::1 PING statistics ---
3 packets transmitted, 3 packets received, 0% packet loss
round-trip min/avg/max = 0.554/0.637/0.746 ms
```

4. close the node, goto 2.

#### Expected results

The host is always reachable.

#### Actual results

After a few iterations the node can not reach the host (or the border router) using the global address anymore.
This is despite the node receiving a valid prefix:

##### broken node

```
Iface  7  HWaddr: 6F:E5  Channel: 26  NID: 0x23 
          Long HWaddr: AE:BA:1A:DA:7E:03:6F:E5 
          L2-PDU:102  MTU:1280  HL:64  RTR  
          RTR_ADV  6LO  IPHC  
          Source address length: 8
          Link type: wireless
          inet6 addr: fe80::acba:1ada:7e03:6fe5  scope: link  VAL
          inet6 addr: 2001:db8::acba:1ada:7e03:6fe5  scope: global  VAL
          inet6 group: ff02::2
          inet6 group: ff02::1
          inet6 group: ff02::1:ff03:6fe5
          
          Statistics for Layer 2
            RX packets 22  bytes 1466
            TX packets 29 (Multicast: 1)  bytes 1787
            TX succeeded 58 errors 0
          Statistics for IPv6
            RX packets 19  bytes 1680
            TX packets 29 (Multicast: 1)  bytes 1968
            TX succeeded 29 errors 0

> ping fdea:dbee:f::1

--- fdea:dbee:f::1 PING statistics ---
3 packets transmitted, 0 packets received, 100% packet loss
> nib neigh
fe80::ac95:485f:763b:179d dev #7 lladdr AE:95:48:5F:76:3B:17:9D router REACHABLE GC
> nib abr
2001:db8::a87c:972e:9ae:7c50 v0 expires 9994min
> ping 2001:db8::a87c:972e:9ae:7c50
ping 2001:db8::a87c:972e:9ae:7c50

--- 2001:db8::a87c:972e:9ae:7c50 PING statistics ---
3 packets transmitted, 0 packets received, 100% packet loss
```

##### working node

```
Iface  7  HWaddr: 17:9D  Channel: 26  NID: 0x23 
          Long HWaddr: AE:95:48:5F:76:3B:17:9D 
          L2-PDU:102  MTU:1280  HL:64  RTR  
          RTR_ADV  6LO  IPHC  
          Source address length: 8
          Link type: wireless
          inet6 addr: fe80::ac95:485f:763b:179d  scope: link  VAL
          inet6 addr: 2001:db8::ac95:485f:763b:179d  scope: global  VAL
          inet6 group: ff02::2
          inet6 group: ff02::1
          inet6 group: ff02::1:ff3b:179d
          
          Statistics for Layer 2
            RX packets 71  bytes 4348
            TX packets 57 (Multicast: 1)  bytes 3701
            TX succeeded 114 errors 0
          Statistics for IPv6
            RX packets 69  bytes 4932
            TX packets 53 (Multicast: 1)  bytes 3828
            TX succeeded 53 errors 0

> ping fdea:dbee:f::1
12 bytes from fdea:dbee:f::1: icmp_seq=0 ttl=63 rssi=255 dBm time=0.780 ms
12 bytes from fdea:dbee:f::1: icmp_seq=1 ttl=63 rssi=255 dBm time=0.696 ms
12 bytes from fdea:dbee:f::1: icmp_seq=2 ttl=63 rssi=255 dBm time=0.752 ms

--- fdea:dbee:f::1 PING statistics ---
3 packets transmitted, 3 packets received, 0% packet loss
round-trip min/avg/max = 0.696/0.742/0.780 ms

> nib neigh
fe80::a87c:972e:9ae:7c50 dev #7 lladdr AA:7C:97:2E:09:AE:7C:50 router REACHABLE GC
2001:db8::acba:1ada:7e03:6fe5 dev #7 lladdr AE:BA:1A:DA:7E:03:6F:E5  STALE REGISTERED
> nib abr
2001:db8::a87c:972e:9ae:7c50 v0 expires 9993min
> ping 2001:db8::a87c:972e:9ae:7c50
ping 2001:db8::a87c:972e:9ae:7c50
12 bytes from 2001:db8::a87c:972e:9ae:7c50: icmp_seq=0 ttl=64 rssi=255 dBm time=0.636 ms
12 bytes from 2001:db8::a87c:972e:9ae:7c50: icmp_seq=1 ttl=64 rssi=255 dBm time=0.640 ms
12 bytes from 2001:db8::a87c:972e:9ae:7c50: icmp_seq=2 ttl=64 rssi=255 dBm time=0.664 ms

--- 2001:db8::a87c:972e:9ae:7c50 PING statistics ---
3 packets transmitted, 3 packets received, 0% packet loss
round-trip min/avg/max = 0.636/0.646/0.664 ms
```

##### border router

```
Iface  7  HWaddr: 7C:50  Channel: 26  NID: 0x23 
          Long HWaddr: AA:7C:97:2E:09:AE:7C:50 
          L2-PDU:102  MTU:1280  HL:64  RTR  
          RTR_ADV  6LO  IPHC  
          Source address length: 8
          Link type: wireless
          inet6 addr: fe80::a87c:972e:9ae:7c50  scope: link  VAL
          inet6 addr: 2001:db8::a87c:972e:9ae:7c50  scope: global  VAL
          inet6 group: ff02::2
          inet6 group: ff02::1
          inet6 group: ff02::1:ffae:7c50
          
Iface  6  HWaddr: 06:91:B1:6B:24:DE 
          L2-PDU:1500  MTU:1500  HL:64  RTR  
          Source address length: 6
          Link type: wired
          inet6 addr: fe80::491:b1ff:fe6b:24de  scope: link  VAL
          inet6 addr: fe80::2  scope: link  VAL
          inet6 group: ff02::2
          inet6 group: ff02::1
          inet6 group: ff02::1:ff6b:24de
          inet6 group: ff02::1:ff00:2
          
> nib neigh
fe80::1 dev #6 lladdr 06:91:B1:6B:24:DD  STALE GC
2001:db8::e43f:9e61:85:a5ef dev #7 lladdr E6:3F:9E:61:00:85:A5:EF  STALE REGISTERED
2001:db8::ac95:485f:763b:179d dev #7 lladdr AE:95:48:5F:76:3B:17:9D  STALE REGISTERED
2001:db8::b8d0:a392:e58a:c0ac dev #7 lladdr BA:D0:A3:92:E5:8A:C0:AC  PROBE REGISTERED
2001:db8::b85c:8804:9015:357f dev #7 lladdr BA:5C:88:04:90:15:35:7F  PROBE REGISTERED
2001:db8::a8fc:c166:2b08:82b2 dev #7 lladdr AA:FC:C1:66:2B:08:82:B2  PROBE REGISTERED
> nib prefix
2001:db8::/64 dev #7 
```

#### Versions

RIOT master

miri64, 2021-03-22:
For that you do need a routing protocol, e.g. like RPL. NDP router discovery can only help you with finding default routers, but not the downlink routes. `nib route` shows you that ;-).

benpicco, 2021-03-22:
I don't understand - if there is only a single hop, shouldn't it be enough to know the default router?

miri64, 2021-03-22:
> I don't understand - if there is only a single hop, shouldn't it be enough to know the default router?

For this the broken node needs to be in the BR's neighbor cache, but as far as I can see, it is not but in the neighbor cache of the "working node". So maybe it is just full and you need to increase the size of the BR's neighbor cache and not make the nodes routers but only hosts (so use `gnrc_ipv6_default` instead of `gnrc_ipv6_router_default)`, if you aim for single hop?

benpicco, 2021-03-22:
Ok, so the issue here is that I have several `gnrc_ipv6_router_default` nodes and sometimes they select each other as the default route instead of the border router.

This does not happen with `gnrc_ipv6_default` nodes.

closed 2021-03-22 by benpicco