RIOT-OS/RIOT #16240: openmote-b, 915MHz

edgaralb, 2021-03-28:
Hi @benpicco, Please, when will the openmote-b (at86rf215) settings for the 915MHz frequencies be available, or what configuration files or changes can be added to support those frequencies.
Best regards.

benpicco, 2021-04-09:
Hi @edgaralb,
unfortunately there is not a good way to achieve this yet.
You can of course manually adjust the frequencies in the driver (e.g. `QPSK_CHANNEL_SPACING_SUBGHZ` or `_chan_center_freq0_subghz_25khz`) but you might also have to modify channel spacing according to local regulations. 

There might be some more settings to adjust to comply to local regulations, as well as which modulations parameters you can choose.

Now all this information should not be in the driver.
The plan is to have a regulatory database that can be used by *all* drivers and keep the drivers 'dump' so we only set center frequency directly and let the upper layer choose what's appropriate depending on the region and channel.

Now as you can see, this has not materialized yet. 
So if you want a quick solution, you can patch the driver, but the proper solution will involve moving this out of the driver entirely.

jeandudey, 2021-04-12:
In the case of legacy modulations, in this case O-QPSK where the channel page and number specifies the center frequencies something like https://github.com/RIOT-OS/RIOT/compare/master...jeandudey:2020_04_10-at86rf215?expand=1 could be done, it's not tested yet but more as a proof of concept, will open the PR as a draft to see if the approach is viable.

closed 2022-04-18 by stale[bot]