RIOT-OS/RIOT #16242: I found stm32 DMA periph driver bugs!  when I tested stm32l431rc board.

skyarm, 2021-03-29:
#### Description

  The next issue, at cpu/stm32/periph/spi.c, The issue can run correctly, But it is just a coincidence!!!
     dma_setup(spi_config[bus].tx_dma,
                  spi_config[bus].tx_dma_chan,
                  (uint32_t*)&(dev(bus)->DR),
                  DMA_MEM_TO_PERIPH,
                  0,
                  DMA_DATA_WIDTH_BYTE);

  But the prototype of the dma_setup is:
    void dma_setup(dma_t dma, int chan, void *periph_addr, dma_mode_t mode,
               uint8_t width, bool inc_periph);
So correctly call must be:
     dma_setup(spi_config[bus].tx_dma,
                  spi_config[bus].tx_dma_chan,
                  (uint32_t*)&(dev(bus)->DR),
                  DMA_MEM_TO_PERIPH,
                  DMA_DATA_WIDTH_BYTE, 
                  0);
Although the correct result can be obtained, it is just a coincidence!!!
  
#### Steps to reproduce the issue

#### Expected results

#### Actual results

#### Versions

skyarm, 2021-03-29:
The next issue: The IRQ Vector value isn't correct if the stream parameter >= 7 or >=  12
  at cpu/stm32/periph/dma.c line 208

   #if defined(DMA2_BASE)
#if defined(CPU_FAM_STM32F1)
    else if (stream < 11) {
#else
    else if (stream < 13 ) {
#endif
        return ((IRQn_Type)((int)DMA2_Channel1_IRQn + stream));
    }
#if !defined(CPU_FAM_STM32L1)
    else {
#if defined(CPU_FAM_STM32F1)
        return (DMA2_Channel4_5_IRQn);
#else
        return ((IRQn_Type)((int)DMA2_Channel6_IRQn + stream));
#endif
    }
#endif /* !defined(CPU_FAM_STM32L1) */
#endif /* defined(DMA2_BASE) */

skyarm, 2021-03-29:
The next issue, at cpu/stm32/periph/spi.c. line 236, The issue can run correctly, But it is just a coincidence!!!
dma_setup(spi_config[bus].tx_dma,
spi_config[bus].tx_dma_chan,
(uint32_t*)&(dev(bus)->DR),
DMA_MEM_TO_PERIPH,
0,
DMA_DATA_WIDTH_BYTE);

But the prototype of the dma_setup is:
void dma_setup(dma_t dma, int chan, void periph_addr, dma_mode_t mode,
uint8_t width, bool inc_periph);
So correctly call must be:
dma_setup(spi_config[bus].tx_dma,
    spi_config[bus].tx_dma_chan,
    (uint32_t)&(dev(bus)->DR),
    DMA_MEM_TO_PERIPH,
    DMA_DATA_WIDTH_BYTE,
   false);
Although the correct result can be obtained, it is just a coincidence!!!

maribu, 2023-01-05:
Let me just claim https://github.com/RIOT-OS/RIOT/pull/18711 fixed the issue. Please reopen if that is not correct.

closed 2023-01-05 by maribu