RIOT-OS/RIOT #16249: Getting UART error in ESp32-wroom

Eleftheria-99, 2021-03-29:
#### Description

#### Steps to reproduce the issue

#### Expected results

#### Actual results

#### Versions

jeandudey, 2021-04-01:
@Eleftheria-99 could you provide more information?

jeandudey, 2021-04-03:
Closing since no information was provided, feel free to reopen this or open a new issue with more details about the problem.

Eleftheria-99, 2021-04-03:
@jeandudey I am so sorry, I just saw, I added all the necessary info , can we reopen the issue ??

Eleftheria-99, 2021-04-03:
@jeandudey  also how did you add a label ?

jeandudey, 2021-04-03:
>@jeandudey also how did you add a label ?

Only members of the repository can add labels

jeandudey, 2021-04-03:
@Eleftheria-99 what version of RIOT are you using? The version (e.g.: 2021.01) or the last commit (`git log` on RIOT directory).

gschorcht, 2021-04-04:
The warning message is thrown if the flash mode is either `qout` or `qio`. In that case, the Quad SPI interface is used for SPI flash which in turn occupies GPIO 9 an GPIO 10 so that they can't be used as UART interface.

Could you check whether this is the case by using the the following command:
```
BOARD=esp32-wroom-32 make -C <your app directory> info-debug-variable-FLASH_MODE
```

Eleftheria-99, 2021-04-04:
> @Eleftheria-99 what version of RIOT are you using? The version (e.g.: 2021.01) or the last commit (`git log` on RIOT directory).

I think 2021.01

Eleftheria-99, 2021-04-04:
> The warning message is thrown if the flash mode is either `qout` or `qio`. In that case, the Quad SPI interface is used for SPI flash which in turn occupies GPIO 9 an GPIO 10 so that they can't be [...]
I did not understand what the warning means  why the warning is being thrown since I did not declare a flash mode when running the code !! The command we used to run the code was : 
cd into directory 
sudo BUILD_IN_DOCKER=1 BOARD=esp32-wroom-32 make  
but we can try also the one you proposed

gschorcht, 2021-04-05:
> I did not understand what the warning means why the warning is being thrown since I did not declare a flash mode when running the code

It's only a warning and just means that UART1 can't be used as configured for the board by default because the flash mode is `qout` or `qio`. `FLASH_MODE` variable might be set automatically, for example if you use module `esp_spi_ram`. Therefore, please send the output of following commands:

```
BOARD=esp32-wroom-32 make info-debug-variable-USEMODULE
BOARD=esp32-wroom-32 make info-debug-variable-FLASH_MODE
```
This should also work in docker.

Eleftheria-99, 2021-04-05:
results : 

i just ran it again with the command you gave and it retunred: 
make: * info-debug-variable-FLASH_MODE: No such file or directory.  Stop.

[image]

gschorcht, 2021-04-05:
> i just ran it again with the command you gave and it retunred:
> make: * info-debug-variable-FLASH_MODE: No such file or directory. Stop.

I'm really sorry, but you are wrong. I never gave you command
```
BOARD=esp32-wroom-32 make -C info-debug-variable-FLASH_MODE
```
Of course, this command doesn't work because option `-C` requires to specify a build directory. The first time I gave you the following command
```
BOARD=esp32-wroom-32 make -C <your app directory> info-debug-variable-FLASH_MODE
```
in which you would have to replace `<your app directory>` by your application directory. But since you have used command 
```
sudo BUILD_IN_DOCKER=1 BOARD=esp32-wroom-32 make
```
according to your answer, I thought you are calling `make` inside your application directory and it is not necessary to specify application directory by option `-C`. So I gave you command
```
BOARD=esp32-wroom-32 make info-debug-variable-USEMODULE
BOARD=esp32-wroom-32 make info-debug-variable-FLASH_MODE
```
last time without option `-C`!

Please read more carefully next time.

gschorcht, 2021-04-05:
Regarding the result. As supposed, the flash mode is `qout` because module `esp_spi_ram´ is used. You will find these information in the output of the last two commands.

Since the `esp32-wroom-32` board doesn't have SPI RAM, module `esp_spi_ram` shouldn't be enabled. To be able to figure out what the problem might be in your case, I would need the `Makefile` of your application. Could you please provide it?

Eleftheria-99, 2021-04-07:
ok we will run it again

Eleftheria-99, 2021-04-09:
[image]

Eleftheria-99, 2021-04-09:
@gschorcht 

[image]

gschorcht, 2021-04-09:
> @gschorcht
> 
> [image]

Ok, this is the same as we had seen before, variable `FLASH_MODE` is set to `qout` because module `esp_spi_ram` is used.

According to your screenshot you are using `examples/saul` as application, right? I suspect that you have modified it, since `examples/saul` in the master repository neither uses the `esp_spi_ram` module nor has the `FLASH_MODE` variable set to `qout`. 

Again, to find out what the problem in your case might be, I would need the Makefile of your application. Therefore, may I ask you again to provide your Makefile?

Eleftheria-99, 2021-04-09:
Makefile:

```
# name of your application
APPLICATION = saul_example

# If no BOARD is found in the environment, use this default:
BOARD ?= native

# This has to be the absolute path to the RIOT base directory:
RIOTBASE ?= $(CURDIR)/../..

# we want to use SAUL:
USEMODULE += saul_default
# include the shell:
USEMODULE += shell
USEMODULE += shell_commands
# additional modules for debugging:
USEMODULE += ps

# Include packages that pull up and auto-init the link layer.
# NOTE: 6LoWPAN will be included if IEEE802.15.4 devices are present
USEMODULE += gnrc_netdev_default
USEMODULE += auto_init_gnrc_netif
# Activate ICMPv6 error messages
USEMODULE += gnrc_icmpv6_error
# Specify the mandatory networking modules for IPv6 and UDP
USEMODULE += gnrc_ipv6_router_default
USEMODULE += gnrc_udp
# Add a routing protocol
USEMODULE += gnrc_rpl
USEMODULE += auto_init_gnrc_rpl
# This application dumps received packets to STDIO using the pktdump module
USEMODULE += gnrc_pktdump
# Additional networking modules that can be dropped if not needed
USEMODULE += gnrc_icmpv6_echo
# Add also the shell, some shell commands
USEMODULE += netstats_l2
USEMODULE += netstats_ipv6
USEMODULE += netstats_rpl

#USEMODULE += esp_spiffs
USEMODULE += esp_head
USEMODULE += esp_hw_counter
USEMODULE += esp_spi_ram
FEATURES_PROVIDED += periph_can

USEMODULE += saul_default
USEMODULE += printf_float
USEMODULE += xtimer

USEMODULE += newlib_nano
USEMODULE += gnrc_udp
USEMODULE += sock_udp  
USEMODULE += gnrc_ipv6_default  
# Comment this out to disable code in RIOT that does safety checking
# which is not needed in a production environment but helps in the
# development process:
DEVELHELP ?= 1

# Change this to 0 show compiler invocation lines by default:
QUIET ?= 1

#add current dir to the include path, 
#putting it in cflags will make it go to the beginning, before the stnadard includes. That way slip_params.h gets included and auto configuration can pick it up
CFLAGS += -I$(CURDIR)

# development process:
CFLAGS += -DDEVELHELP
CFLAGS=-DAUTO_INIT_LED0

#features
FEATURES_REQUIRED += periph_adc
FEATURES_REQUIRED += periph_gpio
FEATURES_REQUIRED += periph_gpio_irq
include $(RIOTBASE)/Makefile.include

gschorcht, 2021-04-09:
> #USEMODULE += esp_spiffs
> USEMODULE += esp_head
> USEMODULE += esp_hw_counter
> USEMODULE += esp_spi_ram

Here you have explicitly enabled module `esp_spi_ram`. Why, the ESP32-WROOM-32 module doesn't have SPI RAM? This is the reason that `FLASH_MODE` is set to `qout` and you get this warning message.

jeandudey, 2021-05-15:
Closing as the issue as been already solved, if not, please re-open it.

closed 2021-05-15 by jeandudey