RIOT-OS/RIOT #16251: XFA support on AVR and MSP430 broken with binutils 2.36.1

maribu, 2021-03-30:
#### Description

It seems there is an issue in the linker scripts for AVR and MSP430 (didn't test all archs, but ARM and RISC-V still work) that surfaces after updating to binutils 2.36.1

#### Steps to reproduce the issue

- `make BOARD=arduino-uno -C examples/hello_work`
- `make BOARD=z1 -C examples/hello_work`

with binutils 2.36.1

#### Expected results

Linking succeeds

#### Actual results

AVR:

```
/usr/lib/gcc/avr/10.2.0/../../../../avr/bin/ld: section .bss VMA [0000000000800100,000000000080047e] overlaps section .data VMA [0000000000800060,0000000000800187]
collect2: error: ld returned 1 exit status
make: *** [/home/maribu/Repos/software/RIOT/examples/hello-world/../../Makefile.include:625: /home/maribu/Repos/software/RIOT/examples/hello-world/bin/arduino-uno/hello-world.elf] Error 1
```

MSP430:

```
/usr/lib/gcc/msp430-elf/10.2.0/../../../../msp430-elf/bin/ld: warning: /home/maribu/Repos/software/RIOT/cpu/msp430_common/ldscripts/xfa.ld contains output sections; did you forget -T?
/usr/lib/gcc/msp430-elf/10.2.0/../../../../msp430-elf/bin/ld: .rodata not found for insert
collect2: error: ld returned 1 exit status
make: *** [/home/maribu/Repos/software/RIOT/examples/hello-world/../../Makefile.include:625: /home/maribu/Repos/software/RIOT/examples/hello-world/bin/z1/hello-world.elf] Error 1
```

Node: Commenting out the including of the corresponding `xfa.ld` linker script fixes the issue.

#### Versions

Current `master`.

maribu, 2021-03-30:
Note that AVR until recently contained custom linker scripts for backward compatibility with ancient toolchains. It might very well be that dropping those resulted in the regression.

maribu, 2021-03-30:
The binaries generated with `binutils-avr` in version 2.36.1 are broken even without XFA for ATxmega. With 2.35.2 and 2.35.1 it works fine. Maybe the bug is in binutils?

kaspar030, 2021-04-26:
> The binaries generated with `binutils-avr` in version 2.36.1 are broken even without XFA for ATxmega. With 2.35.2 and 2.35.1 it works fine. Maybe the bug is in binutils?

Maybe... Anyhow, fixing this won't be done in time for the release.

maribu, 2021-08-11:
For AVR linking with binutils 2.37 works again, but the generated binaries won't boot :-/

Anyone cares to confirm the issue with an alternative toolchain to rule out issues with the Alpine packages?