RIOT-OS/RIOT #16256: Refactor microbit LED matrix into generic LED matrix

chrysn, 2021-03-30:
#### Description

The current microbit module takes a somewhat odd role in being a driver that is specific to a group of boards, and in that it has a top block of defines that pick based on the board how to be configured. (This is where this all came up, in https://github.com/RIOT-OS/RIOT/pull/16250#discussion_r603981661).

There isn't anything microbit specific in the module implementation itself, though -- it's "just" a timer-driven setting of row and column GPIOs fast enough to show a good matrix display image. The parameters like row and column count that are alike for all microbits are already generalized, and the microbit versions are distinct enough to show that the module can both be used with logically aligned rows / columns (as is in the v2) and with arbitrary wiring (as in the v1, which drives 3x9 LEDs in a matrix but places them in a 5x5 grid spatially).

### Useful links

In the current docs:
* [microbit matrix display module](https://riot-os.org/api/group__boards__common__microbit.html)

### Open questions

@aabadie, if there was good reason to not do this, please let me know here so I won't waste time trying.

aabadie, 2021-04-30:
> if there was good reason to not do this, please let me know here so I won't waste time trying.

The main reason was to keep the change (grouping common code across boards) small initially. But I agree this module could go under drivers with sensible parameters that could overloaded at board level.

chrysn, 2022-03-02:
I still hope to resume this at one time, especially with the special case that one axis may have PWM pins assigned.