RIOT-OS/RIOT #16267: Is ADC driver implemented for SAMD21G18A?

IchiroKuroki, 2021-04-01:
The code does not compile if I use the adc.h interface with arduino-nano-33-iot.

#include "periph/adc.h"
adc_init(ADC_LINE(1));

make flash BOARD=arduino-nano-33-iot

undefined reference to `adc_init'
collect2: error: ld returned 1 exit status

If I add in the Makefile:

FEATURES_REQUIRED += periph_adc 

then: There are unsatisfied feature requirements: periph_adc

jeandudey, 2021-04-01:
Hi @IchiroKuroki , could you try to add `FEATURES_PROVIDED += periph_adc` to `RIOT/boards/arduino-nano-33-iot` and `select HAS_PERIPH_ADC` to `RIOT/boards/arduino-nano-33-iot` and try recompiling and verifying that the ADC works?

IchiroKuroki, 2021-04-01:
Hi @jeandudey. After following your advice, I got new errors:

/home/ichiro/Documents/projects/RIOT/cpu/sam0_common/periph/adc.c:90:9: error: 'ADC_REF_DEFAULT' undeclared (first use in this function)
   90 |     if (ADC_REF_DEFAULT == ADC_REFCTRL_REFSEL_INT1V) {
      |         ^~~~~~~~~~~~~~~
/home/ichiro/Documents/projects/RIOT/cpu/sam0_common/periph/adc.c:90:9: note: each undeclared identifier is reported only once for each function it appears in
/home/ichiro/Documents/projects/RIOT/cpu/sam0_common/periph/adc.c: In function '_setup_clock':
/home/ichiro/Documents/projects/RIOT/cpu/sam0_common/periph/adc.c:113:26: error: 'ADC_PRESCALER' undeclared (first use in this function)
  113 |     ADC_DEV->CTRLB.reg = ADC_PRESCALER;
      |                          ^~~~~~~~~~~~~
/home/ichiro/Documents/projects/RIOT/cpu/sam0_common/periph/adc.c: In function '_adc_configure':
/home/ichiro/Documents/projects/RIOT/cpu/sam0_common/periph/adc.c:221:28: error: 'ADC_REF_DEFAULT' undeclared (first use in this function)
  221 |     ADC_DEV->REFCTRL.reg = ADC_REF_DEFAULT;
      |                            ^~~~~~~~~~~~~~~
/home/ichiro/Documents/projects/RIOT/cpu/sam0_common/periph/adc.c: In function 'adc_init':
/home/ichiro/Documents/projects/RIOT/cpu/sam0_common/periph/adc.c:245:17: error: 'ADC_NUMOF' undeclared (first use in this function); did you mean 'DAC_NUMOF'?
  245 |     if (line >= ADC_NUMOF) {
      |                 ^~~~~~~~~
      |                 DAC_NUMOF
/home/ichiro/Documents/projects/RIOT/cpu/sam0_common/periph/adc.c:251:15: error: 'adc_channels' undeclared (first use in this function)
  251 |     gpio_init(adc_channels[line].pin, GPIO_IN);
      |               ^~~~~~~~~~~~
/home/ichiro/Documents/projects/RIOT/cpu/sam0_common/periph/adc.c: In function 'adc_sample':
/home/ichiro/Documents/projects/RIOT/cpu/sam0_common/periph/adc.c:260:17: error: 'ADC_NUMOF' undeclared (first use in this function); did you mean 'DAC_NUMOF'?
  260 |     if (line >= ADC_NUMOF) {
      |                 ^~~~~~~~~
      |                 DAC_NUMOF
/home/ichiro/Documents/projects/RIOT/cpu/sam0_common/periph/adc.c:274:30: error: 'adc_channels' undeclared (first use in this function)
  274 |                            | adc_channels[line].muxpos
      |                              ^~~~~~~~~~~~
/home/ichiro/Documents/projects/RIOT/cpu/sam0_common/periph/adc.c:275:30: error: 'ADC_NEG_INPUT' undeclared (first use in this function)
  275 |                            | ADC_NEG_INPUT;

jeandudey, 2021-04-01:
That's because there's no configuration for the ADC on the board, you could try to use a configuration and add it to the periph_conf.h file in that board, adapting it accordingly of course. Just for testing of course.

I will look into it to see if Arduino has some pins designated for the ADC on that board

IchiroKuroki, 2021-04-01:
Arduino-zero has a close mcu - SAMD21G18(without A). I will try to compose the ADC and PWM configuration based on the configuration of this board.

IchiroKuroki, 2021-04-03:
I started with PWM and this is the configuration I got:

#define PWM_0_EN             1
#define PWM_1_EN             1

#if PWM_0_EN
static const pwm_conf_chan_t pwm_chan0_config[] = {
    // GPIO pin, MUX value, TCC channel
    { GPIO_PIN(PA, 4), GPIO_MUX_E, 0},
    { GPIO_PIN(PA, 5), GPIO_MUX_E, 1},
};
#endif
#if PWM_1_EN
static const pwm_conf_chan_t pwm_chan1_config[] = {
    /* GPIO pin, MUX value, TCC channel */
    { GPIO_PIN(PA, 10), GPIO_MUX_E,  0 },
    { GPIO_PIN(PA, 11), GPIO_MUX_E,  1 },
};
#endif

static const pwm_conf_t pwm_config[] = {
#if PWM_0_EN
    {TCC_CONFIG(TCC0), pwm_chan0_config, ARRAY_SIZE(pwm_chan0_config), SAM0_GCLK_MAIN},
#endif
#if PWM_1_EN
    {TCC_CONFIG(TCC1), pwm_chan1_config, ARRAY_SIZE(pwm_chan1_config), SAM0_GCLK_MAIN},
#endif
}

I tested each pin with the stated PWM support using an oscilloscope. The following pins with declared PWM support are not present in this configuration:
PA02 - This pin is also used for DAC.
PB09 - This pin is also used for I2C.
PA16, PA19 - This pin is also used for SPI.
PA20, PA21, PB10, PB11 - PWM does not work on this pins(for my board).

I'll be working on ADC soon.

IchiroKuroki, 2021-04-04:
For ADC, I got the following configuration:

#define ADC_PRESCALER                       ADC_CTRLB_PRESCALER_DIV512

#define ADC_NEG_INPUT                       ADC_INPUTCTRL_MUXNEG_GND
#define ADC_GAIN_FACTOR_DEFAULT             ADC_INPUTCTRL_GAIN_1X
#define ADC_REF_DEFAULT                     ADC_REFCTRL_REFSEL_INT1V

static const adc_conf_chan_t adc_channels[] = {
    /* port, pin, muxpos */
    {GPIO_PIN(PA, 2), ADC_INPUTCTRL_MUXPOS_PIN0},    
    {GPIO_PIN(PB, 2), ADC_INPUTCTRL_MUXPOS_PIN10},
    {GPIO_PIN(PA, 11), ADC_INPUTCTRL_MUXPOS_PIN19},
    {GPIO_PIN(PA, 10), ADC_INPUTCTRL_MUXPOS_PIN18},
    {GPIO_PIN(PB, 8), ADC_INPUTCTRL_MUXPOS_PIN2},
    {GPIO_PIN(PB, 9), ADC_INPUTCTRL_MUXPOS_PIN3},
    {GPIO_PIN(PA, 9), ADC_INPUTCTRL_MUXPOS_PIN17},
    {GPIO_PIN(PB, 3), ADC_INPUTCTRL_MUXPOS_PIN11},
};

But on my board, the ADC pin starts working only if it is also described in the configuration for PWM and the pwm_init function was called. I managed to find out that the execution of the following code makes the ADC pin start working:

file: cpu/sam0_common/periph/pwm.c
function: poweron
code:
GCLK->CLKCTRL.reg = GCLK_CLKCTRL_CLKEN
                      | GCLK_CLKCTRL_GEN(cfg->gclk_src)
                      | GCLK_CLKCTRL_ID(cfg->tim.gclk_id);
PM->APBCMASK.reg |= cfg->tim.pm_mask;

benpicco, 2021-04-09:
If you have found a working configuration for `arduino-nano-33-iot`, feel free to open a PR.

PR #16301 (2021-04-20): Configuration extension for arduino-nano-33-iot

### Contribution description
Added missing configuration for ADC / PWM / SPI

### Testing procedure
Tested with:
tests/periph_adc
tests/periph_spi
tests/periph_pwm

### Issues/PRs references
Discussed in #16247 and #16267
Closes #16267
closed 2021-04-20 by fjmolinas