RIOT-OS/RIOT #16281: esp8266: Hangs when erasing spi sector on mtd0 if using esp_wifi

iosabi, 2021-04-05:
#### Description

esp8266 hangs when the esp_wifi module is built in and one attempts to format a spiffs on mtd0.  

I'm using the ESP12x board (in my case the [Adafruit Huzzah breakout board](https://www.adafruit.com/product/2471) without anything connected to it other than the UART).

#### Steps to reproduce the issue

1. Download `xtensa-esp8266-elf-gcc` and make it available in your PATH
2. Download [ESP RTOS SDK](https://github.com/gschorcht/RIOT-Xtensa-ESP8266-RTOS-SDK.git) to `/opt/esp/ESP8266-RTOS-SDK`
3. Compile the following example:

4. Connect with your favorite terminal. Check that the board is working (type `help` for example).
5. Run `format` (if it runs ok then rung `format` again).

#### Expected results

#### Actual results

#### Versions

iosabi, 2021-04-05:
Maybe @gschorcht has some info.

#### Debugging
I enabled `ENABLE_DEBUG` in `cpu/esp_common/periph/flash.c` and I can see that the device hangs in the erase function. Adding a few printf I see that it always hangs inside `spi_flash_erase_sector`.

My guess is that this is related to a WiFi-related ISR triggering while we are waiting for the flash to erase a block. Erasing a block happens in the ROM function `SPI_sector_erase`, which is inside a block of `Cache_Read_Disable_2` and `Cache_Read_Enable_2` when called from `spi_flash_erase_sector`.

I listed the ISR handlers in the `_xt_interrupt_table` and handlers 8 (ETS_WDT_INUM, registered to `esp_task_wdt_isr`) and 10 (registered to ets_timer_handler_dsr) are the only ones in 0x402xxxxx address range (IROM - flash), while others are in IRAM. I don't know if this makes a difference.

I replaced ISR 10 with a wrapper in IRAM that raises a GPIO, then calls the original handler and then lowers the GPIO after returning from the handler. I can see ISR 10 firing every 100ms in the esp_wifi case. This doesn't occur at all when esp_wifi module is not compiled (the ISR 10 is also not registered to anything). 

My guess would be that the CPU is trying to execute something from flash (IROM) but the flash is disabled / not mapped or something like that which causes the CPU to stall, and my guess would be that this code is the ISR 10; however I don't see the GPIO high when the CPU hangs which would indicate that's not inside the ISR 10.

Any other idea of where could it be hanging? I couldn't get the JTAG debugger to work so I don't know what's the CPU doing before WDT triggers. I'll try to debug a bit more but I'm out of ideas.

iosabi, 2021-10-30:
Can't repro anymore after that patch. Sounds like it fixed it. I see 1332 bytes more in IRAM with that patch in the examples/filesystem which is pretty reasonable. Thanks for getting to the bottom of this!

I think we need to have a way to produce a more obvious crash when trying to run code from flash if the cache is disabled. This will likely happen again. Maybe a `break` instruction and a comment next to it with information about how to find out what flash address caused the fault to quickly know which function should be moved to IRAM.

gschorcht, 2021-10-30:
> Thanks for getting to the bottom of this!

I must thank you for finding this problem and investigating it. Perhaps it was also the cause of other instabilities, especially when using `esp_wifi`.

> Maybe a `break` instruction and a comment next to it with information about how to find out what flash address caused the fault to quickly know which function should be moved to IRAM.

What `break` instruction should be used and especially where should it be placed. If understand you correctly, the `break` instruction should be executed each time a function in IROM is called while the cache is disabled.

iosabi, 2021-10-30:
Any `break` instruction works. xtensa ISA doc says that `break 8, x` to `break 15, x` are all user-defined.

I believe that if you try to execute from IROM when the cache is disabled you get an exception, probably NMI but haven't check, so we should add code that checks this condition and panics with a reasonable message or breaks in the debugger.

iosabi, 2021-10-31:
@gschorcht so, there are quite a few problems going on here.

1. Nothing happens when accessing flash memory (read from 0x402xxxxx) when the flash cache is disabled. Nothing. You just read 0.
2. `00 00 00` happens to be the opcode for `ill`, the illegal instruction so trying to execute from flash with the cache disable at least produces an `IllegalInstructionCause` exception. The PC will point to the `ill` instruction and there's no problem fetching it.
3. `exception_handler()` is in IRAM, but calls functions in flash, in particular `ps()` and `heap_stats()`. In theory it should also call `putchar()` which is also in flash, but `ets_printf()` doesn't ever call it (see 5).
4. `ets_printf()`, the .text function code is in IRAM (See https://github.com/RIOT-OS/RIOT/blob/9256970517798bd844f4ee1dc4cddd9f61cef6e1/cpu/esp8266/ld/esp8266.riot-os.ld#L236 ) but, the `xtensa-esp8266-elf` toolchain places the literal strings in `.rodata`, not in `.literal`, so `ets_printf` doesn't actually work with the cache disabled because all its const data is not placed in IRAM. It is probably useful to have that function in RAM anyway.
5. Going one step back, `exception_handler()` is tagged with IRAM in the source code (not mentioned in the linker script), but the string literals it uses don't end up in IRAM, those are in `.rodata` which according to the .map file end up in flash, and that includes the format strings we pass to ets_printf(). Reading those strings from flash is not a problem (not an exception), it just reads 0s which happens to be a valid empty C string and ets_printf() happily prints it without ever calling `putchar`, which is why we get no output on the serial but also no crash in putchar().
6. Even after the specific fix for `vTaskEnterCritical`, causing any exception while the flash cache is disabled would trigger the infinite exception.

gschorcht, 2021-10-31:
> 2\. `IllegalInstructionCause` exception

Yes, that is exactly what I saw when the cache was disabled when calling a function in IROM, but after not working on the ESP8266 port for at least 2 years, I forgot.

> 3\. `exception_handler()` is in IRAM, but calls functions in flash, in particular `ps()` and `heap_stats()`. In theory it should also call `putchar()` which is also in flash, but `ets_printf()` doesn't ever call it (see 5).

That is, they would have to be placed in IRAM too.

> 4\. `xtensa-esp8266-elf` toolchain places the literal strings in `.rodata`, not in `.literal`, so `ets_printf` doesn't actually work with the cache

Ah, thats why the original linker script includes `.rodata` for `ets_printf.o`:
```
#ifdef CONFIG_LINK_ETS_PRINTF_TO_IRAM
    *libesp8266.a:ets_printf.o(.literal .text .literal.* .text.* .rodata.* .rodata)
#endif
```
Finally, we would have to extend `esp8266.riot-os.ld` in IRAM section by:
```diff
diff --git a/cpu/esp8266/ld/esp8266.riot-os.ld b/cpu/esp8266/ld/esp8266.riot-os.ld
index e99aa855fe..06df3af98d 100644
--- a/cpu/esp8266/ld/esp8266.riot-os.ld
+++ b/cpu/esp8266/ld/esp8266.riot-os.ld
@@ -233,7 +233,7 @@ SECTIONS
     /* SDK libraries that expect their .text or .data sections to link to iram */
     /* TODO *libcore.a:(.bss .data .bss.* .data.* COMMON) */
     *esp_idf_spi_flash/spi_flash_raw.o(.literal .text .literal.* .text.*)
-    *esp_idf_esp8266/ets_printf.o(.literal .text .literal.* .text.*)
+    *esp_idf_esp8266/ets_printf.o(.literal .text .literal.* .text.* .rodata.* .rodata)
     /*
     *cpu.a:*.o(.literal .text .literal.* .text.*)
     */
@@ -241,6 +241,8 @@ SECTIONS
     *esp_wifi/*(.literal .text .literal.* .text.*)
     *freertos/*(.literal .text .literal.* .text.*)
     *periph/*(.literal .text .literal.* .text.*)
+    *ps/*(.literal .text .literal.* .text.* .rodata.* .rodata)
+    *newlib_syscalls_default/*(.literal .text .literal.* .text.* .rodata.* .rodata)
     *xtimer/*(.literal .text .literal.* .text.*)
 
     *libhal.a:clock.o(.literal .text .literal.* .text.*)
@@ -287,6 +289,7 @@ SECTIONS
     *libc.a:*findfp.o(.literal .text .literal.* .text.*)
     *libc.a:*fputwc.o(.literal .text .literal.* .text.*)
     */
+    *libg.a:*putchar.o(.literal .text .literal.* .text.*)
 
     *enc28j60/*(.literal .text .literal.* .text.*)
```

gschorcht, 2021-10-31:
I would prefer to place everything in IRAM. But with 48 kByte it is so small that we have to limit what goes into the IRAM. It always requires a bit of trial and error what absolutely has to go into the IRAM.

iosabi, 2021-10-31:
Why do we have both printf() in newlib and ets_printf()?

gschorcht, 2021-10-31:
If I remember correctly, there were the following reasons:
1. The SDK redefines `ets_printf` which is usually a ROM function for libraries that require `ets_printf`. [`ets_printf` source code in SDK](https://github.com/RIOT-OS/RIOT/blob/9256970517798bd844f4ee1dc4cddd9f61cef6e1/cpu/esp8266/vendor/esp-idf/esp8266/source/ets_printf.c#L298-L300) gives some explanation. We just use a slightly adapted version.
2. `ets_printf` is the only `printf` function that should work even if the IROM cache is disabled, which is not the case at the moment, as we learned.
3. `ets_printf` is the only `printf` function that works before the IROM cache is enabled during startup.
4. `ets_printf` was intended to work also inside interrupt context. The `newlib` in `xtensa-gcc` uses the reentrant version of `printf` which uses lock functions which in turn use mutexes. RIOT's mutex, however, doesn't work in interrupt context.