RIOT-OS/RIOT #16282: SIGFPE on native architecture when printing double floats on Ubuntu 21.04

gdoffe, 2021-04-06:
Hi all,

#### Description
Cannot printf double floats on native architecture for Ubuntu 20.10 and later.
The build is ok but at execution SIGFPE signal is received.
This bug does not occur on Ubuntu 20.04

#### Steps to reproduce the issue
Apply this patch to just print a double in `version` command:
```diff
diff --git a/sys/shell/commands/sc_sys.c b/sys/shell/commands/sc_sys.c
index a79df983f4..595e235bb5 100644
--- a/sys/shell/commands/sc_sys.c
+++ b/sys/shell/commands/sc_sys.c
@@ -54,6 +54,7 @@ int _version_handler(int argc, char **argv)
     (void) argv;
 
     puts(RIOT_VERSION);
+    printf("%lf\n", 0.25);
 
     return 0;
 }
 ```
Then build and execute the application:
```bash
$ make -C tests/ps_schedstatistics/ && tests/ps_schedstatistics/bin/native/tests_ps_schedstatistics.elf
```

#### Expected results
Expected result is available when building on Ubuntu 20.04:
```shell
$ make -C tests/ps_schedstatistics/ && tests/ps_schedstatistics/bin/native/tests_ps_schedstatistics.elf
make : on entre dans le répertoire « /home/gdo/Developpement/Informatique/Personnel/cogip/RIOT/tests/ps_schedstatistics »
Building application "tests_ps_schedstatistics" for "native" with MCU "native".

"make" -C /home/gdo/Developpement/Informatique/Personnel/cogip/RIOT/boards/native
"make" -C /home/gdo/Developpement/Informatique/Personnel/cogip/RIOT/boards/native/drivers
"make" -C /home/gdo/Developpement/Informatique/Personnel/cogip/RIOT/core
"make" -C /home/gdo/Developpement/Informatique/Personnel/cogip/RIOT/cpu/native
"make" -C /home/gdo/Developpement/Informatique/Personnel/cogip/RIOT/cpu/native/periph
"make" -C /home/gdo/Developpement/Informatique/Personnel/cogip/RIOT/cpu/native/stdio_native
"make" -C /home/gdo/Developpement/Informatique/Personnel/cogip/RIOT/drivers
"make" -C /home/gdo/Developpement/Informatique/Personnel/cogip/RIOT/drivers/periph_common
"make" -C /home/gdo/Developpement/Informatique/Personnel/cogip/RIOT/sys
"make" -C /home/gdo/Developpement/Informatique/Personnel/cogip/RIOT/sys/auto_init
"make" -C /home/gdo/Developpement/Informatique/Personnel/cogip/RIOT/sys/div
"make" -C /home/gdo/Developpement/Informatique/Personnel/cogip/RIOT/sys/ps
"make" -C /home/gdo/Developpement/Informatique/Personnel/cogip/RIOT/sys/schedstatistics
"make" -C /home/gdo/Developpement/Informatique/Personnel/cogip/RIOT/sys/shell
"make" -C /home/gdo/Developpement/Informatique/Personnel/cogip/RIOT/sys/shell/commands
"make" -C /home/gdo/Developpement/Informatique/Personnel/cogip/RIOT/sys/test_utils/interactive_sync
"make" -C /home/gdo/Developpement/Informatique/Personnel/cogip/RIOT/sys/xtimer
   text	   data	    bss	    dec	    hex	filename
  47656	    736	  89388	 137780	  21a34	/home/gdo/Developpement/Informatique/Personnel/cogip/RIOT/tests/ps_schedstatistics/bin/native/tests_ps_schedstatistics.elf
make : on quitte le répertoire « /home/gdo/Developpement/Informatique/Personnel/cogip/RIOT/tests/ps_schedstatistics »
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2021.01)
Help: Press s to start test, r to print it is ready
s
START
Creating thread #0, next=1
Creating thread #1, next=2
Creating thread #2, next=3
Creating thread #3, next=4
Creating thread #4, next=0
[... 280 lines cut ...]
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: missing (missing)

Installed development tools
---------------------------
                   ccache: missing
                    cmake: missing
                 cppcheck: missing
                  doxygen: missing
                      git: git version 2.30.2
                     make: GNU Make 4.3
                  openocd: missing
                   python: missing
                  python2: missing
                  python3: Python 3.9.2
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing
```

Feel free to ask any test or informations about this.

gdoffe, 2021-04-07:
I found some related issues:

- https://github.com/RIOT-OS/RIOT/issues/15870
- https://github.com/RIOT-OS/RIOT/issues/495
- https://github.com/RIOT-OS/RIOT/issues/15878

I try to find a workaround to avoid reinstalling my computer with Ubuntu 20.04.

gdoffe, 2022-07-07:
Hi friends,

I up this bug as I can still reproduce it on Ubuntu 22.04.
I am limited by my skills to investigate further on this. Any help would be appreciated.

jia200x, 2022-09-22:
Hi @gdoffe 

See https://github.com/RIOT-OS/RIOT/issues/495#issuecomment-1254664077. This seems to be the cause of this issue.

maribu, 2023-01-05:
Duplicate of https://github.com/RIOT-OS/RIOT/issues/495

closed 2023-01-05 by maribu