RIOT-OS/RIOT #16293: Conflict between semtech-loramac and hashes modules

donsez, 2021-04-08:
#### Description

Modules semtech-loramac and hashes are in conflict

I can not sha1 functions.

#### Steps to reproduce the issue

```
> cd github/RIOT-OS/RIOT/tests/pkg_semtech-loramac
> USEMODULE=hashes BOARD=p-nucleo-wb55 gmake
Building application "tests_pkg_semtech-loramac" for "p-nucleo-wb55" with MCU "stm32".

"gmake" -C /Users/dd/github/RIOT-OS/RIOT/pkg/semtech-loramac
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/build/pkg/semtech-loramac/src/boards/mcu -f /Users/dd/github/RIOT-OS/RIOT/pkg/semtech-loramac/Makefile.loramac_arch
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/build/pkg/semtech-loramac/src/system/crypto -f /Users/dd/github/RIOT-OS/RIOT/pkg/semtech-loramac/Makefile.loramac_crypto
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/build/pkg/semtech-loramac/src/mac -f /Users/dd/github/RIOT-OS/RIOT/pkg/semtech-loramac/Makefile.loramac_mac
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/build/pkg/semtech-loramac/src/mac/region -f /Users/dd/github/RIOT-OS/RIOT/pkg/semtech-loramac/Makefile.loramac_region
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/boards/p-nucleo-wb55
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/boards/common/nucleo
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/core
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/cpu/stm32
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/cpu/cortexm_common
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/cpu/cortexm_common/periph
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/cpu/stm32/periph
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/cpu/stm32/stmclk
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/cpu/stm32/vectors
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/drivers
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/drivers/periph_common
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/drivers/sx127x
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/pkg/semtech-loramac/contrib
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/sys
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/sys/auto_init
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/sys/auto_init/loramac
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/sys/crypto
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/sys/fmt
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/sys/frac
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/sys/hashes
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/sys/iolist
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/sys/isrpipe
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/sys/luid
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/sys/malloc_thread_safe
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/sys/newlib_syscalls_default
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/sys/pm_layered
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/sys/random
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/sys/random/tinymt32
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/sys/shell
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/sys/shell/commands
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/sys/stdio_uart
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/sys/test_utils/interactive_sync
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/sys/tsrb
"gmake" -C /Users/dd/github/RIOT-OS/RIOT/sys/ztimer
/usr/local/Cellar/arm-none-eabi-gcc/8-2018-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld: /Users/dd/github/RIOT-OS/RIOT/tests/pkg_semtech-loramac/bin/p-nucleo-wb55/semtech_loramac_crypto//aes.o: in function `aes_encrypt':
/Users/dd/github/RIOT-OS/RIOT/build/pkg/semtech-loramac/src/system/crypto/aes.c:569: multiple definition of `aes_encrypt'; /Users/dd/github/RIOT-OS/RIOT/tests/pkg_semtech-loramac/bin/p-nucleo-wb55/crypto//aes.o:/Users/dd/github/RIOT-OS/RIOT/sys/crypto/aes.c:1032: first defined here
collect2: error: ld returned 1 exit status
gmake: *** [/Users/dd/github/RIOT-OS/RIOT/tests/pkg_semtech-loramac/../../Makefile.include:634: /Users/dd/github/RIOT-OS/RIOT/tests/pkg_semtech-loramac/bin/p-nucleo-wb55/tests_pkg_semtech-loramac.elf] Error 1
```

#### Expected results

#### Actual results

#### Versions
```
Operating System Environment
----------------------------
         Operating System: Mac OS X 10.15.7
                   Kernel: Darwin 19.6.0 x86_64 i386
./dist/tools/ci/print_toolchain_versions.sh: line 78: realpath: command not found
             System shell:  (probably dash)
             make's shell: GNU bash, version 3.2.57(1)-release (x86_64-apple-darwin19)

Installed compiler toolchains
-----------------------------
               native gcc: Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/c++/4.2.1
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.1 20181213 (release) [gcc-8-branch revision 267074]
                  avr-gcc: missing
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: Apple clang version 12.0.0 (clang-1200.0.32.28)

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.0.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: missing (missing)

Installed development tools
---------------------------
                   ccache: ccache version 4.0
                    cmake: cmake version 3.19.2
                 cppcheck: missing
                  doxygen: missing
                      git: git version 2.24.3 (Apple Git-128)
                     make: GNU Make 3.81
                  openocd: Open On-Chip Debugger 0.10.0
                   python: Python 2.7.16
                  python2: Python 2.7.16
                  python3: Python 3.8.2
                   flake8: error: /Applications/Xcode.app/Contents/Developer/usr/bin/python3: No module named flake8
               coccinelle: missing
```

fjmolinas, 2021-04-09:
I think this can be fixed by providing our own implementation of the LoraMACCrypto module, only users of aes, this was the original approach taken by in #6645, see https://github.com/RIOT-OS/RIOT/blob/0d24ab1c12a3ccb849c39da589df533a4f01bb0b/sys/net/gnrc/link_layer/gnrc_lorawan/gnrc_lorawan_crypto.c. @jia200x do you know why this was not the case in the final pkg?

jia200x, 2021-04-09:
> I think this can be fixed by providing our own implementation of the LoraMACCrypto module, only users of aes, this was the original approach taken by in #6645, see [...]

Back then during the early adoption of the pkg (#6645) there was a custom LoRaMAC Crypto, but the ifnal pkg just took the reference implementation. I completely agree that we should write a custom implementation.
In fact, we can unify GNRC LoRaWAN and LoRaMAC cryto (e.g, we can define macros or static inline function that point to these "common" LoRaMAC crypto).
What do you think?

fjmolinas, 2021-04-09:
> Back then during the early adoption of the pkg (#6645) there was a custom LoRaMAC Crypto, but the ifnal pkg just took the reference implementation. I completely agree that we should write a custom [...]

Yes that sound like the best course!

aabadie, 2021-04-09:
Note that this problem was addressed by #10124.

jia200x, 2021-04-09:
> Note that this problem was addressed by #10124.

We could reactivate that

jia200x, 2021-04-12:
It's already solved in master. Thanks for reporting!

closed 2021-04-12 by jia200x