RIOT-OS/RIOT #16318: Can GNRC TCP run over BLE?

lvjiamei, 2021-04-12:
#### Description
I'm trying to run GNRC TCP on BLE with two nrf52840 boards. But these two boards are failed in estabilishing the connection. The related logs is "ipv6: no link-layer address or interface for next hop to fe80::xxxx:xxxx:xxxx:xxxx".

#### Steps to reproduce the issue
I reuse the code in /tests/gnrc_tcp. The steps are as follows:
1. <BOARD=nrf52840dk USEMODULE=nimble_netif make>
2. Flash the .bin file to the boards.
3. Connect two boards by <ble adv> and <ble connect E3:F6:47:6B:XX:XX 80>
4. On the board A:
  a. Creat an end point： <gnrc_tcp_ep_from_str [FE80::E3F6:XXXX:XXXX:XXXX]:80>
  b. Intial a TCP structure: <gnrc_tcp_tcb_init>
  c. Start a TCP connection: <gnrc_tcp_open_active [FE80::E3F6:XXXX:XXXX:XXXX]:80>
There is no packet sent on radio when I tried to open a TCP connection. The log is " "ipv6: no link-layer address or interface for next hop to fe80::xxxx:xxxx:xxxx:xxxx".

miri64, 2021-04-13:
Did you use the interface identifier with the link-local address? Most of the time you need that as link local addresses are local to the link, so the OS does not know which interface to use on its own (you will see the same behavior when trying to use link-local addresses e.g. on Linux in that manner). E.g. with RIOT `fe80::xxxx:xxxx:xxxx:xxxx%5` to send over interface 5, with Linux you use `fe80::xxxx:xxxx:xxxx:xxxx%eth0` to send over interface `eth0`.

miri64, 2021-04-13:
In case this does not solve the problem, I assigned both the authors of BLE and TCP respectively. Maybe they have an idea.

brummer-simon, 2021-04-13:
From a TCP point of view, the underlaying network technology is not important, the network layer handles everything below GNRC_TCP. As @miri64 mentioned above the missing device identifier propably causes the issue.

lvjiamei, 2021-04-15:
> Did you use the interface identifier with the link-local address? Most of the time you need that as link local addresses are local to the link, so the OS does not know which interface to use on its [...]

Thank you for your reminding! I solve the problem exactly!

closed 2021-04-15 by lvjiamei