RIOT-OS/RIOT #16320: Does RIOT's xtimer accurately wake up threads?

liyue75, 2021-04-13:
#### Description

Sensors send data to my computer. 
These sensors use stm32f4discovery MCU, and running RIOT-OS. I use ipv6 to send data and netwrok stack is GNRC. In main loop, I use `xtimer_usleep(20000)` to wake up the main thread every 20ms and send data.
On the computer, I run a program to receive data, but I found that the receive data interval was not always 20ms, 
[image]
The figure above shows the receive data interval for the past 10 minutes, with some data intervals greater than 25ms, sometimes even greater than 30ms.

Do you think that these interval is caused by  `xtimer_usleep`, in other words, will the error of `xtimer_usleep` greater than 5ms?

liyue75, 2021-04-13:
I find it may related to the computer side network protocol stack. My host system is Ubuntu18.04. In my receive program, I use`recvfrom` to accept data from stm32.

jeandudey, 2021-04-13:
Have you tried to use [`xtimer_periodic_wakeup`](https://doc.riot-os.org/group__sys__xtimer.html#ga77ff235d491f651378348a46519d9dba) at the end of your main loop? It will account for the difference in time when prior processing waking up the thread at the correct time

liyue75, 2021-04-14:
@jeandudey Yes, I have tried`xtimer_periodic_wakeup`, but it's about the same.
Xtimer should not cause such a large error, the jitter should be caused by the Linux network protocol stack on computer. Although the data is sent to the computer, it does not seem to be processed by the receiver immediately.

jeandudey, 2021-04-14:
Then it may be caused by the linux network stack and the latency of the connection too, and the processing time GNRC uses before sending a packet, it all accounts. To see if xtimer is the problem you could try to send the packets with a timestamp, the difference in the timestamp between each packet should show the interval of 20 ms

liyue75, 2021-04-14:
@jeandudey Thanks a lot.

closed 2021-04-14 by liyue75