RIOT-OS/RIOT #16336: No output in pyterm after flashing to STM32F0discovery board

hoekstraarie, 2021-04-15:
Hello everyone,

I'm trying to get started with RIOT-OS, I followed the tutorials and can compile and run my code on my native system. But when I try to flash my code on my board I don't get any output on my terminal. 

My OS can find the USB so that is not the issue, it just doesn't give any output. Though I expect hello world to be printed. 

I compile and flash my code on a virtualbox with Ubuntu and my host is a Windows 10 machine. 

[image]

[image]

I got some pictures attached with my commands and the output.

jeandudey, 2021-04-21:
Hello @hoekstraarie !

Have you added your user to the `dialout` group in your OS? This might be necessary to access serial ports with your user. If thats the case and still no output is shown, resetting the board while the serial terminal is open shows anything?

Another thing to do is to check if it's flashing process is working and the board works, you might want to flash the `tests/leds` test to see if the on board leds blink. E.g.:

- `BOARD=stm32f0discovery make -C tests/leds flash reset`

hoekstraarie, 2021-04-22:
Hello @jeandudey,

Thanks for your comment on my issue. Fortunately I solved the problem, there was a problem with the USART breakoutboard. 

Have a great day!

jeandudey, 2021-04-22:
Thanks for the response @hoekstraarie !

Closing this now since it was solved

closed 2021-04-22 by jeandudey