RIOT-OS/RIOT #16355: Can't add a static library to RIOT

neuromancer-valis, 2021-04-20:
#### Description

Hey guys, I can't add a static library (cross-compiled) `samplelib.a` to my project via the `USEMODULE += samplelib.a` field.

#### Steps to reproduce the issue

#### Expected results

#### Actual results

Instead I get the following warning message:

`error: ISO C99 requires whitespace after the macro name
#define MODULE_LIBEXAMPLE.A 1
`

#### Versions

kaspar030, 2021-04-20:
> `USEMODULE += libexample.a` field.

try `USEMODULE += libexample`(without `.a`) instead.

kaspar030, 2021-04-21:
@neuromancer-valis I'll close this, as I'm certain this solved your issue. Please reopen if you think otherwise!

closed 2021-04-21 by kaspar030

aabadie, 2021-04-21:
Maybe have a look at the [bindist example](https://github.com/RIOT-OS/RIOT/tree/master/examples/bindist). It seems that you can't use `USEMODULE` with archive but rather have to use `BIN_DIRS` and `BIN_USEMODULE` variables:

https://github.com/RIOT-OS/RIOT/blob/855a6fa36a9f807227121ca3fc54a1927317079c/examples/bindist/Makefile#L20-L26

Let us know if that solves your problem.

aabadie, 2021-04-21:
It seems that another possibility to a `.a` file to the build is to use the `ARCHIVES` variable: https://github.com/RIOT-OS/RIOT/search?q=ARCHIVES

ubik-valis, 2021-09-14:
Thank you all for your answers! I followed the bindist example, but still have include errors. Adding the include files with `INCLUDES` throws an error that no such file or directory exists.