RIOT-OS/RIOT #16369: Unsatisfied features RIOT-OS stm32f0discovery

hoekstraarie, 2021-04-22:
#### Description
Get an error when building lorawan and mpu9x50 examples. With stm32f0discovery board.

#### Steps to reproduce the issue
My studentgroup and I want to setup a Lorawan network, so we can send data from a mpu accelerometer. We ordered a sx1276 lora module. 

[lora module](https://www.hoperf.com/modules/lora/RFM95.html)
[mpu module](https://www.electronicwings.com/sensors-modules/mpu6050-gyroscope-accelerometer-temperature-sensor-module)

We tested the lora modules on an arduino and we can send data to our gateway. We can read the data on the console of the things network, both in our application and in our gateway. 

Now we tought it was a good step to test our lora module on a stm32f0discovery board with RIOT-OS, because of the hardware independent operating system. Beneath you can find our connections from our stm board to the modules.

LORA MODULE

MISO  -> PA6
MOSI  -> PA7
SCK    -> PA5
3v3     -> 3v3
GND   -> GND

MPU MODULE

SCL    -> PB10
SDA   -> PB11
3V3    -> 3V3
GND  -> GND

Now if we try to flash the lora example, we get an error with **rtt unsatisfied features**,
also if we try to flash the mpu example we get the error **i2c unsatisfied features**.

We know there is a problem in the global_conf.json file because i2c is not initialized here. So it would be nice to know how to initialize rtt and i2c for example.

The test of the sx127x driver works, but we can't get any input or output out of the terminal.

Thanks in advance!

#### Actual results
Error unsatisfied features 
[image]

#### Versions
Operating system: UBUNTU on a virtualbox with Windows 10 as host
Build environment: ARM toolchain

jeandudey, 2021-04-22:
Hello @hoekstraarie !

Seems that the configuration for the `stm32f0discovery` for the I2C and RTT peripherals are missing, for the I2C you should be able to add the missing configuration, e.g.: by looking for another board with similar configurations and changing it on the `stm32f0discovery`.  STM32 boards define common include files for the I2C configurations, [for example](https://github.com/RIOT-OS/RIOT/blob/master/boards/common/stm32/include/cfg_i2c1_pb6_pb7.h) and they are included on the respective `periph_conf.h` file of the board, and the `periph_i2c` features are added to the `Makefile.features` and `Kconfig` files of the board.

I don't know much about the pins of the `stm32f0discovery` board, but if there are pins defined/meant to be used for the I2C you could add them and contribute the patches to RIOT 😉 .

For the RTT the story is similar, configuration needs to be added and tested for the board.

hoekstraarie, 2021-04-23:
Hello @jeandudey ,

Thanks for your fast response, we got the i2c part working now. We will commit this change to the RIOT-OS git, so everyone can enjoy the working of i2c on the stm32f0discovery. Where can we commit this change?

However the rtt part is still a bit vague, do you happen to have an example for this config. We cannot find any boards with this kind of configuration. Or is it just a hardware problem? The datasheet of the stm32f0discovery board tells us that there is a RTC clock but no RTT clock. 

Hope you can help us further with the next problem.

Thanks in advance. 👍

jeandudey, 2021-04-23:
For the RTT there is a common configuration too, the [cfg_rtt_default.h](https://github.com/RIOT-OS/RIOT/blob/master/boards/common/stm32/include/cfg_rtt_default.h) file which can be included on the `periph_conf.h` file of the `stm32f0discovery` just like the I2C, remember to add the `FEATURES_PROVIDED += periph_rtt` on the `Makefile.features` and the respective `select HAS_PERIPH_RTT` on the board `Kconfig` file.

>However the rtt part is still a bit vague, do you happen to have an example for this config. We cannot find any boards with this kind of configuration. Or is it just a hardware problem? The datasheet of the stm32f0discovery board tells us that there is a RTC clock but no RTT clock.

The rtt is a RIOT peripheral abstraction for real time timers, on STM32s RIOT uses the LPTIM1 hardware timer peripheral so it should be available for most/all STM32s.

>Thanks for your fast response, we got the i2c part working now. We will commit this change to the RIOT-OS git, so everyone can enjoy the working of i2c on the stm32f0discovery. Where can we commit this change?

You can fork the project, commit and push the changes to your branch and make a PR. See the [CODING_CONVENTIONS.md](https://github.com/RIOT-OS/RIOT/blob/master/CODING_CONVENTIONS.md) and [CONTRIBUTING.md](https://github.com/RIOT-OS/RIOT/blob/master/CONTRIBUTING.md) for more information about this :wink:. Thanks for contributing!

hoekstraarie, 2021-04-23:
Thanks @jeandudey ,

I tried this and it is compiling, only I get some errors. 

The periph_conf.h of the stm32f0discovery board:
[image]

The Kconfig file:
[image]

The Makefile.features file:
[image]

The errors I get:
[image]

It looks like there is something not included yet. Do you know which file that is?

Thanks for your good support.

jeandudey, 2021-04-23:
Can you do a `make clean` and test again? Also:

[image]

Check the `FEATURES_PROVIDE += periph_i2c` 👀

hoekstraarie, 2021-04-23:
Excuse me didn't see that mistake 💯 . Still the same erros, even after a make clean. I try to run the sx1276 test with the lora module that is connected to my board. Still the same errors.

jeandudey, 2021-04-23:
It seems that the stm32f0 model used by that board doesn't provides an RTT so that's the reason it wont work :-/, the patches with the I2C can be PRed though.

hoekstraarie, 2021-04-23:
Hello @jeandudey ,

So this means I can’t use my sx1276 Lora module with my stm32f0discovery board?

We will contribute the I2c part of my project to riot.

Have a great day.

jeandudey, 2021-04-23:
>So this means I can’t use my sx1276 Lora module with my stm32f0discovery board? We will try to test it without the rtt peripheral after the weekend.

You should be able to use it if RTT is not present, the dependency that uses RTT is ztimer for msec/usec backends, but there is an implementation for general purpose timers using periph_timer, so in theory it should work without RTT

hoekstraarie, 2021-04-26:
Hi @jeandudey ,

So that should say I can't run the examples from RIOT on my board, because they all use this RTT. Or are there examples that use the general purpose timers?

fjmolinas, 2021-04-26:
Hi @hoekstraarie let me add some more context here @aabadie who is maintaining the `semtech_loramac` pkg decided that `ztimer_periph_rtt` would be the required backend for `ztimer_msec`, which means that `periph_rtt` is required. But if I'm not mistaken the `stm32f0discovery` `BOARD` does not have a LSE (low speed external clock) to use a source for the RTT, which is why the RTT is not provided. There is some reasoning on why this was done in the documentation see "Low power considerations" in [semtec-loramac pkg doc](https://doc.riot-os.org/group__pkg__semtech-loramac.html), also the original PR discussion https://github.com/RIOT-OS/RIOT/pull/14547#discussion_r470018566.

What you can do is simply apply this patch to remove that hard requirement.

```diff
diff --git a/pkg/semtech-loramac/Makefile.dep b/pkg/semtech-loramac/Makefile.dep
index 16eb716643..fa732e521e 100644
--- a/pkg/semtech-loramac/Makefile.dep
+++ b/pkg/semtech-loramac/Makefile.dep
@@ -9,7 +9,6 @@ USEMODULE += semtech_loramac_crypto
 USEMODULE += semtech_loramac_arch
 
 USEMODULE += ztimer_msec
-USEMODULE += ztimer_periph_rtt
 
 # The build fails on MSP430 because the toolchain doesn't provide
 # EXIT_SUCCESS/EXIT_FAILURE macros
```

But with this do not expect for the pkg to work with sleep modes.

hoekstraarie, 2021-04-26:
Hi @fjmolinas ,

We applied the patch and it's flashing to the board now. Finally 👍 . 

But in the pyterm we can't get any output in the terminal. So that's our next objective for the coming week.

It isn't giving any errors back, and it is running. We can see that it is stuck at the loramac stack init. 

Thanks for the help.

closed 2021-04-28 by hoekstraarie

jeandudey, 2021-04-28:
@hoekstraarie did the loramac issue got solved?

fjmolinas, 2021-04-28:
> Hi @fjmolinas ,
> 
> We applied the patch and it's flashing to the board now. Finally +1 .
> 
> But in the pyterm we can't get any output in the terminal. So that's our next objective for the coming week.
> 
> It isn't giving any errors back, and it is running. We can see that it is stuck at the loramac stack init.
> 
> With debug enable = 1 we can see output that says, initializing loramac. And after that it is stuck.
> 
> Thanks for the help.

Sorry I missed your answer. Can you maybe enable debug output in the `sx1276` driver code? It might be stuck there?

Also is the chip you are using does it actually integrate a `sx1276x`? Maybe its not the exact same LoRa chip...

hoekstraarie, 2021-04-29:
Hi @jeandudey and @fjmolinas ,

I closed the issue because we got it working somehow. I opened a new issue, because those problems are not related to this issue.  Maybe you can look at my new issue thanks for your support. #16407 

Have a great day!