RIOT-OS/RIOT #16381: I2C not working under RIOT with U8G2 pkg

silklaasboer, 2021-04-23:
Description

Getting no i2c clockline and data line change. With the nucleo-l452re board. when using the u8g2 pkg included in the makefile.

Steps to reproduce the issue

My studentgroup and I want to setup a Oled display, so we can display battery info on it. sadly enough my i2c is not doing anything when analysing it with a logic analyser attachted to my computer.

I am using an SSD1306 128x32 Oled display by Univision

We first tested the display with an arduino and it all worked fine with the display. now we are trying it on the nucleo-l452re and it is not showing anything.

display I/O connections

SCL -> PB8
SDA -> PB9
3V3 -> 3V3 OR 5V -> 5V
GND -> GND

Now if i try to flash the file to the board i'll get no error's at all and its looping to the code just fine, i tested it with extra print statements to the serial monitor.

Thanks in advance!
Actual results
[image]

[image]

Operating system: UBUNTU on a docker with Windows 10 as host
Build environment: ARM toolchain

hoekstraarie, 2021-04-23:
Kijk naar mijn vraag daar is het al beantwoord. ClOsInG thIS IsSuE nOW!!!

jeandudey, 2021-04-23:
If you tried with [tests/pkg_u8g2](https://github.com/RIOT-OS/RIOT/tree/master/tests/pkg_u8g2), did you use `TEST_OUTPUT=4` to test using SSD1306 I2C?

silklaasboer, 2021-04-23:
> 
> 
> If you tried with [tests/pkg_u8g2](https://github.com/RIOT-OS/RIOT/tree/master/tests/pkg_u8g2), did you use `TEST_OUTPUT=4` to test using SSD1306 I2C?

Yes i used this configuration ;
#define TEST_OUTPUT 4
#define TEST_OUTPUT_I2C 4
#define TEST_I2C 0  //also tried I2C_DEV(0)
#define TEST_ADDR 0x3C
#define TEST_DISPLAY u8g2_Setup_ssd1306_i2c_128x32_univision_f
#define TEST_PIN_CS GPIO_PIN(PORT_B,8)
#define TEST_PIN_DC   GPIO_PIN(PORT_B,9)
#define TEST_PIN_RESET U8X8_PIN_RESET // also tried GPIO_UNDEF

jeandudey, 2021-04-23:
So it should work :-/, can you provide the output of [tests/periph_i2c](https://github.com/RIOT-OS/RIOT/tree/master/tests/periph_i2c) by sending on the terminal the command `i2c_scan 0`?

silklaasboer, 2021-04-23:
> 
> 
> So it should work :-/, can you provide the output of [tests/periph_i2c](https://github.com/RIOT-OS/RIOT/tree/master/tests/periph_i2c) by sending on the terminal the command `i2c_scan 0`?

i2c_scan 0 is not one of the commands that is recognised. if i type help these are the only commands that are known.
[image]

jeandudey, 2021-04-23:
On recent versions it should be available:

https://github.com/RIOT-OS/RIOT/blob/c0c3a76fb88b695b56cefdc844de8807baf206d6/tests/periph_i2c/Makefile#L9

So, you can try with the latest RIOT release or directly from master

silklaasboer, 2021-04-23:
> 
> 
> On recent versions it should be available:
> 
> https://github.com/RIOT-OS/RIOT/blob/c0c3a76fb88b695b56cefdc844de8807baf206d6/tests/periph_i2c/Makefile#L9
> 
> So, you can try directly from master to see if it helps

it looks something like this 
[image]

jeandudey, 2021-04-23:
Please disable `ENABLE_DEBUG` on the i2c.c file to see the correct output :eyes:

silklaasboer, 2021-04-23:
> 
> 
> Please disable `ENABLE_DEBUG` on the i2c.c file to see the correct output 👀

[image]

jeandudey, 2021-04-23:
> > Please disable `ENABLE_DEBUG` on the i2c.c file to see the correct output eyes
> 
> [image]

It seems that the display responds to I2C reads/writes, the problem is with u8g2 package or the test

silklaasboer, 2021-04-23:
> 
> 
> > > Please disable `ENABLE_DEBUG` on the i2c.c file to see the correct output eyes
> > 
> > 
> > [image]
> 
> It seems that the display responds to I2C reads/writes, the problem is with u8g2 package or the test

yes but is the u8g2 package not included within RIOT? they even wrote device specific u8x8_byte_hw_i2c_riotos and the u8x8_gpio_and_delay_riotos. dont now were to look anymore because i get no error's at all

jeandudey, 2021-04-23:
Could you try by using the following command? (and leaving the main.c file as original)

- `make BOARD=nucleo-l452re TEST_OUTPUT=4 TEST_DISPLAY=u8g2_Setup_ssd1306_i2c_128x32_noname_1`

As the `#define`s you're using may not be getting picked up since the `Makefile` adds it's own definitions.

silklaasboer, 2021-04-23:
> 
> 
> Could you try by using the following command? (and leaving the main.c file as original)
> 
>     * `make BOARD=nucleo-l452re TEST_OUTPUT=4 TEST_DISPLAY=u8g2_Setup_ssd1306_i2c_128x32_noname_1`
> 
> 
> As the `#define`s you're using may not be getting picked up since the `Makefile` adds it's own definitions.

i think you mean: TEST_DISPLAY=u8g2_Setup_ssd1305_i2c_128x32_noname_1 ?

silklaasboer, 2021-04-23:
> 
> 
> Could you try by using the following command? (and leaving the main.c file as original)
> 
>     * `make BOARD=nucleo-l452re TEST_OUTPUT=4 TEST_DISPLAY=u8g2_Setup_ssd1306_i2c_128x32_noname_1`
> 
> 
> As the `#define`s you're using may not be getting picked up since the `Makefile` adds it's own definitions.

i can compile without an error if i use : make BOARD=nucleo-l452re TEST_OUTPUT=4 TEST_DISPLAY=u8g2_Setup_ssd1305_i2c_128x32_noname_1` and the standard MakeFile and standard main.c. but still nothing happends with de display while uart is giving me feedback that it did init i2c, init display and that its supposed to be drawing on screen.

LarsGrootkarzijn, 2021-05-12:
dit is wel echt flink kut sil hopelijk heb je het al opgelost groetjes aan je zusje

maribu, 2023-05-17:
Is this issue still relevant?

If so maybe

```
make BOARD=nucleo-l452re TEST_OUTPUT=4 TEST_DISPLAY=u8g2_Setup_ssd1306_i2c_128x32_univision_1
```

Does the trick? (Other candidates for values of `TEST_DISPLAY` based on your description are `u8g2_Setup_ssd1306_i2c_128x32_winstar_2` and `u8g2_Setup_ssd1306_i2c_128x32_winstar_f`)

I have to admit that I never really used the u8g2 package myself. I could give it a look, though.