RIOT-OS/RIOT #16388: Pinetime board gets stuck in cpu_init()

arteeh, 2021-04-25:
#### Description
When creating an application for the Pinetime, cpu_init() and consequently board_init() will get stuck somewhere, meaning any code after that does not seem to run.

#### Steps to reproduce the issue
Create a simple application, with the RIOT folder checked out to 2021.01-branch. Make with the following definitions:
```
BOARD ?= pinetime
USEMODULE += ili9341
```
Create a main function where the following code is run:
```
board_init(); // or cpu_init() and setting the display registers manually
BACKLIGHT_ON;
ili9341_init(&display, &ili9341_params[0]);
ili9341_fill(&display, 0, 319, 0, 239, 0x0000);
ili9341_fill(&display, 10, 59, 10, 109, 0x001F);
```

I then run `PROGRAMMER=openocd DEBUG_ADAPTER=stlink make flash` with an ST-Link V2 wired up to my Pinetime to flash the program.

#### Expected results
The Pinetime display shows a blue rectangle on a black background, at low brightness.

#### Actual results
The Pinetime doesn't show any activity, and nothing happens on the screen.

#### Versions
Operating System: Fedora Silverblue 34
GCC: arm-none-eabi-gcc (Fedora 10.2.0-4.fc34) 10.2.0
CLIB: arm-none-eabi-newlib: "4.1.0"
Make: GNU Make 4.3
Debugger: Open On-Chip Debugger 0.11.0

kaspar030, 2021-04-25:
does `tests/driver_ili9341` display anything?

arteeh, 2021-04-25:
> does `tests/driver_ili9341` display anything?

The [test](https://github.com/RIOT-OS/RIOT/blob/master/tests/driver_ili9341/main.c) works well, but it doesn't use board_init() or cpu_init() which is what this issue is about. I just tried to add cpu_init() to the beginning of the test, and nothing shows up on the display.

kaspar030, 2021-04-25:
`board_init()` and `cpu_init()` are called automatically by RIOT's startup procedure, on every board, before `main()`. Are you calling it again in your application?

arteeh, 2021-04-25:
I didn't know board_init() gets run before main(). This issue is a misunderstanding of mine then, closing.

closed 2021-04-25 by arteeh