RIOT-OS/RIOT #16402: esp8266 precompiled bootloaders don't support partitions past 1MB

iosabi, 2021-04-27:
#### Description
The esp8266 cpu ships with some bootloader code pre-compiled from the vendor SDK. I couldn't find what revision and options where used to compile these bootloaders, they have the following bug:

If the partition table contains any partition that starts or ends at the 1MB boundary or after that you get the following error even if using a chip with 4MB flash:

```
I (69) boot: SPI Flash Size : 4MB
E (72) flash_parts: partition 3 invalid - offset 0x100000 size 0x2000 exceeds flash chip size 0x100000
E (81) boot: Failed to verify partition table
E (85) boot: load partition table error!
user code done
```

Note that the code detects the flash at 4MB but then complains about the partition being larger than 1MB.

#### Steps to reproduce the issue
Create a custom partition table with the vendor recommended OTA scheme, for example:

```
$ cat mypartitions.csv
# Name,   Type, SubType, Offset,   Size
nvs,      data, nvs,     0x9000,   0x6000
phy_init, data, phy,     0xf000,   0x1000
ota_0,    app,  ota_0,   0x010000, 0xf0000
otadata,  data, ota,     0x100000, 0x2000
ota_1,    app,  ota_1,   0x110000, 0xf0000
```

Then compile and run on a esp12 board the following example program:

```USEMODULE=esp_log_startup make -C tests/shell BOARD=esp8266-esp-12x PARTITION_TABLE_CSV=`pwd`/mypartitions.csv -j flash```

#### Expected results
The board should boot all the way to the shell.

#### Actual results
The ROM bootloader loads the flash bootloader at 0x0000 and prints messages in the UART at 74880 bps while doing that; and then the flash bootloader switches to 115200 and prints the following error:

```
I (69) boot: SPI Flash Size : 4MB
E (72) flash_parts: partition 3 invalid - offset 0x100000 size 0x2000 exceeds flash chip size 0x100000
E (81) boot: Failed to verify partition table
E (85) boot: load partition table error!
```

iosabi, 2021-04-27:
I was able to compile a new bootloader from scratch from the vendor repo and pass it as `BOOTLOADER_BIN` (unfortunately that needs a relative path in esptool.inc.mk, but ok, doable) and boot with this partition table. We are using in RIOT a now aging version of the SDK from @gschorcht fork.

Could we make the SDK dependency a PKG and do the symbol manipulation needed in the PKG Makefile instead? Is there any problem with that approach?

I can also just update the bootloader binaries, but if I can build the bootloader from source I'd rather do that than ship a binary.

gschorcht, 2021-04-27:
> Could we make the SDK dependency a PKG and do the symbol manipulation needed in the PKG Makefile instead? Is there any problem with that approach?

Indeed, we already had the idea to define the SDK as a package as it is done for the Geko SDK for EFM/EFR/EZR32 MCUs. However, at that time the packages were downloaded for each application separately to the application's bin directory. This means that when compiling all test applications, for example, one would have had a lot of copies of the SDK, each 112 MB in size. Therefore, we provided a preconfigured SDK for download. Now that packages are downloaded into a separate directory, the SDK could be implemented as a package with according modifications via the Makefile.

However, having the SDK as a package does not solve the problem of the aging SDK, because the RIOT port of the ESP8266 requires much more changes than renaming some symbols. Some parts of the SDK source code have to be extracted and need to be changed. Also, a FreeRTOS adaptation layer has to be realized to link RIOT with the functions in the SDK that assume to work on a FreeRTOS. Therefore, the RIOT port is quite tricky.

Another problem is that the interface how FreeRTOS functions are used by the binary libraries of the SDK has been completly changed from version 3.1 to version 3.3. Before version 3.3, there was an OS interface that defines an abstract version of used FreeRTOS functions that could be linked to functions implemented by RIOT. Since version 3.3 the SDK links directly to FreeRTOS functions. I had to learn this last when I upgraded the RIOT to the ESP8266_RTOS_SDK version 3.3. I had to change a lot. The upgrade was almost finished and I had only one remaining problem with the hardware timer when WiFi was enabled. But then Corona came and I had absolutely no time anymore to continue working on this upgrade.

iosabi, 2021-04-30:
@gschorcht  thanks for your explanation and even more for the work you did to bring up ESP here.

I looked at the [release/v3.1-for-riot-os](https://github.com/gschorcht/RIOT-Xtensa-ESP8266-RTOS-SDK/commits/release/v3.1-for-riot-os) branch in your fork and it is only 2 commits ahead of https://github.com/gschorcht/RIOT-Xtensa-ESP8266-RTOS-SDK/commit/913a06a9ac3b2f18009e8fee8f092ca9ffeccd38 which is the first parent in espressif's 3.1 branch. In those two commits all I see is the binary .a files of three libraries being edited. I looked at the diff between the symbols and made a list of the symbol renames you did. Basically reverse engineering what you did to be able to replicate the same change. From what you said my understanding is that there's nothing blocking *today* a change to stop using your fork and instead create a SDK pkg/ to pull espressif at that specific SHA and apply the same symbol renaming you did, given that the SDK repo will be pulled only once (to $RIOTBASE/build) and even if we need to make a copy of all the vendor's components .a per app they are only 2.5MB. 

I think that the advantage of doing this is that it would be documented what changes were made to the vendor SDK so anybody else can come and fix stuff there. There's a bug elsewhere about documenting how were the toolchain packages created which is a similar problem. You are now busy so I think that enabling other people to help would be a great thing to have. For example, espressif have added more commits in their release/v3.1 branch after the 913a06a9ac3b2f18009e8fee8f092ca9ffeccd38 you used as a parent so even if upgrading to 3.3 is complicated, if we had the documentation of what changes were made or even better the tooling to make those automatically we would be able to upgrade to the latest 3.1* release more easily and maybe fix some of these issues (they have released tags 3.1.1 and 3.1.2 since).

I'll prepare a commit to create a pkg/esp8266_sdk following the Gecko SDK example and rename the symbols in their binary libraries in that pkg at build time. My idea is to write a python script to do this rename based on a simple rename table in python. It would be great to be able to use pyelftools but I think I can manage with just the toolchain. Does this sound good? Do you have any advice about this? how did you do the symbol renaming?

iosabi, 2021-05-06:
I pushed this branch: https://github.com/iosabi/RIOT/tree/esp8266_bldr which builds the bootloader from source from the SDK code and it fixes this particular issue. That branch depends on the SDK PR #16425 so I'll wait until that on is merged.

closed 2023-01-03 by bors[bot]