RIOT-OS/RIOT #16407: SX127x driver test

hoekstraarie, 2021-04-28:
#### Description

Test code for sx127x driver runs on board, but cannot setup frequency.
#### Steps to reproduce the issue

We compiled the code and flashed it to my stm32f0discovery board. It said it initialized the sx1276 module but we cannot change frequency. We got it working a few times but that was more by luck. 

We connected our sx1276 module the following way:

MISO -> PA6
MOSI -> PA7
SCK -> PA5
NSS -> PA10
RESET -> PA11
GND -> GND
DIO0 -> PA12
DIO1 -> PA13
3V3 -> 3V3

When we flash our stm32f0discovery board we first have to take out the clk cable else it won't flash the board. After we flash the board we can put the clk cable back in and reset the board, than it recognizes the module.

I changed the gpio pins to our own setup in the params file of the sx127x driver. 

#### Expected results

We expect the module to send the payload to our already set up gateway that is working. 
#### Actual results

[image]

Here you can see that I want to change the frequency but it does not change anything. We got it working a few times like I earlier mentioned. We also got it in listening mode once, and received the same payload as we received on our gateway. So that means the lora module works right?

#### Versions

Operating system: Linux Ubuntu on a windows host with virtual box
Build environment: ARM toolchain

fjmolinas, 2021-04-29:
Hi @hoekstraarie reposting from #16369:

> Sorry I missed your answer. Can you maybe enable debug output in the sx1276 driver code? It might be stuck there?
> 
> Also is the chip you are using does it actually integrate a sx1276x? Maybe its not the exact same LoRa chip...

So what is the exact LoRa module you have?

hoekstraarie, 2021-04-29:
Hi @fjmolinas ,

My exact module is a rfm95w module,

[lora module](https://nl.rs-online.com/web/p/lpwan/1251259/?cm_mmc=NL-PLA-DS3A-_-google-_-PLA_NL_NL_Semiconductors_Whoop-_-(NL:Whoop!)+LPWAN-_-1251259&matchtype=&pla-326768158449&gclid=Cj0KCQjwsqmEBhDiARIsANV8H3aLLTgzNgoFIvvs74jDvkwoEB9PWrwJU_9Sg8Yhiiie_zv3_XU0uPMaAig6EALw_wcB&gclsrc=aw.ds)

It worked one time we received some data when we were in listen mode. But now it seems it doesn't do anything anymore.

fjmolinas, 2021-04-29:
> Hi @fjmolinas ,
>
> My exact module is a rfm95w module,
>
> [lora [...]

Ahh so a sx1272, are you specifying that when compiling the application?

hoekstraarie, 2021-04-29:
Hi @fjmolinas ,

It is the same module but on our module it says sx1276 so that's why we thought it was a sx1276 look at my picture included.

[image]

Excuse me for our soldering, still a work in progress :)

fjmolinas, 2021-04-29:
> Hi @fjmolinas ,
> 
> It is the same module but on our module it says sx1276 so that's why we thought it was a sx1276 look at my picture included.
> Excuse me for our soldering, still a work in progress :)

Ah yes the the driver module seems to be fine, I said sx1272 because it was mentioned in the link you inserted before.

hoekstraarie, 2021-04-29:
Hi @fjmolinas ,

That's true, that's why I think it's weird that it is not working.

fjmolinas, 2021-04-29:
Well here is what I would do:

- first thing would be to check your soldering, maybe something is loose?
- check the exact version of your module sine they don't all support all frequency bands, and the default one in RIOT is th 868Mhz one
- verify you can communicate by spi: e.g.: read some registers check that the content makes sense
- check that the GPIO's you are using are note connected/used by another module in RIOT
- its also sometimes useful to test your hardware with some code from the manufacturer (if there is any), at least you can verify that the chip is operational

Let me know how that goes, but erradic behaviour might mean faulty electrical contacts.

hoekstraarie, 2021-04-29:
Hi @fjmolinas ,

We checked the soldering and there were some things that could be better so we made some changes to it. We can initialize the device now. I think it's because of the usart breakoutboard that is connected to flash our board, we can flash the board when we take out the spi clk pin and when we connect the spi clk pin again it can initialize the module again. When let the clk pin connected it cannot flash the board for some reason. So I think the usart and spi are conflicting.

Thanks for the tips!

fjmolinas, 2021-04-30:
>  We can initialize the device now.

Just initialize or communicate?

hoekstraarie, 2021-04-30:
Hi @fjmolinas ,

We can init the device and set the frequency and send with it. But we cannot see the payload on our gateway.

fjmolinas, 2021-04-30:
> Hi @fjmolinas ,
> 
> We can init the device and set the frequency and send with it. But we cannot see the payload on our gateway.

So you have multiple devices? Can you sniff packets between devices?

hoekstraarie, 2021-04-30:
Hi @fjmolinas ,
 
No we have one device that is sending to our gateway. We tested the lora module with an arduino and we can send data to it. But on the stm32f0discovery board with the sx127x driver test we can set frequency and send payload but can't see data on the gateway.

akshaim, 2021-05-04:
> No we have one device that is sending to our gateway. We tested the lora module with an arduino and we can send data to it. But on the stm32f0discovery board with the sx127x driver test we can set frequency and send payload but can't see data on the gateway.

Hey, Did you check the CR, SF values ? Does it match with the LoRa GW configuration ?

closed 2021-05-04 by hoekstraarie

akshaim, 2021-05-04:
@hoekstraarie  Did you solve the issue ?

hoekstraarie, 2021-05-04:
Yeah we solved it. We ordered a new board with built in Lora. Our teacher recommended that because he wasn’t sure that it was possible with the old board.